esphome/issues #279: Unable to find action with the name 'output.tun_off', is the component loaded?.

kquinsland, 2019-05-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**
I am using a [d1 mini pro](https://wiki.wemos.cc/products:d1:d1_mini_pro)
```
esphome:
  name: test_motion
  platform: ESP8266
  board: d1_mini_pro
```
**Affected component:**
I am not 100% sure, i think [output / https://esphome.io/components/output/gpio.html](https://esphome.io/components/output/gpio.html).

**Description of problem:**

This _might_ be simple user-error / misunderstanding.
I am trying to toggle a gpio output (the builtin LED) when a binary sensor state changes. Namely, the LED should be on when the binary sensor / GPIO in is high, and the led should be off when low.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test_motion
  platform: ESP8266
  board: d1_mini_pro

# See: https://esphome.io/components/wifi.html
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass

# Enable logging
logger:

# Enable status LED; very low resolution logging :)
# See: https://esphome.io/components/status_led.html
status_led:
  # Wemos uses GPIO2 for the built in LED
  pin: 
    number: GPIO2
    # It needs an inversion (active low)
    inverted: True

# We *also* want to use the status LED to indicate when motion is detected
#   to do this, we need to define an output
output:
  - platform: gpio
    pin: GPIO2
    id: status_led

# See: https://esphome.io/components/mqtt.html
mqtt:
  broker: mqtt.host
  username: !secret mqtt_user
  password: !secret mqtt_pass

# the Lux and BME sensor will be on software i2c bus
# See: https://esphome.io/components/i2c.html
i2c:
  # d1_mini_pro uses 4,5
  # See: https://wiki.wemos.cc/products:d1:d1_mini_pro
  sda: 4
  scl: 5
  # Does a default bus scan on startup
  scan: True

# the PIR and "alive" sensor are binary
binary_sensor:
  # PIR
  # See: https://esphome.io/cookbook/pir.html
  ##
  # Pin14 is wired up to motion sensor
  - platform: gpio
    pin: 14
    name: "Motion Sensor"
    device_class: motion
    id: pir
    on_press:
      then:
        - output.tun_on: status_led
    on_release:
      then:
        - output.tun_off: status_led

  # A "connected?" status sensor
  # See: https://esphome.io/components/binary_sensor/status.html
  - platform: status
    name: "Motion Sensor Status"

# See: https://esphome.io/components/sensor/tsl2561.html
sensor:
  - platform: tsl2561
    name: "Ambient Light"
    address: 0x39
    update_interval: 15s

  # See: https://esphome.io/components/sensor/bme280.html
  - platform: bme280
    temperature:
      name: "Temperature"
      id: bme280_temperature
    pressure:
      name: "Air Pressure"
      id: bme280_pressure
    humidity:
      name: "Relative Humidity"
      id: bme280_humidity
    address: 0x76
    update_interval: 15s
    # A bit of maths to do some useful inference
    # See: https://esphome.io/cookbook/bme280_environment.html
  - platform: template
    name: "Absolute Humidity"
    lambda: |-
      const float mw = 18.01534;    // molar mass of water g/mol
      const float r = 8.31447215;   // Universal gas constant J/mol/K
      return (6.112 * powf(2.718281828, (17.67 * id(bme280_temperature).state) /
        (id(bme280_temperature).state + 243.5)) * id(bme280_humidity).state * mw) /
        ((273.15 + id(bme280_temperature).state) * r); // in grams/m^3
    accuracy_decimals: 2
    update_interval: 15s

```

**Traceback (if applicable):**

```
There is no stacktrace to share, just a compilation error
```

**Additional information and things you've tried:**

The error during (pre) build:

```
  on_press:  [source test.yaml:79]
    then:  [source test.yaml:80]
      - [source test.yaml:80]

        Unable to find action with the name 'output.tun_on', is the component loaded?. Got 'status_led'
        output.tun_on: status_led
  on_release:  [source test.yaml:82]
    then:  [source test.yaml:83]
      - [source test.yaml:83]

        Unable to find action with the name 'output.tun_off', is the component loaded?. Got 'status_led'
        output.tun_off: status_led
```

As soon as i comment out the `on_press` and `on_release` components, i have no issue compiling the code.

Because i am only commenting out the "trigger" and *not* getting an error about the `output` yaml block, i do believe that the `output` component *is* properly loaded.  The docs certainly do make it appear that `output` objects have a `turn_on` and `turn_off` method which takes the ID that they are to act on; `status_led` in this case.

Things i have tried:

- change the `id: status_led` to `id: out_led` and adjust the `output.tun_{on,off}` directives
- disable the `status_led:` component alltogether

Please let me know if there's anything else you need from me :).

HFsi, 2019-05-01:
The action is turn_on and turn_off, not tun_on and tun_off

kquinsland, 2019-05-01:
**doh**.

Ticket: **closed**
Reason: **user is idiot**

_sigh_.

closed 2019-05-01 by kquinsland

OttoWinter, 2019-05-01:
On the positive side: in the latest dev branch the validator will suggest close matches for unknown action names :)