esphome/issues #282: No ACK / ACK time-out using a Nextion NX3224T024 connected to Wemos D1 Mini #281

JBS5, 2019-05-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Using the ESPHome 1.12.2 add-on on HASS.io running on a Intel NUC with Ubuntu.

**ESP (ESP32/ESP8266, Board/Sonoff):**
I am trying to get a Nextion NX3224T024 to work connected to a Wemos D1 Mini.

Connected like this:

[image]
[image]

**Affected component:**
https://esphome.io/components/display/nextion.html

**Description of problem:**
A screenshot of the upload followed by the errors:

[image]

**Problem-relevant YAML-configuration entries:**
```yaml
esphomeyaml:
  name: touchscreen
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: 'WLANJBSSAM'
  password: '7W7Zt66ybVpUajc3SyFx'
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.1.221
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.1.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0  

uart:
  rx_pin: D4
  tx_pin: D7
  baud_rate: 9600
  
binary_sensor:
  - platform: nextion
    page_id: 0
    component_id: 2
    name: "Nextion Component 2 Touch"

display:
  - platform: nextion
    lambda: |-
      it.set_component_text("t0", "hello");

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

jascdk, 2019-05-02:
I am having the exact same issue:(

bigstation17, 2019-05-02:
I have exact same problem :(
Commands are working but not all the time

HFsi, 2019-05-03:
Same here.....switching the pins give the same result

lordfiSh, 2019-05-03:
@JBS5 you may want to remove/mask your wlan config ;)

JBS5, 2019-05-03:
@lordfiSh Thanks :-) replaced that part of the YAML in the code.

jascdk, 2019-05-03:
@JBS5 just wondering. Have you tried to use a esp32 instead of a esp8266? I have made some other displays (the PCF8574) using the esp8266 , and it messed up. Was not working good. After I switched to esp32 it runs smoothly. Just a thought

HFsi, 2019-05-03:
I have a PCF8574 controlled by a esp01 (the cheapest on earth, I think) working 24/7 with no issues at all.
The esp32 test is a good idea, I'll do it when the ones I bought arrive

JBS5, 2019-05-03:
@jascdk Will try it with an esp32, I do not have some available so I have to order one, takes some time.
Not sure if this one is suitable https://www.amazon.com/Gikfun-ATmega328-Micro-controller-Arduino-EK1620x3/dp/B00SGMEH7G ?

bigstation17, 2019-05-03:
I tried it with esp32, it's exactly the same errors.

JBS5, 2019-05-03:
Thanks @bigstation17 for your feedback, then I save myself the time to try it with a esp32 and hope there will be some input how to fix this problem. Or, happy to help if there is more input needed to addresss the issue.

bigstation17, 2019-05-03:
Yes for sure. I saw in dev version that are more commands for nextion that are undocumented for now. I paused my nextion project and i will wait for next esphome version :)

JBS5, 2019-05-03:
Just installed the HA SwitchPlate project (https://github.com/aderusha/HASwitchPlate) again on my Nextion display and Wemos D1 Mini to make sure both are still ok and that works without any problem.

Should be an ESPHome thing for sure...

thedexboy, 2019-05-05:
*** Possible solution at the end of my ranting! ***

One quick side note though, you might want to use a level shifter for your project. I mean, I've read all sorts of things and people say it's working and all but ESP's work on 3.3 V and Nextion works on 5 V.

Secondly, this is the error I'm getting:
```
[15:06:51][I][logger:119]: Log initialized
[15:06:51][C][ota:484]: There have been 0 suspected unsuccessful boot attempts.
[15:06:51][I][application:055]: Running through setup()...
[15:06:51][C][uart:111]: Setting up UART bus...
[15:06:51][W][display.nextion:041]: Waiting for ACK timed out!
[15:06:51][W][display.nextion:088]: Sending command 'bkcmd=3' failed because no ACK was received
[15:06:51][W][display.nextion:041]: Waiting for ACK timed out!
[15:06:51][W][display.nextion:088]: Sending command 'page 0' failed because no ACK was received
[15:06:51][C][wifi:027]: Setting up WiFi...
```

My code from ESPHome:
```
esphome:
  name: delavnica_nextion
  platform: ESP8266
  board: d1_mini_pro

wifi:
  networks:
  - ssid: !secret primarywifi
    password: !secret primarywifipass
  manual_ip:
    static_ip: !secret ip_delavnicaNextion
    gateway: 192.168.2.254
    subnet: 255.255.255.0
    dns1: 192.168.2.22
    dns2: 192.168.2.22

# Enable Home Assistant API
#api:

web_server:
  port: 80

ota:
  safe_mode: True
  password: !secret otapass

# Enable logging
logger:
#  level: VERY_VERBOSE

uart:
  rx_pin: D7
  tx_pin: D4
  baud_rate: 115200

binary_sensor:
  - platform: nextion
    page_id: 0
    component_id: 6
    name: "Nextion beep button"
    id: nextion_beep_button

display:
  - platform: nextion
    lambda: |-
      if (id(nextion_beep_button).state) {
        it.set_component_text("textbox", "It's alive!");
      } else {
      }
```

What I've noticed is now I don't need to enable `bkcmd=3` anymore. I've connected 2 additional USB to UART adapters and logged the whole communication from ESP and Nextion display (both ways).
ESP sends the correct data and Nextion replies correctly. For some reason, ESPHome doesn't read the data back - it did receive it!

----------

While writing this reply, I moved tx_pin to D6 instead of D4. Everything started working. W. T. F.?

thedexboy, 2019-05-05:
One more thing I noticed:
```
[17:24:48][W][display.nextion:149]: Received unknown filler end bytes from Nextion!
[17:24:48][W][display.nextion:149]: Received unknown filler end bytes from Nextion!
[17:24:49][W][display.nextion:160]: Nextion reported invalid instruction!
```

I get quite a bit of these when pushing buttons. Could this be because of softwareserial?
Once I get the thang of things, I'll switch over to real UART and disable logging - or push logging to softwareserial.

OttoWinter, 2019-05-08:
@thedexboy I see you have enabled VERY_VERBOSE logs, could you share those logs?

> While writing this reply, I moved tx_pin to D6 instead of D4. Everything started working. W. T. F.?

Interesting, it could be softwareserial, but on the other hand the D6 pin on that board has no hw serial attached. So both are using a soft serial - maybe a pin is damaged?

bigstation17, 2019-05-08:
```[19:12:34][D][binary_sensor:037]: 'button test': Sending state OFF
[19:12:34][W][display.nextion:149]: Received unknown filler end bytes from Nextion!
[19:12:34][W][display.nextion:041]: Waiting for ACK timed out!
[19:12:34][W][display.nextion:088]: Sending command 'n0.val=0' failed because no ACK was received
[19:12:34][W][display.nextion:149]: Received unknown filler end bytes from Nextion!
[19:12:34][W][display.nextion:041]: Waiting for ACK timed out!
[19:12:34][W][display.nextion:088]: Sending command 't1.txt="release"' failed because no ACK was received
[19:12:34][D][binary_sensor:037]: 'button test': Sending state ON
[19:12:34][W][display.nextion:149]: Received unknown filler end bytes from Nextion!
[19:12:34][W][display.nextion:041]: Waiting for ACK timed out!
[19:12:34][W][display.nextion:088]: Sending command 'n0.val=1' failed because no ACK was received
[19:12:34][W][display.nextion:149]: Received unknown filler end bytes from Nextion!
[19:12:35][W][display.nextion:041]: Waiting for ACK timed out!
[19:12:35][W][display.nextion:088]: Sending command 't1.txt="press"' failed because no ACK was received 

with VERY_VERBOSE

[19:16:07][D][binary_sensor:037]: 'button test': Sending state OFF
[19:16:07][VV][uart:176]:     Wrote "n0.val=0"
[19:16:07][VV][uart:165]:     Wrote 0b11111111 (0xFF)
[19:16:07][VV][uart:165]:     Wrote 0b11111111 (0xFF)
[19:16:07][VV][uart:165]:     Wrote 0b11111111 (0xFF)
[19:16:07][VV][uart:186]:     Read 0b00000001 (0x01)
[19:16:07][VV][uart:186]:     Read 0b11111111 (0xFF)
[19:16:07][VV][uart:186]:     Read 0b11111111 (0xFF)
[19:16:07][VV][uart:186]:     Read 0b11111111 (0xFF)
[19:16:07][VV][uart:176]:     Wrote "t1.txt="release""
[19:16:07][VV][uart:165]:     Wrote 0b11111111 (0xFF)
[19:16:07][VV][uart:165]:     Wrote 0b11111111 (0xFF)
[19:16:07][VV][uart:165]:     Wrote 0b11111111 (0xFF)
[19:16:07][VV][uart:186]:     Read 0b00000001 (0x01)
[19:16:07][VV][uart:186]:     Read 0b11111111 (0xFF)
[19:16:07][VV][uart:186]:     Read 0b11111111 (0xFF)
[19:16:07][VV][uart:186]:     Read 0b11111111 (0xFF)
[19:16:07][D][binary_sensor:037]: 'button test': Sending state ON
[19:16:07][VV][uart:176]:     Wrote "n0.val=1"
[19:16:07][VV][uart:165]:     Wrote 0b11111111 (0xFF)
[19:16:07][VV][uart:165]:     Wrote 0b11111111 (0xFF)
[19:16:07][VV][uart:165]:     Wrote 0b11111111 (0xFF)
[19:16:07][VV][uart:186]:     Read 0b00000001 (0x01)
[19:16:07][VV][uart:186]:     Read 0b11111111 (0xFF)
[19:16:07][VV][uart:186]:     Read 0b11111111 (0xFF)
[19:16:07][VV][uart:186]:     Read 0b11111111 (0xFF)
[19:16:07][VV][uart:176]:     Wrote "t1.txt="press""
[19:16:07][VV][uart:165]:     Wrote 0b11111111 (0xFF)
[19:16:07][VV][uart:165]:     Wrote 0b11111111 (0xFF)
[19:16:07][VV][uart:165]:     Wrote 0b11111111 (0xFF)
[19:16:07][VV][uart:186]:     Read 0b00000001 (0x01)
[19:16:07][VV][uart:186]:     Read 0b11111111 (0xFF)
[19:16:07][VV][uart:186]:     Read 0b11111111 (0xFF)
[19:16:07][VV][uart:186]:     Read 0b11111111 (0xFF)
[19:16:07]pm open,type:0 0
[19:16:08][VV][component:126]: Running interval 'update':0 with interval=5000 last_execution=13243 (now=18249)```

thedexboy, 2019-05-08:
@OttoWinter if I uncomment and try getting very_verbose logs, ESP won't boot - it will get stuck in a bootloop.

mmotley999, 2019-05-10:
FWIW, I'm seeing the same behavior on a Wemos D1 mini clone, even when I should be using the hardware UART0 (I think?) and moving logging to UART1.  Perhaps disabling serial logging altogether would help? I haven't tried that yet.

```
logger:
  hardware_uart: UART1
  id: mainlog

uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 19200
```

However, I'm not seeing these errors when I converted the same code to an ESP32 board... I put it through it's paces with no ACK errors:

```
uart:
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 19200
```

OttoWinter, 2019-05-10:
> However, I'm not seeing these errors when I converted the same code to an ESP32 board... I put it through it's paces with no ACK errors:

That probably means the issue is with software serial indeed. ESP32 has 3 HW serial controllers on all pins (which esphome uses automatically).

However, when using tx_pin GPIO1 and rx_pin GPIO3 on ESP8266, it does use the UART0 hw serial too - weird.

OttoWinter, 2019-07-11:
Closing due to inactivity

closed 2019-07-11 by OttoWinter