esphome/issues #286: ESP8266 power loss, cant reconnect to WIFI

det-peralta, 2019-05-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

hassio
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**Affected component:**

WIFI

**Description of problem:**
I created a code to switch/binarysensor/cover, its basically to open a two garage doors, however im testing removing the power cable, to simulate a power loss. My problem is that after the power loss, the esp never reconnects again to the wifi.
How can i fix this poblem?

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp_demo
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "ssid"
  password: "pass"
  manual_ip:
    static_ip: 192.168.1.200
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1
    dns2: 1.1.1.1

logger:

api:

ota:

cover:

  - platform: template
    name: Garage Door 1
    id: garage_door1
    lambda: |-
      if (id(sensor1).state) {
        return cover::COVER_OPEN;
      } else {
        return cover::COVER_CLOSED;
      }
    open_action:
      - switch.turn_on: momentary_button1
    close_action:
      - switch.turn_on: momentary_button1
    stop_action:
      - switch.turn_on: momentary_button1
      
  - platform: template
    name: Garage Door 2
    id: garage_door2
    lambda: |-
      if (id(sensor2).state) {
        return cover::COVER_OPEN;
      } else {
        return cover::COVER_CLOSED;
      }
    open_action:
      - switch.turn_on: momentary_button2
    close_action:
      - switch.turn_on: momentary_button2
    stop_action:
      - switch.turn_on: momentary_button2

binary_sensor:
  - platform: gpio
    pin: 
      number: D2
      mode: INPUT_PULLUP
    id: sensor1
    internal: True
  - platform: gpio
    pin: 
      number: D3
      mode: INPUT_PULLUP
    id: sensor2
    internal: True

switch:

  - platform: gpio
    id: relay1
    pin: GPIO5
    restore_mode: ALWAYS_OFF
    internal: True
    
  - platform: template
    id: momentary_button1
    turn_on_action:
      - switch.turn_on: relay1
      - delay: 150ms
      - switch.turn_off: relay1
    internal: True
    
  - platform: gpio
    id: relay2
    pin: GPIO12
    restore_mode: ALWAYS_OFF
    internal: True
    
  - platform: template
    id: momentary_button2
    turn_on_action:
      - switch.turn_on: relay2
      - delay: 150ms
      - switch.turn_off: relay2
    internal: True
    

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
I tried 1.12.2 and the dev version 1.13.0-dev, same problem

det-peralta, 2019-05-08:
i have solve the problem however, i didnt make any change

closed 2019-05-08 by det-peralta

OttoWinter, 2019-06-18:
Moving the discussion to #455