esphome/issues #287: SH1106 128x64 I2C Display not working correctly

lordfiSh, 2019-05-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**
NodeMCU Lua Lolin V3 Module ESP8266 ESP-12E

**Affected component:**
https://esphome.io/components/display/ssd1306.html

**Description of problem:**
Only a few lines are working, updating works too, rest is garbage
https://media.discordapp.net/attachments/429907082955718657/572140538900512769/unknown.png

maybe relevant:

- https://github.com/keirf/FlashFloppy/issues/219
- https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/issues/228

**Problem-relevant YAML-configuration entries:**
```yaml
i2c:
  sda: D1
  scl: D2
display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    reset_pin: D0
    address: 0x3C
    lambda: |-
      it.print(0, 0, id(font), "Hello World!");
```

**Traceback (if applicable):**
n/a
```

```

**Additional information and things you've tried:**
n/a

randybb, 2019-05-05:
I have the same NodeMCU, the same display with I2C interface (it has no reset pin, so you can remove it) and it work like a charm. Do you have pull up resistors on SDA/SCL?

lordfiSh, 2019-05-07:
@randybb no, basicly this: https://esphome.io/cookbook/display_time_temp_oled.html

OttoWinter, 2019-05-07:
Yeah the second link you posted might be related.

Have you tried `"SH1106 128x64"` as model? As your title states, that should be the correct model.

randybb, 2019-05-07:
I had the same setup based the cookbook on my desk, had no issue to run it. Still have it as base for my future heating controller. My model is "SSD1306 128x64", but as I have mentioned - pullup resistors are needed for most of ESP8266 modules - there are no internal pullups, some board have them on GPIO4 and GPIO5,...

lordfiSh, 2019-05-08:
> Have you tried `"SH1106 128x64"` as model? As your title states, that should be the correct model.

thanks!
[image]

closed 2019-05-08 by lordfiSh