esphome/issues #288: Impossible to dim Sonoff B1 using brightness alone

MTrab, 2019-05-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff B1

**Affected component:**

Dimming the light

**Description of problem:**
When dimming the light, you first have to set brightness to 0%, which dims the color LEDs. After this you can dim the white light using white level.
When calling the brightness from ie. Google Home thru Home Assistant, it's only the brightness that is adjusted - hence no brightness is set when using white light.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: bedroom_ceiling_1
  upper_devicename: Bedroom Ceiling 1
  platform: ESP8266
  board: esp01_1m

<<: !include .common.yaml

binary_sensor:
  - platform: status
    name: "$upper_devicename Status"

my9231:
  data_pin: GPIO12  # GPIO13 for AiLight
  clock_pin: GPIO14  # GPIO15 for AiLight
  num_channels: 6
  num_chips: 2

output:
  - platform: my9231
    id: output_blue
    channel: 0
  - platform: my9231
    id: output_red
    channel: 1
  - platform: my9231
    id: output_green
    channel: 2
  - platform: my9231
    id: output_warm_white
    channel: 4
  - platform: my9231
    id: output_cold_white
    channel: 5

light:
  - platform: rgb
    name: ${upper_devicename}
    red: output_red
    green: output_green
    blue: output_blue
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

brandond, 2019-05-03:
It appears you've set up the channels for a rgbww light, but then set up a rgb light - so I'm not really surprised that the white channels aren't getting changed. Do you see a different result if you use `platform: rgbww` instead of `platform: rgb`?

MTrab, 2019-05-04:
Sorry - for some reason I posted the wrong version of the config.
I have updated the initial post now, with the correct config.

With RGB I don't have the while levels (obviously) and dimming works - althou I can't set a true white as it doesn't use the white channels.

With RGBWW as platform it is as described in the issue.

OttoWinter, 2019-05-07:
This is intentional - white_value doesn't scale with brightness as otherwise it would be impossible to only control the white channel of a light. It's not ideal but a limitation by Home Assistant (and needs to be fixed there, see also https://github.com/home-assistant/architecture/issues/123)

closed 2019-05-07 by OttoWinter