esphome/issues #289: Incorrect VCC value on NodeMCU?

fredrikbaberg, 2019-05-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

 Add-on version: dev
 You are running the latest version of this add-on.
 System: HassOS 2.11  (armv7 / raspberrypi3)
 Home Assistant version: 0.92.2
 Supervisor version: 162

ESPHome version: 1.13.0.dev0

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCUv2 8266 (may not be original, purchased from AliExpress a few years ago).

**Affected component:**

https://esphome.io/components/sensor/adc.html?highlight=adc#esp8266-measuring-vcc

**Description of problem:**

I'm powering the board through USB port. When using the ADC cmponent to measure VCC, the reported value is 64.0V. I've tried to power the board through computer USB port and USB powerbank with the same result, and with two different boards.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

esphome:
  name: blue
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_key
  fast_connect: true

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret esphome_password

# Enable OTA updates
ota:
  password: !secret esphome_password

sensor:
# Voltage
- platform: adc
  pin: VCC
  name: "Blue VCC Voltage"

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

fredrikbaberg, 2019-05-04:
I now tried to change to the normal (not beta or dev) version, removed the build folder from `config/esphome/` (since I got error when trying to upload), and it now reports a more reasonable value of 2.96V.

```
 Add-on version: 1.12.2
 You are running the latest version of this add-on.
 System: HassOS 2.11  (armv7 / raspberrypi3)
 Home Assistant version: 0.92.2
 Supervisor version: 162
```

fredrikbaberg, 2019-05-09:
It seems to report 0.0V when tried with the dev version (again through Hass.IO dev addon). That is probably better than 64V, but still not correct.

Since this was for the dev branch, not the main or beta, it is OK for me if you want to close this issue.

OttoWinter, 2019-05-10:
progress :)

Anyway, now it should be fixed (at least is giving me reasonable values now).

closed 2019-05-10 by OttoWinter