esphome/issues #290: Native API service not showing up in HA

ToSa27, 2019-05-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**Affected component:**

Native API user defined service

**Description of problem:**
I'm working on a secure RFID component for EV1 DESfire cards based on the existing pn532 component but I don't think the issue is related to that component: Creating a user-defined service as outlined [here](https://esphome.io/components/api.html#user-defined-services) works fine and compiles ok without any errors or warnings but after detection of the ESP by HA I would have expected the service to show up on the developer tools services page in HA. I don't see any errors in the HA log or the ESP logs that appear related. The version below is already trimmed down to the minimum (no variables etc.) to eliminate potential failure sources...

**Problem-relevant YAML-configuration entries:**
```yaml
api:
  services:
    - service: new_card
      then:
        - pn532.encode: rfid_reader
```

**Traceback (if applicable):**

```
no traceback - no error found in logs
```

**Additional information and things you've tried:**

ToSa27, 2019-05-04:
I just added another switch to the device, compiled and uploaded but the new switch is not showing up - so maybe the issue is completely unrelated to the API service and the device integration is not working properly for some other reason. I first deleted the integration through the UI but when it was detected again it did not have any entities listed. Then deleted from the registry (removed all possible references from core.config_entries / core.device_registry / core.entity_registry) and restarted HA but still the same - integration was detected without adding any entities...
What is the most appropriate way to ensure that a device is completely redetected after uploading a fresh firmware with e.g. some new components?

ToSa27, 2019-05-04:
Tested with a different ESP with different hardware (no RFID reader) which is based on ESPHome 1.12.2 and it's working fine there. The service name is actually the concatenation of the device id and the service name. Shows up without any manual intervention (no need to "trigger" the re-detection of the device by deleting something in .storage). I'll try with a plain 1.13.0-dev next without my modifications to ensure I didn't break it (but I don't expect that as I only touched the pn532 component and added a few constants).

ToSa27, 2019-05-04:
Updates the same device that was working fine with 1.12.2 to 1.13.0-dev (unchanged dev branch) and the service is immediately gone. Seems to be an issue with services detection only in 1.13.0-dev because the switch and the cam are detected properly, just the service is gone.

OttoWinter, 2019-05-07:
Yeah that could be broken in 1.13 - the dev branch is a bit buggy in places right now due to a recent big refactor.

closed 2019-05-08 by OttoWinter