esphome/issues #292: Allow multiple xiaomi flower redundancy

jokerigno, 2019-05-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHOME running on Docker conntainer

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**Affected component:**

xiaomi_miflora

**Description of problem:**

I have many plants around the house tracked with mi Flora sensors.

I have setup a single yaml file (with different name as gateway1 gateway2 etc ) with the details of every flower sensor (MAC adress and sensor for tracking), so I thought I could move the plant around the house without change the gateway settings.

Issue is that if the same plant is tracked from 2 gateway all sensor are only in 1 integration and the plants next to the second gateway are missing (while they "see" info looking at the log). 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: gateway2
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "Casa Concordia"
  password: "3^y^F8Qhw*$8"

  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.2.209
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.2.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

switch:
  - platform: restart
    name: "Gateway 2 Restart"

esp32_ble_tracker:

binary_sensor:
#  - platform: esp32_ble_tracker
#    mac_address: E9:1D:FF:5C:1E:EA
#    name: "Mi Band 3"

#  - platform: esp32_ble_tracker
#    mac_address: 88:0F:10:AA:CD:6D
#    name: "Bilancia Xiaomi"
    
sensor:
  - platform: xiaomi_miflora
    mac_address: C4:7C:8D:6A:39:57
    temperature:
      name: "Xiaomi MiFlora Temperature Cuciphotos"
    moisture:
      name: "Xiaomi MiFlora Moisture Cuciphotos"
    illuminance:
      name: "Xiaomi MiFlora Illuminance Cuciphotos"
    conductivity:
      name: "Xiaomi MiFlora Soil Conductivity Cuciphotos"
      
  - platform: xiaomi_miflora
    mac_address: C4:7C:8D:67:37:C9
    temperature:
      name: "Xiaomi MiFlora Temperature Adele"
    moisture:
      name: "Xiaomi MiFlora Moisture Adele"
    illuminance:
      name: "Xiaomi MiFlora Illuminance Adele"
    conductivity:
      name: "Xiaomi MiFlora Soil Conductivity Adele"
      
  - platform: xiaomi_miflora
    mac_address: C4:7C:8D:6A:39:77
    temperature:
      name: "Xiaomi MiFlora Temperature Irene"
    moisture:
      name: "Xiaomi MiFlora Moisture Irene"
    illuminance:
      name: "Xiaomi MiFlora Illuminance Irene"
    conductivity:
      name: "Xiaomi MiFlora Soil Conductivity Irene"
      
  - platform: xiaomi_miflora
    mac_address: C4:7C:8D:67:3B:B0
    temperature:
      name: "Xiaomi MiFlora Temperature Patty"
    moisture:
      name: "Xiaomi MiFlora Moisture Patty"
    illuminance:
      name: "Xiaomi MiFlora Illuminance Patty"
    conductivity:
      name: "Xiaomi MiFlora Soil Conductivity Patty"
      
  - platform: xiaomi_miflora
    mac_address: C4:7C:8D:6A:52:27
    temperature:
      name: "Xiaomi MiFlora Temperature Ortenzio"
    moisture:
      name: "Xiaomi MiFlora Moisture Ortenzio"
    illuminance:
      name: "Xiaomi MiFlora Illuminance Ortenzio"
    conductivity:
      name: "Xiaomi MiFlora Soil Conductivity Ortenzio"
      
  - platform: xiaomi_miflora
    mac_address: C4:7C:8D:67:7A:DD
    temperature:
      name: "Xiaomi MiFlora Temperature Vita"
    moisture:
      name: "Xiaomi MiFlora Moisture Vita"
    illuminance:
      name: "Xiaomi MiFlora Illuminance Vita"
    conductivity:
      name: "Xiaomi MiFlora Soil Conductivity Vita"
      
  - platform: xiaomi_miflora
    mac_address: C4:7C:8D:67:3B:9E
    temperature:
      name: "Xiaomi MiFlora Temperature Vita Grande"
    moisture:
      name: "Xiaomi MiFlora Moisture Vita Grande"
    illuminance:
      name: "Xiaomi MiFlora Illuminance Vita Grande"
    conductivity:
      name: "Xiaomi MiFlora Soil Conductivity Vita Grande"

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-05-07:
This has already been done in dev - The entity from each device will show up in HA as a separate entity. As for merging the states that will then need to be done with a template sensor (checking for `unknown` state) in HA - merging is not an ESPHome but a HA concern.

closed 2019-05-07 by OttoWinter

jokerigno, 2019-05-08:
If the mi flora entities are set on both gateway it is available only in the latter added. So I think issue is related to ESPHOME integration maybe.

For your information I solved the merging issue using min_max sensor with "last" attribute.

OttoWinter, 2019-05-08:
> If the mi flora entities are set on both gateway it is available only in the latter added.

yes, and that has been fixed in dev.

jokerigno, 2019-05-08:
sorry 👍