esphome/issues #293: esp32cam listed as board by espressif but not supported

y0ast, 2019-05-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32 CAM

**Affected component:**
ESP32 camera: https://esphome.io/components/esp32_camera.html

**Description of problem:**
https://esphome.io/devices/esp32.html states to visit http://docs.platformio.org/en/latest/platforms/espressif32.html for selecting the board type.

That page lists http://docs.platformio.org/en/latest/boards/espressif32/esp32cam.html#board-espressif32-esp32cam as a board, however setting the yaml to that board gives an error:

```
INFO Reading configuration...
ERROR Error while reading config: Could not find board 'esp32cam'. Valid boards are nodemcuv2, esp8285, wifinfo, esp01_1m, d1_mini, espduino, esp12e, heltec_wifi_kit_8, espinotee, espino, espectro, thingdev, xinabox_cw01, sparkfunBlynk, huzzah, phoenix_v1, phoenix_v2, d1, esp01, wio_node, esp07, thing, esp_wroom_02, modwifi, oak, wio_link, wifi_slot, gen4iod, nodemcu, wifiduino, esp210, d1_mini_lite, espresso_lite_v1, espresso_lite_v2, d1_mini_pro
```

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: camera
  platform: ESP32
  board: esp32cam
```

**Additional information and things you've tried:**

fabaff, 2019-05-05:
I can conform this error.

```bash
$ pip freeze | grep platformio
platformio==3.6.7
```

`esp-wrover-kit` doesn't work for me.

```bash
src/main.cpp: In function 'void setup()':
src/main.cpp:31:3: error: 'ESP32Camera' was not declared in this scope
ESP32Camera *esp32camera = App.register_component(new ESP32Camera("Camera"));
^
src/main.cpp:31:16: error: 'esp32camera' was not declared in this scope
```

y0ast, 2019-05-05:
It can be worth trying `esp32dev`. Both work for me.

OttoWinter, 2019-05-07:
@fabaff That is a different issue, see https://github.com/esphome/issues/issues/139

@y0ast Looks like that board has only been added recently - ESPHome has a delay in applying these new platformio package versions because historically they need some time to be bug-free.

apeeters, 2020-11-22:
I confirmed the `esp32cam`works with the current 1.15.3 Docker container.

closed 2020-11-23 by y0ast