esphome/issues #294: Sensor HDC1080 communication failed!

22MarioZ, 2019-05-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

PIO

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**Affected component:**

Sensor HDC1080

**Description of problem:**
When using the sensor HDC1080 I get the following error: 

>
[21:08:09][C][sensor.hdc1080:043]: HDC1080:
[21:08:09][C][sensor.hdc1080:044]: Address: 0x40
[21:08:09][E][sensor.hdc1080:046]: Communication with HDC1080 failed!
[21:08:09][C][sensor.hdc1080:048]: Update Interval: 60000 ms
[21:08:09][C][sensor.hdc1080:049]: Temperature 'temp'
[21:08:09][C][sensor.hdc1080:049]: Unit of Measurement: '°C'
[21:08:09][C][sensor.hdc1080:049]: Accuracy Decimals: 1
[21:08:09][C][sensor.hdc1080:049]: Icon: 'mdi:thermometer'
[21:08:09][C][sensor.hdc1080:050]: Humidity 'hum'
[21:08:09][C][sensor.hdc1080:050]: Unit of Measurement: '%'
[21:08:09][C][sensor.hdc1080:050]: Accuracy Decimals: 0
[21:08:09][C][sensor.hdc1080:050]: Icon: 'mdi:water-percent'
> 

Using the same sensor with the CloseCube library example the sensor is working without any problem.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: esp32hdc1080

esphome:
  name: $devicename
  platform: ESP32
  board: lolin32

wifi: !include common/wifi.yaml
mqtt: !include common/mqtt.yaml
ota: !include common/ota.yaml

logger:
  level: DEBUG

i2c:
  sda: 15
  scl: 13
  scan: False

sensor:
  - platform: hdc1080
    temperature:
      id: hdc1080_temp
      name: "temp"
      unit_of_measurement: "°C"
      icon: "mdi:thermometer"
      accuracy_decimals: 1
      filters:
        - offset: 0
    humidity:
      id: hdc1080_hum
      name: "hum"
      unit_of_measurement: "%"
      icon: "mdi:water-percent"
      accuracy_decimals: 0
    update_interval: 60s
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-05-08:
More information please, VERY_VERBOSE usb logs would be nice. Also make sure you're using the correct address (check with scan option in i2c).

22MarioZ, 2019-05-08:
Hello Otto

Here is the  requested logs:

> 
[20:47:23]?ׁJun  8 2016 00:22:57
[20:47:23]
[20:47:23]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[20:47:23]configsip: 0, SPIWP:0xee
[20:47:23]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[20:47:23]mode:DOUT, clock div:2
[20:47:23]load:0x3fff0018,len:4
[20:47:23]load:0x3fff001c,len:928
[20:47:23]ho 0 tail 12 room 4
[20:47:23]load:0x40078000,len:9280
[20:47:23]load:0x40080400,len:5860
[20:47:23]entry 0x40080698
[20:47:24][I][logger:119]: Log initialized
[20:47:24][VV][preferences:036]: LOAD 0: valid=YES, 0=0x00000002 1=0x13D89D5A (Type=233825507, CRC=0x13D89D5A)
[20:47:24][C][ota:484]: There have been 2 suspected unsuccessful boot attempts.
[20:47:24][VV][preferences:049]: SAVE 0: 0=0x00000003 1=0x60BCD276 (Type=233825507, CRC=0x60BCD276)
[20:47:24][VV][sensor.sensor:063]: Sensor(0x3ffb8f88)::add_filter(0x3ffb8870)
[20:47:24][VV][sensor.filter:035]: Filter(0x3ffb8870)::initialize(parent=0x3ffb8f88 next=0x0)
[20:47:24][I][application:055]: Running through setup()...
[20:47:24][V][application:056]: Sorting components by setup priority...
[20:47:24][V][esp32-hal-i2c.c:1483] i2cInit(): num=0 sda=15 scl=13 freq=0
[20:47:24][V][esp32-hal-i2c.c:1677] i2cSetFrequency(): freq=100000Hz
[20:47:24][V][esp32-hal-i2c.c:1692] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=100000Hz
[20:47:24][V][esp32-hal-i2c.c:1698] i2cSetFrequency(): Fifo delta=1
[20:47:24][V][esp32-hal-i2c.c:1677] i2cSetFrequency(): freq=50000Hz
[20:47:24][V][esp32-hal-i2c.c:1692] i2cSetFrequency(): cpu Freq=240Mhz, i2c Freq=50000Hz
[20:47:24][V][esp32-hal-i2c.c:1698] i2cSetFrequency(): Fifo delta=1
[20:47:24][C][sensor.hdc1080:030]: Setting up HDC1080...
[20:47:24][VV][i2c:063]: Beginning Transmission to 0x40:
[20:47:24][VV][i2c:100]:     Writing 0b00000010 (0x02)
[20:47:24][VV][i2c:100]:     Writing 0b00000000 (0x00)
[20:47:24][VV][i2c:100]:     Writing 0b00000000 (0x00)
[20:47:24][VV][i2c:068]:     Transmission ended. Status code: 0x02
[20:47:24][W][i2c:077]: Received NACK on transmit of address 0x40
[20:47:24][VV][component:051]: set_interval(name='update', interval=60000, offset=12041)
[20:47:24][C][wifi:027]: Setting up WiFi...
[20:47:24][V][wifi_esp32:033]: Enabling STA.
[20:47:24]I (185) wifi: wifi driver task: 3ffb5884, prio:23, stack:3584, core=0
[20:47:24]I (544) wifi: wifi firmware version: 6b44342
[20:47:24]I (544) wifi: config NVS flash: enabled
[20:47:24]I (546) wifi: config nano formating: disabled
[20:47:24]I (568) wifi: Init dynamic tx buffer num: 32
[20:47:24]I (568) wifi: Init data frame dynamic rx buffer num: 10
[20:47:24]I (568) wifi: Init management frame dynamic rx buffer num: 10
[20:47:24]I (572) wifi: Init static rx buffer size: 1600
[20:47:24]I (576) wifi: Init static rx buffer num: 4
[20:47:24]I (579) wifi: Init dynamic rx buffer num: 10
[20:47:24]I (653) wifi: mode : softAP (b4:e6:2d:b0:2f:2a)
[20:47:24]I (655) wifi: Init max length of beacon: 752/752
[20:47:24]I (655) wifi: Init max length of beacon: 752/752
[20:47:24][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 0 - WIFI_READY
[20:47:24][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 13 - AP_START
[20:47:24][V][wifi_esp32:337]: Event: WiFi AP start
[20:47:24][V][wifi_esp32:286]: Event: WiFi ready
[20:47:24]I (677) wifi: mode : sta (b4:e6:2d:b0:2f:29)
[20:47:24][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[20:47:24][V][wifi_esp32:295]: Event: WiFi STA start
[20:47:24][D][WiFiGeneric.cpp:342] _eventCallback(): Event: 2 - STA_START
[20:47:24][V][wifi_esp32:295]: Event: WiFi STA start
[20:47:24]I (701) wifi: Set ps type: 0
[20:47:24]
[20:47:24][I][wifi:167]: WiFi Connecting to 'UPC2F66A8D'...
[20:47:24][V][wifi:169]: Connection Params:
[20:47:24][V][wifi:170]:   SSID: 'UPC2F66A8D'
[20:47:24][V][wifi:175]:   BSSID: Not Set
[... 96 lines cut ...]
[20:47:29][C][sensor.mqtt:029]:   State Topic: 'esp32hdc1080/sensor/temp/state'
[20:47:29][C][sensor.mqtt:025]: MQTT Sensor 'hum':
[20:47:29][C][sensor.mqtt:027]:   Expire After: 300s
[20:47:29][C][sensor.mqtt:029]:   State Topic: 'esp32hdc1080/sensor/hum/state'
[20:47:29][I][application:109]: Running through first loop()
[20:47:29][I][application:136]: First loop finished successfully!
[20:48:12][VV][component:126]: Running interval 'update':0 with interval=60000 last_execution=4294955420 (now=48140)
[20:48:12][VV][i2c:063]: Beginning Transmission to 0x40:
[20:48:12][VV][i2c:100]:     Writing 0b00000000 (0x00)
[20:48:12][VV][i2c:068]:     Transmission ended. Status code: 0x02
[20:48:12][W][i2c:077]: Received NACK on transmit of address 0x40
[20:49:12][VV][component:126]: Running interval 'update':0 with interval=60000 last_execution=48124 (now=108129)
[20:49:12][VV][i2c:063]: Beginning Transmission to 0x40:
[20:49:12][VV][i2c:100]:     Writing 0b00000000 (0x00)
[20:49:12][VV][i2c:068]:     Transmission ended. Status code: 0x02
[20:49:12][W][i2c:077]: Received NACK on transmit of address 0x40
[20:49:24][I][ota:152]: Boot seems successful, resetting boot loop counter.
[20:49:24][VV][preferences:049]: SAVE 0: 0=0x00000000 1=0x0DEFE4E3 (Type=233825507, CRC=0x0DEFE4E3)
[20:50:12][VV][component:126]: Running interval 'update':0 with interval=60000 last_execution=108124 (now=168125)
[20:50:12][VV][i2c:063]: Beginning Transmission to 0x40:
[20:50:12][VV][i2c:100]:     Writing 0b00000000 (0x00)
[20:50:12][VV][i2c:068]:     Transmission ended. Status code: 0x02
[20:50:12][W][i2c:077]: Received NACK on transmit of address 0x40

However now I get different message...

OttoWinter, 2019-07-11:
Closing due to inactivity - see also https://github.com/esphome/issues/issues/504#issuecomment-509215008

closed 2019-07-11 by OttoWinter