esphome/issues #302: Compile api issue

aero83, 2019-05-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HassIO/Docker AMDx64

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini
**Affected component:**

https://esphome.io/components/sensor/bme280.html

**Description of problem:**
When attempting to compile to change or update  I get this in the compile window. 
`[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22AsyncMqttClient%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f71c7716510>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=1)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22AsyncMqttClient%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f71c7716590>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=2)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22AsyncMqttClient%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f71c7716850>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=3)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22AsyncMqttClient%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f71c7716d90>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=4)
[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22AsyncMqttClient%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f71c7982610>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=5)
Error: [API] Could not connect to PlatformIO API Service. Please try later.`

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: alexander_room
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: 'NSA'
  password: 'redacted'

mqtt:
  broker: '192.168.10.45'
  username: ''
  password: ''
# Example configuration entry
i2c:
  sda: 5
  scl: 4
  scan: False
# Example configuration entry
sensor:
  - platform: bme280
    temperature:
      name: "Alexanders Room Temp"
      oversampling: 8x
    pressure:
      name: "Alexanders Room Pressure"
    humidity:
      name: "Alexanders Room Humidity"
    address: 0x76
    update_interval: 60s
# Enable logging
logger:

ota:
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-05-07:
Probably a duplicate of https://github.com/esphome/issues/issues/296

More info here https://github.com/esphome/esphome/issues/40

paradoxbound, 2019-05-17:
Looks like we have the same problem.

brandond, 2019-05-17:
DNS issues. Seems to be most common with folks running Docker on Ubuntu, or with a piHole, or something like that.

paradoxbound, 2019-05-18:
Yes running the Pi-hole mod, I was running it for a while without issues, however, I will move the Pi-hole elsewhere and see if there is any change.

killerm1, 2019-05-18:
Im having the same issue..
only seems to have started since adding pi-hole addon. but even if I disable or stop pi-hole the issue still occurs. would removing it completely help? I have tried the work around in esphome/esphome#40 but makes no diff

HFsi, 2019-05-18:
You have to restart HA without the pi-hole add-on started....I think it have to do with the dns cache of HA...compile and then start the add on

killerm1, 2019-05-18:
I stoped pinhole and restarted HA via config /general /server mangement. But still didn't find api.platformio in esphome compile. Should I just completly remove pihole? It's a nice addition but my whole reason for HA is esphome.

HFsi, 2019-05-18:
You can try whitelisting platformio.org, too

killerm1, 2019-05-18:
Whitelist was the first thing i tried 😉

paradoxbound, 2019-05-19:
I removed Pi-hole and put it on its own instance, the problem went away. Whitelisting won't work as it is not blacklisted in the first place, can confirm this by checking the api from another instance also using the Pi-hole for DNS. I have no problem with non-localhost using Pi-hole so I am guessing that it is a Hassio issue, either core routing or the Pi-hole module.

HFsi, 2019-05-19:
I have all in the same hassio and the compilations works fine, the only different was whitelisting and restarting the first time without pihole....I don't know what the ideal solution is, but that worked for me

Wriman, 2019-05-29:
reinstall a new hassio, and install one addon, but a problem is same. :(

brandond, 2019-05-29:
@Wriman did you change anything on the pihole? That seems to be the cause of the issue.

Wriman, 2019-05-29:
> @Wriman did you change anything on the pihole? That seems to be the cause of the issue.

i dont have pihole… never have before

kenmaples, 2019-06-01:
I have been having this issue. pihole was showing the address resolving, but the failure continued. As soon as I removed cloudflare as one of my upstream servers in pihole, things started work. Adding cloudflare back causes the failure to return. Can anyone else confirm?

OttoWinter, 2019-06-01:
@kenmaples I've seen reports of some pages not resolving under cloudflare DNS, but a quick check shows that that's not the case here:

```txt
$ dig api.platformio.org @1.1.1.1

; <<>> DiG 9.10.6 <<>> api.platformio.org @1.1.1.1
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 22442
;; flags: qr rd ra; QUERY: 1, ANSWER: 2, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1452
;; QUESTION SECTION:
;api.platformio.org.		IN	A

;; ANSWER SECTION:
api.platformio.org.	1736	IN	CNAME	platformio.org.
platformio.org.		941	IN	A	193.222.52.25

;; Query time: 58 msec
;; SERVER: 1.1.1.1#53(1.1.1.1)
;; MSG SIZE  rcvd: 77
```

kenmaples, 2019-06-02:
I am seeing resolution also, further testing seems to show the issue only occurs when DNSSEC is enabled. I will try to setup a better test.

g14nn86, 2019-06-24:
hey, i have the same messages as aero83  i am running hassio. I have been searching around for a few days an i opened the above issue and i have read it all here but still haven't understand what should i do.  Have you found any solution for that?
Thanks in advance for your time.

aero83, 2019-06-24:
I am running pihole on a separate device.  This was a weird issue, because I could resolve the url from within the docker container, yet it would not work when attempting to compile.  Even when pi-hole was disabled from the admin interface.   All of this started with no changes to any of the settings or components of my network, router or pihole.

I ended up changing a few settings between my OPNsense router/firewall and the Pihole and it went back to working. Unfortunately I'm not sure exactly WHICH setting was the culprit.

slapecrade, 2019-06-30:
Had the same problem, running from hassos image and same issue after moving to debian. Finally solved it by unticking "Use DNSSEC" in Pihole.

This took me a month...

OttoWinter, 2019-07-11:
Closing because it's an external problem with DNS settings.

As a quick guide, these things appear to have helped:

 - Disable any DNS blockers like pi-hole adguard
 - Disable the DNSSEC setting in there

It's probably an issue with DNSSEC settings on platformio's end somewhere, but that's an external issue and should be handled in their bug trackers.

closed 2019-07-11 by OttoWinter