esphome/issues #303: Ultrasonic distance sensor issues in 1.12.2

teeling, 2019-05-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Running through docker, not through hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU 8266

**Affected component:**

Ultrasonic

**Description of problem:**
I'm having very similar to issue 147 (https://github.com/esphome/issues/issues/147) where my sensor only reports values up to 0.07cm.
The value also seem to shift a bit sporadically if I just leave it there for a while.
I also get a lot of "Distance measurement timed out!" in the log. These seem to occur more around updates of other sensors, but not only, so may be related or not.

I have tried the "branch: fix-ultrasonic", (which may have the same code for ultrasonic as the 1.12.2 ?), but any way the results are the same. I have also been experimenting with the update_inteval and timeout, but no noticeable difference in performance.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

esphome:
  name: nodemcutest
  platform: ESP8266
  board: nodemcuv2
#  esphome_core_version:
#    branch: fix-ultrasonic
    
wifi:
  ssid: "mywlan"
  password: "mysecretpass" 
  
  # Optional manual IP
  manual_ip:
    static_ip: 192.168.1.163
    gateway: 192.168.1.1
    subnet: 255.255.255.0

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:

ota:

binary_sensor:
  - platform: gpio
    name: "Pin D0"
    pin: D0
    
dallas:
  - pin: D2
  
# Individual sensors
sensor:
  - platform: dallas
    address: 0x8003079779B3BC28
    name: "Living Room Temperature"
    
  - platform: ultrasonic
    trigger_pin: D6
    echo_pin: D5
    name: "Ultrasonic Sensor"
    update_interval: 1s
    timeout: 2.0m
    
[... 117 lines cut ...]
[20:50:22][V][sensor.ultrasonic:032]: Echo took 306µs
[20:50:22][V][sensor.sensor:020]: 'Ultrasonic Sensor': Received new state 0.052479
[20:50:22][D][sensor.ultrasonic:040]: 'Ultrasonic Sensor' - Got distance: 0.05 m
[20:50:22][V][sensor.sensor:020]: 'Ultrasonic Sensor': Received new state 0.052479
[20:50:23][V][sensor.ultrasonic:032]: Echo took 308µs
[20:50:23][V][sensor.sensor:020]: 'Ultrasonic Sensor': Received new state 0.052822
[20:50:24][D][sensor.ultrasonic:040]: 'Ultrasonic Sensor' - Got distance: 0.05 m
[20:50:24][V][sensor.sensor:020]: 'Ultrasonic Sensor': Received new state 0.052822
[20:50:25][V][sensor.ultrasonic:032]: Echo took 308µs
[20:50:25][V][sensor.sensor:020]: 'Ultrasonic Sensor': Received new state 0.052822
[20:50:25][D][sensor.ultrasonic:040]: 'Ultrasonic Sensor' - Got distance: 0.05 m
[20:50:25][V][sensor.sensor:020]: 'Ultrasonic Sensor': Received new state 0.052822
[20:50:26][V][sensor.ultrasonic:032]: Echo took 308µs
[20:50:26][V][sensor.sensor:020]: 'Ultrasonic Sensor': Received new state 0.052822
[20:50:26][D][sensor.ultrasonic:040]: 'Ultrasonic Sensor' - Got distance: 0.05 m
[20:50:26][V][sensor.sensor:020]: 'Ultrasonic Sensor': Received new state 0.052822
[20:50:26][V][sensor.ultrasonic:032]: Echo took 0µs
[20:50:27][D][sensor.ultrasonic:035]: 'Ultrasonic Sensor' - Distance measurement timed out!
[20:50:27][V][sensor.sensor:020]: 'Ultrasonic Sensor': Received new state nan
[20:50:28][V][sensor.ultrasonic:032]: Echo took 0µs
```

**Additional information and things you've tried:**
I'm a bit new to electronics, so I'm not sure if it could be a hardware problem, but since I get reasonably accurate readings between 2 and 7 cm I assume the connections are about right. Please correct me if I'm mistaken in that statement. :-)
Should I get a few new sensor units and try? Have seen people complaining about the quality in these, so could be worth a shot.

mbo18, 2019-05-09:
What if you try:
`timeout: '2m'`
?

teeling, 2019-05-09:
Unfortunately, no difference that I can see.

OttoWinter, 2019-05-09:
@teeling Sounds like a hardware issue to me. Please add what model of ultrasonic sensor you're using. Some models need a higher `pulse_time`.

teeling, 2019-05-09:
@OttoWinter I have the HC-SR05, which I was told was compatible with the SR04 I see people use all the time. I'm running both trigger and echo through a 3.3v to 5v level converter, and I've also tried connecting the trigger straight from D6 to the SR05 trig pin.
You mention a higher pulse_time, is that a config option in ESPHome? Any suggestion for a value I can try? Thanks!

rradar, 2019-06-06:
@teeling just wired my first ultrasonic sensor and it just works! :rocket: 

Three things:

1. Get rid of the level shifter/converter if you are using a esp8266/esp32 (not needed as these one's a 5v tolerant on the logic level)
2. Catch the update to 1.13.x and try again
3. Report back that everything works and close this issue :wink:

OttoWinter, 2019-07-11:
Closing due to inactivity

closed 2019-07-11 by OttoWinter