esphome/issues #305: GPIO 14 wont work in esphome, in tasmota the gpio is working

marrobHD, 2019-05-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**Affected component:**

https://esphome.io/components/binary_sensor/gpio.html

**Description of problem:**

GPIO pin 14 wont work in ESPHome. In tasmota the pin is working fine. I have outdoor socket flashed over tuya convert with esphome. Now I want to use the build in button to control the relais. 
**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  plug_name: "outdoor1"

esphome:
  name: outdoorplug1
  platform: ESP8266
  board: esp8285

switch:
- platform: gpio
  name:  "${plug_name}_Relay1"
  id: "rel1"
  pin: GPIO12
  restore_mode: RESTORE_DEFAULT_OFF
- platform: gpio
  name:  "${plug_name}_Relay2"
  id: "${plug_name}_Relay2"
  pin: GPIO4
  restore_mode: RESTORE_DEFAULT_OFF
- platform: gpio
  name: "${plug_name}_LED_Blue"
  pin: GPIO5
  inverted: True
  restore_mode: ALWAYS_OFF
- platform: gpio
  name: "${plug_name}_LED_Red"
  pin: GPIO13
  inverted: True
  restore_mode: ALWAYS_OFF
- platform: gpio
  name: "${plug_name}_LED_Green"
  pin: GPIO16
  inverted: True
  restore_mode: ALWAYS_OFF
  
binary_sensor:
  - platform: gpio
    pin: 14
    name: "${plug_name}_button1"
    on_press:
      then:
        - switch.toggle: "rel1"
        - logger.log: "Hello World1"

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
I have tried to use pin 14 not gpio14

OttoWinter, 2019-05-11:
Please explain in more detail:

 - What is the behavior you see. Is the binary sensor not getting any events or something else?
 - What the HW looks like. How is the button wired up?

Also see the GPIO binary sensor page. You might need to set the pin to have a pullup and invert it. The docs explain how.

marrobHD, 2019-05-11:
The binary sensor won't turn on. There are no messages from the sensor in the console. That is my socket: https://s.click.aliexpress.com/e/XjlQMXm Why should a pullup be necessary. That was not necessary in tasmota

marrobHD, 2019-05-12:
Thanks! Now its working!
LEORY 10A Dual EU Plug Outdoor 
[My Outdoor Plug](https://de.aliexpress.com/item/LEORY-10A-Dual-EU-Plug-Outdoor-IP44-Waterproof-Smart-WIFI-Socket-Outlet-Support-Alexa-Google-Home/32997653700.html?spm=a2g0x.10010108.1000016.1.7b072587C0b1IE&isOrigTitle=true)

**MyESPHome Code:**
```yaml
substitutions:
  plug_name: "outdoor1"

esphome:
  name: outdoorplug1
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "yourwifissid"
  password: "yourwifipassword"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'yourapipassword'

ota:
  password: 'yourotapassword'

web_server:
  port: 80

debug:

time:
  - platform: homeassistant
    id: homeassistant_time

switch:
- platform: gpio
  name:  "${plug_name}_Relay1"
  id: "${plug_name}_Relay1"
  pin: GPIO12
  restore_mode: RESTORE_DEFAULT_OFF
  on_turn_on:
  - switch.turn_on: "${plug_name}_LED_Blue"
  on_turn_off:
  - switch.turn_off: "${plug_name}_LED_Blue"
- platform: gpio
  name:  "${plug_name}_Relay2"
  id: "${plug_name}_Relay2"
  pin: GPIO4
  restore_mode: RESTORE_DEFAULT_OFF
  on_turn_on:
  - switch.turn_on: "${plug_name}_LED_Green"
  on_turn_off:
  - switch.turn_off: "${plug_name}_LED_Green"
- platform: gpio
  name: "${plug_name}_LED_Blue"
  id: "${plug_name}_LED_Blue"
  pin: GPIO5
  inverted: True
  restore_mode: RESTORE_DEFAULT_OFF
- platform: gpio
  name: "${plug_name}_LED_Red"
  id: "${plug_name}_LED_Red"
  pin: GPIO13
  inverted: True
  restore_mode: RESTORE_DEFAULT_OFF
- platform: gpio
  name: "${plug_name}_LED_Green"
  id: "${plug_name}_LED_Green"
  pin: GPIO16
  inverted: True
  restore_mode: RESTORE_DEFAULT_OFF
- platform: shutdown
  name: "${plug_name} Shutdown"
- platform: restart
  name: "${plug_name} Restart"

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
      inverted: True
    name: "${plug_name}_Button"
    on_multi_click:
    - timing:
        - ON for at most 1s
        - OFF for at most 1s
        - ON for at most 1s
        - OFF for at least 0.2s
      then:
        - logger.log: "Double Clicked - turn all Relays on"
        - switch.toggle: "${plug_name}_Relay1"
        - switch.toggle: "${plug_name}_Relay2"
    - timing:
        - ON for 0.5s to 1s
        - OFF for at least 0.5s
      then:
        - logger.log: "Single Long Clicked - toggeling Relay2"
        - switch.toggle: "${plug_name}_Relay2"
    - timing:
        - ON for at most 0.5s
        - OFF for at least 0.5s
      then:
        - logger.log: "Single Short Clicked - toggeling Relay1"
        - switch.toggle: "${plug_name}_Relay1"
sensor:
# Extra sensor to keep track of plug uptime
  - platform: uptime
    name: ${plug_name}_Uptime Sensor
    
status_led:
  pin:
    number: GPIO13
    inverted: True
```

closed 2019-05-12 by marrobHD