esphome/issues #306: Problem with servo component in dev branch

norges, 2019-05-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

dev version  checked out in a virtualenv on a raspberry pi 3+  (and on another machine in a docker container ubuntu on windows) 
 

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266  nodemcu board powered over usb with one  little sg90 servo.

**Affected component:**

https://esphome.io/components/servo.html
or
https://esphome.io/components/output/esp8266_pwm.html

**Description of problem:**
In the dev branch there is a Fatal exception when I try to use a servo at the end of the setup of the servo component.

I know that a dev version is not something to use productive but I encountered it when I tried to create something new and thought I better mention it, if it is known because of work in progress you can close it immediately.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: servo2
  platform: ESP8266
  board: nodemcuv2

# Enable logging
logger:
  level: VERY_VERBOSE
  #level: VERBOSE
debug:

servo:
  - id: my_servo
    output: pwm_output

# Example output platform
output:
  - platform: esp8266_pwm
    id: pwm_output
    pin: D1
    frequency: 50 Hz
```

**Traceback (if applicable):**

When compiled with the dev version I got the following output on the usb log
```
[16:39:19][I][app:027]: Running through setup()...
[16:39:19][V][app:028]: Sorting components by setup priority...
[16:39:19][C][something.something:017]: Setting up ESP8266 PWM Output...
[16:39:19][I][app:059]: setup() finished successfully!
[16:39:20][I][app:063]: esphome-core version 1.13.0-dev compiled on May 11 2019, 16:37:57
[16:39:20][C][something.something:022]: ESP8266 PWM:
[16:39:20][C][something.something:023]:   Pin: GPIO5 (Mode: OUTPUT)
[16:39:20][C][something.something:024]:   Frequency: 50.0 Hz
[16:39:20][C][logger:137]: Logger:
[16:39:20][C][logger:138]:   Level: VERY_VERBOSE
[16:39:20][C][logger:139]:   Log Baud Rate: 115200
[16:39:20][C][logger:140]:   Hardware UART: UART0
[16:39:20][C][servo:012]: Servo:
[16:39:20][C][servo:013]:   Idle Level: 7.5%
[16:39:20][C][servo:014]:   Min Level: 3.0%
[16:39:20][C][servo:015]:   Max Level: 12.0%
[16:39:20][D][debug:022]: ESPHome Core version 1.13.0-dev
[16:39:20][D][debug:024]: Free Heap Size: 49000 bytes
[16:39:20][D][debug:052]: Flash Chip: Size=4096kB Speed=40MHz Mode=DOUT
[16:39:20][D][debug:189]: Chip ID: 0x00185235
[16:39:20][D][debug:190]: SDK Version: 2.2.1(cfd48f3)
[16:39:20][D][debug:191]: Core Version: 2_4_2
[16:39:20][D][debug:192]: Boot Version=31 Mode=1
[16:39:20][D][debug:193]: CPU Frequency: 80
[16:39:20][D][debug:194]: Flash Chip ID=0x001640EF
[16:39:20][D][debug:195]: Reset Reason: External System
[16:39:20][D][debug:196]: Reset Info: Fatal exception:0 flag:6 (EXT_SYS_RST) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000

```
When compile the above source  with the version 12.2  I got the following output on the usb log:
```
[16:45:29][I][logger:119]: Log initialized
[16:45:29][I][application:055]: Running through setup()...
[16:45:29][V][application:056]: Sorting components by setup priority...
[16:45:29][C][output.esp8266_pwm:024]: Setting up ESP8266 PWM Output...
[16:45:29][D][debug:025]: ESPHome Core version 1.12.2
[16:45:29][D][debug:027]: Free Heap Size: 48064 bytes
[16:45:29][D][debug:055]: Flash Chip: Size=4096kB Speed=40MHz Mode=DOUT
[16:45:29][D][debug:192]: Chip ID: 0x00185235
[16:45:29][D][debug:193]: SDK Version: 2.2.1(cfd48f3)
[16:45:29][D][debug:194]: Core Version: 2_4_2
[16:45:29][D][debug:195]: Boot Version=31 Mode=1
[16:45:29][D][debug:196]: CPU Frequency: 80
[16:45:29][D][debug:197]: Flash Chip ID=0x001640EF
[16:45:29][D][debug:198]: Reset Reason: External System
[16:45:29][D][debug:199]: Reset Info: Fatal exception:0 flag:6 (EXT_SYS_RST) epc1:0x00000000 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
[16:45:29][I][application:089]: setup() finished successfully!
[16:45:29][I][application:097]: esphome-core version 1.12.2 compiled on May 11 2019, 16:23:31
[16:45:29][C][output.esp8266_pwm:029]: ESP8266 PWM:
[16:45:29][C][output.esp8266_pwm:030]:   Pin: GPIO5 (Mode: OUTPUT)
[16:45:29][C][output.esp8266_pwm:031]:   Frequency: 50.0 Hz
[16:45:29][C][logger:142]: Logger:
[16:45:29][C][logger:143]:   Level: VERY_VERBOSE
[16:45:29][C][logger:144]:   Log Baud Rate: 115200
[16:45:29][C][logger:145]:   Hardware UART: UART0
[16:45:29][C][servo:028]: Servo:
[16:45:29][C][servo:029]:   Idle Level: 7.5%
[16:45:29][C][servo:030]:   Min Level: 3.0%
[16:45:29][C][servo:031]:   Max Level: 12.0%
[16:45:29][I][application:109]: Running through first loop()
[16:45:29][I][application:136]: First loop finished successfully!
[16:45:29]wifi evt: 7
[16:45:32]wifi evt: 7
```

**Additional information and things you've tried:**
I tried also a different ESP 8266 board with the same result.

norges, 2019-05-11:
Sorry that I don't had thought about minimizing it directly to the PWM component.

OttoWinter, 2019-07-11:
Closing as fixed since the creation of issue

closed 2019-07-11 by OttoWinter