esphome/issues #307: Set WaveshareEPaper to sleep to prevent hardware damage

grote, 2019-05-11:
**Affected component:**

`esphome::display::WaveshareEPaper`

**Description of problem:**

If I understand the code of [`src/esphome/display/waveshare_epaper.cpp`](https://github.com/esphome/esphome-core/blob/dev/src/esphome/display/waveshare_epaper.cpp) correctly, it does *not* put the display to sleep between refreshs.

[In their wiki](https://www.waveshare.com/wiki/4.2inch_e-Paper_Module_(B)#FAQ), the manufacturer of the waveshare e-paper panels says this:

> Please set the e-paper to sleep mode or disconnect it if you needn't refresh the e-paper but need to power on your development board or Raspberry Pi for long time.Otherwise, the voltage of panel keeps high and it will damage the panel 

or in another place:

> Please set the e-Paper to sleep mode or power off it directly, otherwise, the e-Paper is damaged because of working in high voltage for long time. 

So, please correct me if I am mistaken, the WaveshareEPaper code should put the display to sleep to prevent damaging people's display panels over prolonged usage.

OttoWinter, 2019-05-12:
At least my epaper display has been working fine for a long time now (months) without any noticable change in the display - so even if that is true it can't be that bad.

Also, from the way this technology operates, I don't think anything can happen to the panel itself when it is not written to. If there were still a voltage potential after writing, you would see the panel slowly fade to black or white (which it doesn't do). Maybe they're talking more about the voltage regulator circuitry.

Next, as the wiki says: "for long time." - Most people will have the update_interval set at a few min max or so. That doesn't mean a long time to me. Plus each time the display needs to be fully re-initialized after a sleep.

Unless some epaper expert chimes in or there is better documentation (for example a datasheet) explaining this in more detail, I don't plan to act on it.

grote, 2019-05-12:
Thanks for leaving this issue open!

I searched [the datasheet (PDF)](https://www.waveshare.com/w/upload/7/7f/4.2inch-e-paper-b-specification.pdf) and couldn't find what they mention also in their manual about not letting the display on high voltage or outside deep sleep for long.

However, in the datasheet, there's this diagram on page 35:
[image]

If this is the typical operating sequence and deep sleep or turn off is at the end of it, maybe this is how it should be implemented?

How often do you refresh your display? My plan was to refresh it once a minute (since it also shows the time), but do not refresh the screen when nobody is home (or online) which happens at least ~8h every night and potentially more hours during the day.

closed 2019-05-28 by OttoWinter