esphome/issues #308: [MQTT]  Homeassistant discovery causes permanent loop of publish failures

thiemok, 2019-05-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Device: Weemos D1 mini
esphome version: 1.12.2
MQTT: mosquitto 1.6.2 running on k3s on a rpi3

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, Board/Weemos D1 mini

**Affected component:**

MQTT

**Description of problem:**
I added a bunch of sensors for telemetry to my device which caused mqtt to go into an infinite loop of publish failures. The device works fine without homeassistant discovery or with less sensors added.
From some other issues regarding mqtt publish falilures I learned that this is probably caused by the send buffer filling up. My issues is that homesassistant discovery happens so frequently that the buffer never has the chance to clear before the next discovery period. For reference, my sensor state published happen way less frequently than the discovery itself.

**Problem-relevant YAML-configuration entries:**
This is the failing config, I have since then removed the telemetry sensors to get the device working again
(Config pasted from multiple files)
```yaml
substitutions:
  device_name_lower: room_name
  device_name_upper: "Room Name"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_psk
  fast_connect: true
  domain: !secret wifi_domain

# Enable logging
logger:
  level: WARN

mqtt:
  broker: !secret mqtt_broker
  discovery: True
  port: !secret mqtt_port

ota:
  password: !secret ota_passwd

text_sensor:
  - platform: version
    name: ESPHome Version
  - platform: wifi_info
    ip_address:
      name: ${device_name_lower} ESP IP Address
    ssid:
      name: ${device_name_lower} ESP Connected SSID
    bssid:
      name: ${device_name_lower} ESP Connected BSSID

esphome:
  name: ${device_name_lower}_climate
  platform: ESP8266
  board: d1_mini

i2c:
  sda: D2
  scl: D3

sensor:
  - platform: uptime
    name: ${device_name_lower}_climate Uptime Sensor
  - platform: adc
    pin: VCC
    name: ${device_name_lower}_climate VCC Voltage
  - platform: dht
    model: DHT11
    pin: D0
    temperature:
      name: "${device_name_upper} Temperature"
    humidity:
      name: "${device_name_upper} Humidity"
    update_interval: 120s
  - platform: tsl2561
    name: "${device_name_upper} Ambient Light"
    address: 0x39
    update_interval: 120s

```

**Additional information and things you've tried:**
I think this could be avoided by adding a config option to change the discovery interval.

OttoWinter, 2019-05-12:
> My issues is that homesassistant discovery happens so frequently that the buffer never has the chance to clear before the next discovery period.

Discovery data is only sent after the device connects - it is not sent at any intervals or anything like that.

Your broker may however be closing the connection if it receives too much data - please check the broker logs as to what is causing disconnects (if any)

thiemok, 2019-05-12:
I will check that later, thanks for the info.

For further discussion: I was sseing a constant stream of discovery messages on when subscribing to any topic on the broker on my pc

OttoWinter, 2019-07-11:
Closing due to inactivity

closed 2019-07-11 by OttoWinter