esphome/issues #309: [Shelly1] After boot switch always turns "ON"

peterforeman, 2019-05-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker 18.09.5, ESPhome 1.12.2

**ESP (ESP32/ESP8266, Board/Sonoff):**

Shelly 1 (ESP8266)

**Affected component:**

Switch component(?)

**Description of problem:**
I  have multiple Shelly products at home (Shelly1 / 2 / 2.5). This specific shelly always turns the relay ON after boot. This is not what I want: after boot (power loss or whatever) it should stay off.

**Problem-relevant YAML-configuration entries:**
```yaml
mqtt:
  broker: xxx
  username: xxx
  password: xxx
  discovery: False
  topic_prefix: xxx

binary_sensor:
- platform: gpio
  pin:
    number: GPIO5
  name: "Button"
  on_press:
    - switch.toggle: relay
    
- platform: status
  name: "Status"

switch:
- platform: gpio
  name: "Relay"
  pin: GPIO4
  id: relay
  restore_mode: ALWAYS_OFF
```

**Traceback (if applicable):**

```
[08:13:38][I][mqtt.client:213]: MQTT Connected!
[08:13:39][I][application:089]: setup() finished successfully!
[08:13:39][I][application:097]: esphome-core version 1.12.2 compiled on May 13 2019, 08:13:18
[08:13:39][C][wifi:371]: WiFi:
[08:13:39][C][wifi:253]:   SSID: 'xxx'[redacted]
[08:13:39][C][wifi:254]:   IP Address: x
[08:13:39][C][wifi:256]:   BSSID: xxx[redacted]
[08:13:39][C][wifi:257]:   Hostname: 'xxx'
[08:13:39][C][wifi:261]:   Signal strength: -68 dB ▂▄▆█
[08:13:39][C][wifi:262]:   Channel: 11
[08:13:39][C][wifi:263]:   Subnet: 255.255.255.0
[08:13:39][C][wifi:264]:   Gateway: xxx
[08:13:39][C][wifi:265]:   DNS1: xxx
[08:13:39][C][wifi:266]:   DNS2: xxx
[08:13:39][C][binary_sensor.status:035]: Status Binary Sensor 'Status'
[08:13:39][C][binary_sensor.status:035]:   Device Class: 'connectivity'
[08:13:39][C][binary_sensor.gpio:022]: GPIO Binary Sensor 'Button'
[08:13:39][C][binary_sensor.gpio:023]:   Pin: GPIO5 (Mode: INPUT)
[08:13:39][C][switch.gpio:049]: GPIO Switch 'Relay'
[08:13:39][C][switch.gpio:050]:   Pin: GPIO4 (Mode: OUTPUT)
[08:13:39][C][switch.gpio:066]:   Restore Mode: Restore (Defaults to OFF)
[08:13:39][C][logger:142]: Logger:
[08:13:39][C][logger:143]:   Level: DEBUG
[08:13:39][C][logger:144]:   Log Baud Rate: 115200
[08:13:39][C][logger:145]:   Hardware UART: UART0
[08:13:39][C][web_server:126]: Web Server:
[08:13:39][C][web_server:127]:   Address: xxx:80
[08:13:39][C][ota:127]: Over-The-Air Updates:
[08:13:39][C][ota:128]:   Address: xxx:8266
[08:13:39][C][mqtt.client:062]: MQTT:
[08:13:39][C][mqtt.client:064]:   Server Address: xxx:1883 (xxx)
[08:13:39][C][mqtt.client:065]:   Username: 'xxx'[redacted]
[08:13:39][C][mqtt.client:066]:   Client ID: 'xxx'[redacted]
[08:13:39][C][mqtt.client:071]:   Topic Prefix: 'xxx'
[08:13:39][C][mqtt.client:073]:   Log Topic: 'xxx/debug'
[08:13:39][C][mqtt.client:076]:   Availability: 'xxx/status'
[08:13:39][C][binary_sensor.mqtt:021]: MQTT Binary Sensor 'Status':
[08:13:39][C][binary_sensor.mqtt:022]:   State Topic: 'xxx/status'
[08:13:39][C][binary_sensor.mqtt:021]: MQTT Binary Sensor 'Button':
[08:13:39][C][binary_sensor.mqtt:022]:   State Topic: 'xxx/binary_sensor/button/state'
[08:13:39][C][switch.mqtt:042]: MQTT Switch 'Relay': 
[08:13:39][C][switch.mqtt:043]:   State Topic: 'xxx/switch/relay/state'
[08:13:39][C][switch.mqtt:043]:   Command Topic: 'xxx/switch/relay/command'
[08:13:39][I][application:109]: Running through first loop()
[08:13:39][D][binary_sensor:037]: 'Status': Sending state ON
[08:13:39][D][switch:027]: 'Relay' Turning ON.
[08:13:39][D][switch:051]: 'Relay': Sending state ON
[08:13:39][I][application:136]: First loop finished successfully!
```

**Additional information and things you've tried:**
I've tried commenting out the binary_sensor and status sections so that it should not be something with the switch which is pressed/inverted. I checked MQTT for retained messages (not there). Added "restore_mode" which also does not work. Tried "on_boot" with switch.turn_off at different priorities. This does turn off, but again it turns on almost immediately after.

OttoWinter, 2019-05-13:
> 'Relay' Turning ON.

Something is turning the relay on *after* `setup()`, which probably means there's something manually turning it on.

Could you double check for retained messages? Or alternatively, is the traceback you added really the one corresponding to the config (`Restore Mode: Restore (Defaults to OFF)` doesn't correspond with config)?

peterforeman, 2019-05-13:
The traceback was from before the Restore Mode setting. I've sent an empty retained payload to the command topic and reset the ESP again. I do not longer see the "Relay Turning ON" message and the relay stays off (even without Restore Mode setting removed). Though I am pretty sure there was no retained message (double checked with MQTT Explorer and mosquitto_sub), but it seems to do what is expected.

I'll close this, will monitor it and re-open if the problem re-appears.

Thanks for your crazy fast response! 👍

closed 2019-05-13 by peterforeman