esphome/issues #310: Fastled no yellow color in last part of strip

pszafer, 2019-05-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
HA 0.92 PIP, ESPHome 1.12.2

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32

**Affected component:**
fastled_clockless

**Description of problem:**
I have WS2812 led strip light divided into partitions and one main light with all leds.
I cannot set yellow color for main light for all leds. Only half getting yellow color.
What's weird when setting partition light `led201300` yellow color it is working.
When using effect it is working too.

Only setting yellow color to `fast_led` is not working.

**Problem-relevant YAML-configuration entries:**
```yaml
light:
  - platform: fastled_clockless
    chipset: WS2812
    pin: GPIO23
    num_leds: 300
    rgb_order: GRB
    id: fast_led
    name: 'Tasma LED'
    effects:
      - addressable_rainbow:
      - addressable_color_wipe:
      - flicker:
      - strobe:
      - addressable_scan:
      - addressable_twinkle:
          name: Twinkle Effect With Custom Values
          twinkle_probability: 5%
          progress_interval: 4ms
      - addressable_random_twinkle:
      - addressable_fireworks:
  - platform: partition
    name: "LED0-50"
    id: led050
    segments:
      - id: 'fast_led'
        from: 0
        to: 70
    effects:
      - addressable_rainbow:
      - addressable_color_wipe:
      - flicker:
 - platform: partition
    name: "LED51-100"
    id: led51100
    segments:
      - id: 'fast_led'
        from: 71
        to: 200
    effects:
      - addressable_rainbow:
      - addressable_color_wipe:
      - flicker:
      - strobe:
      - addressable_scan:
      - addressable_twinkle:
          name: Twinkle Effect With Custom Values
          twinkle_probability: 5%
          progress_interval: 4ms
      - addressable_random_twinkle:
      - addressable_fireworks:
  - platform: partition
    name: "LED200-300"
    id: led201300
    segments:
      - id: 'fast_led'
        from: 201
        to: 300
    effects:
      - addressable_rainbow:
      - addressable_color_wipe:
      - flicker:
      - strobe:
      - addressable_scan:
      - addressable_twinkle:
          name: Twinkle Effect With Custom Values
          twinkle_probability: 5%
          progress_interval: 4ms
      - addressable_random_twinkle:
      - addressable_fireworks:

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

I changed GPIO, no change.

OttoWinter, 2019-05-13:
That sounds interesting lol.

> I cannot set yellow color for main light for all leds.

Is this via HA frontend? Also is this MQTT or native API?

pszafer, 2019-05-13:
Native API. Do you want me to check MQTT?

OttoWinter, 2019-05-13:
I doubt the native API is it - just wanted to know which one is used here.

Some logs could be useful. Also, if you have an effect active on one of the light partitions that will override the state from the fastled base. So also try turning the partitions off before and/or removing them from the config.

pszafer, 2019-05-13:
Deleting partitions didn't help.
Same problems occurs with such config:
```
esphome:
  name: tomatoes
  platform: ESP32
  board: esp32doit-devkit-v1

wifi:
  ssid: 'mySSID'
  password: 'myPASS'
  domain: .my.domain

# Enable logging
logger:
  level: VERBOSE
api:
ota:
web_server:
  port: 80
sensor:
  - platform: adc
    pin: GPIO32
    name: "moisture"
    update_interval: 10s
    attenuation: 11db
    accuracy_decimals: 0
    unit_of_measurement: '%'
    filters:
      - lambda: return map(x*100, 110, 335, 100, 0);
light:
  - platform: fastled_clockless
    chipset: WS2812
    pin: GPIO22
    num_leds: 300
    rgb_order: GRB
    id: fast_led
    name: 'Tasma LED'
    effects:
      - addressable_rainbow:
      - addressable_color_wipe:
      - flicker:
      - strobe:
      - addressable_scan:
      - addressable_twinkle:
          name: Twinkle Effect With Custom Values
          twinkle_probability: 5%
          progress_interval: 4ms
      - addressable_random_twinkle:
      - addressable_fireworks:
```

This is log for yellow light:
```
[15:19:01][D][light.state:418]: 'Tasma LED' Turning ON
[15:19:01][D][light.state:449]:   Red=100%, Green=100%, Blue=0%
[15:19:01][D][light.state:459]:   Transition Length: 1000 ms
```

When reverted to partitions I got similar log for it:
```
[15:22:03][D][light.state:418]: 'LED200-300' Turning ON
[15:22:03][D][light.state:449]:   Red=100%, Green=100%, Blue=0%
[15:22:03][D][light.state:459]:   Transition Length: 1000 ms
```

I lost connection to ESP32 if I set log level: VERY_VERBOSE :/

pszafer, 2019-05-13:
Ok. Got VERY_VERBOSE.

Non-working yellow:
```
[15:34:39][D][light.state:418]: 'LED200-300' Turning ON
[15:34:39][D][light.state:449]:   Red=100%, Green=100%, Blue=0%
[15:34:39][D][light.state:459]:   Transition Length: 1000 ms
[15:34:39][VV][preferences:049]: SAVE 4: 0=0x00000001 1=0x3F800000 (Type=3303562514, CRC=0xBA33E1CE)
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:39][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:40][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:40][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:40][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:40][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:40][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:40][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:40][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:40][VV][light.fast_led:041]: Writing RGB values to bus...
[15:34:40][VV][light.fast_led:041]: Writing RGB values to bus..
```

Working yellow:
```
[15:33:20][D][light.state:418]: 'Tasma LED' Turning ON
[15:33:20][D][light.state:449]:   Red=100%, Green=100%, Blue=0%
[15:33:20][D][light.state:459]:   Transition Length: 1000 ms
[15:33:20][VV][preferences:049]: SAVE 1: 0=0x00000001 1=0x3F800000 (Type=2993874927, CRC=0xCCA95B33)
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:20][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:21][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:21][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:21][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:21][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:21][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:21][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:21][VV][light.fast_led:041]: Writing RGB values to bus...
[15:33:21][VV][light.fast_led:041]: Writing RGB values to bus..
```

pszafer, 2019-05-13:
Ok, more mess.
I tested only idea I had, change number of leds configured.

| num_leds | yellow color                                     |
|----------|--------------------------------------------------|
| 60       | all OK                                           |
| 100      | all OK                                           |
| 150      | 20 last leds are little bit ORANGE but still OK  |
| 200      | around 50 last leds are little bit ORANGE but OK |
| 250      | 70 last leds are RED                             |
| 300      | 120 last leds are RED                            |

With setting color for partition Yellow is always OK.
I could partition lights and create parition of partitions (sum of partitions) and test if that is gonna work.

I can also test Tasmota if it can help you.

OttoWinter, 2019-05-13:
Ok then that sounds like a power supply issue to me.

pszafer, 2019-05-14:
Ordered new one. I'll get back to you when I check the new one.

OttoWinter, 2019-05-14:
@pszafer Well you could also check the VCC voltage at the points of the strip where you see the error - a new strip itself probably won't fix the issue; if it is what I think it is, then only new wiring or a better PSU will help.

pszafer, 2019-05-14:
Misunderstanding, I ordered new PSU, as I'm gonna need it anyway :)

pszafer, 2019-05-16:
@OttoWinter unfortunately. New PSU 120W, nothing changed.
I recorder about last 100 led's so I can show you - https://youtu.be/z3R-HkAC3uk
With more powerful PSU I can see transtition after turn on from yellow to red...
Maybe you send color signal for too long time or something like that...?

OttoWinter, 2019-05-16:
Puh okay then this is going to be complicated issue.

I don't really understand why this color exactly would not work - the LED just receives R,G,B signals, so any signal that uses the three should in theory work.

Also the partition light directly accesses the RGB output of the base LED. Plus the setting color code is shared between them so there should also be no difference there. So I don't understand why this would not work.

Maybe it's some issue in the fastled library; I doubt it but that could theoretically be it. 

> Maybe you send color signal for too long time or something like that...?

All communication with the LED strip is handled by the fastled library in this case.
Another option would be to test with neopixelbus integration (I believe that one also supports this light strip).

pszafer, 2019-05-16:
With neopixelbus same effect... I'm gonna try to reconnect led strip with another resistor.

pszafer, 2019-05-18:
The more I'm testing, less I'm understand... maybe wrong connection.

I change config to something like this:
```
light:
  - platform: fastled_clockless
    chipset: WS2812
    pin: GPIO22
    num_leds: 300
    rgb_order: GRB
    id: fast_led
    name: 'Tasma LED'
    effects:
      - addressable_rainbow:
      - addressable_color_wipe:
      - flicker:
      - strobe:
      - addressable_scan:
      - addressable_twinkle:
          name: Twinkle Effect With Custom Values
          twinkle_probability: 5%
          progress_interval: 4ms
      - addressable_random_twinkle:
      - addressable_fireworks:
  - platform: partition
    name: "LED0-50"
    id: led050
    segments:
      - id: 'fast_led'
        from: 0
        to: 70
    effects:
      - addressable_rainbow:
      - addressable_color_wipe:
      - flicker:
      - strobe:
      - addressable_scan:
      - addressable_twinkle:
          name: Twinkle Effect With Custom Values
          twinkle_probability: 5% 
          progress_interval: 4ms
      - addressable_random_twinkle:
      - addressable_fireworks:
  - platform: partition
    name: "LED_PROJEKTOR"
    id: ledprojektor
    segments:
      - id: 'fast_led'
        from: 71
        to: 300
    effects:
      - addressable_rainbow:
  - platform: partition
    name: "LED_ALL"
    id: ledall
    segments:
      - id: 'fast_led'
        from: 3
        to: 300
    effects:
      - addressable_rainbow:
```
New results:

| led_id | state                                     |
|----------|--------------------------------------------------|
| fast_led       | all OK (yellow too!!!)                                           |
| led050      | all OK                                           |
| ledprojektor      | last 120 leds are red  |
| ledall      | when turning it on ESP hangs (have to hard reboot it) and led strip got full white color |

So with this config, partitions are broken. Don't understand it at all.
I have resistor installed between GPIO22 and DIN of LED strip. Maybe this is wrong?

BWT would be possible to add some logic that if I turn off `light.fast_led` in HA then all of sub_partitions would be turned off as well?

OttoWinter, 2019-05-27:
> BWT would be possible to add some logic that if I turn off light.fast_led in HA then all of sub_partitions would be turned off as well?

No, and when you use partitions you shouldn't access the parent light at all - for that you can use light groups in HA.

As for the other stuff, I don't know what this is. I don't know how to help with that.

pszafer, 2019-06-25:
With not accessing parent light it is working very good. I will close it as using group in HA is acceptable workaround.

closed 2019-06-25 by pszafer