esphome/issues #312: Homeassistant ESPHome integration always resolves empty

yeralin, 2019-05-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Raspbian GNU/Linux 9 (stretch)
(Raspberry Pi 3B)

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266 (NODEMCUV2 by espressif)

**Affected component:**
API (https://esphome.io/components/api.html)
Homeassistant Integration

**Description of problem:**
I made a simple pet feeder ran by ESP8266 hooked with servo. Everything appears to be working fine. I initially integrated it with Homeassistant (ran on my raspberry pi) successfully, and I'd been using it just fine for about a month. Then, I added a new sensor to my pet feeder to show on Homeassistant dashboard. Uploaded it via Wi-Fi on my board, but it did not appear on the dashboard, so I deleted the integration, and re-added it back again. However, when I tried to add integration back, it found my ESP8266, but does not resolve any exposed metrics (sensors) and api calls:
[image]
As you can see, it found my smart_pet_feeder, but did not resolve any api "endpoints" from it. And first I thought it is a metadata issue, something got cached, and I just need to clean up homeassistant configurations folder. Tried to find anything that's related to esphome/smart_pet_feeder using grep, deleted those files, and restarted Homeassistant. The issue persisted. Then, I reinstalled my homeassistant by deleting previous virtualenv, creating a new one, and install homeassistant using pip. After successful reinstallation, I tried to integrated smart_pet_feeder, but once again it resolved empty. Then, I backed up and deleted homeassistant's configurations directory, reinstalled homeassistant, and started all fresh. **And still when I tried to do integration for my feeder, it resolved empty**
Next step, I pulled a homeassistant docker image on my laptop:
`docker run -p 8123:8123 homeassistant/home-assistant`, waited for it to fully initialize, went to the main page, tried to integrate my smart_pet_feeder with this fresh instance, and **it worked!**
[image]

I believe that some metadata is persisted outside of virtualenv environment, and outside of `~/.homeassistant` configuration's directory that prevents me to successfully re-integrate my esphome instance with my homeassistant. Or maybe there is some deeper issue.

**Problem-relevant YAML-configuration entries:**
```yaml
...
api:
  services:
    # Update weight and time period of the feeder
    - service: change_target_weight
      variables:
        new_target_weight: int
      then:
        - lambda: |-
            id(target_weight) = new_target_weight;
    - service: change_feed_interval
      variables:
        new_feed_interval: int
      then:
        - lambda: |-
            id(feeder_interval) = new_feed_interval;
...
```
The yml file is quite large, you can access it here: https://github.com/yeralin/IoT/blob/master/applications/smart-feeder/smart-feeder.yml

**Traceback (if applicable):**
Does not apply

**Additional information and things you've tried:**
Everything I tried is explained above.

yeralin, 2019-05-15:
@OttoWinter I'm really sorry to bother. Could you _please_ take a look? I've tried **everything**, the last thing left is to reinstall Raspbian 😅

OttoWinter, 2019-05-16:
The list of entities is re-fetched on every HA connect to the device - metadata should not be the issue here since that is only used for restoring stuff but the new values always override that.

Potential Steps:

 1. Remove all ESPHome devices in integration screen (with delete icon in upper right corner)
 2. Re-add them one-by-one. If they're not visible straight away, try reloading the page in the browser; there was an issue in HA's frontend some time ago that prevented newly added entities from being shown without a refresh.
 3. If still experiencing issues, please post your HA error logs (as in the issue template)

yeralin, 2019-05-16:
**Traceback (from Raspberry PI's hass):**
```
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.setup] Setting up esphome
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.setup] Setup of domain esphome took 0.0 seconds.
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [aioesphomeapi.connection] smart_pet_feeder.local: Error while reading incoming messages: Error while receiving data: 0 bytes read on a total of 1 expected bytes
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.esphome
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.components.switch] Setting up switch.esphome
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (Thread-20) [homeassistant.loader] Loaded binary_sensor from homeassistant.components.binary_sensor
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (Thread-16) [homeassistant.loader] Loaded climate from homeassistant.components.climate
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (Thread-11) [homeassistant.loader] Loaded light from homeassistant.components.light
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (Thread-14) [homeassistant.loader] Loaded cover from homeassistant.components.cover
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (Thread-18) [homeassistant.loader] Loaded camera from homeassistant.components.camera
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (Thread-5) [homeassistant.loader] Loaded fan from homeassistant.components.fan
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.setup] Setting up climate
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.setup] Setup of domain climate took 0.0 seconds.
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.setup] Setting up binary_sensor
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.setup] Setup of domain binary_sensor took 0.0 seconds.
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.components.climate] Setting up climate.esphome
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.components.binary_sensor] Setting up binary_sensor.esphome
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.setup] Setting up light
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.setup] Setting up cover
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.setup] Setup of domain cover took 0.0 seconds.
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.setup] Setting up camera
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.setup] Setting up fan
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.setup] Setup of domain fan took 0.0 seconds.
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.components.cover] Setting up cover.esphome
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.components.fan] Setting up fan.esphome
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.setup] Setup of domain light took 0.0 seconds.
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.components.light] Setting up light.esphome
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.setup] Setup of domain camera took 0.0 seconds.
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.components.camera] Setting up camera.esphome
May 16 16:21:48 raspberrypi hass[18950]: 2019-05-16 16:21:48 INFO (MainThread) [homeassistant.components.esphome] Successfully connected to smart_pet_feeder.local
May 16 16:21:56 raspberrypi hass[18950]: 2019-05-16 16:21:55 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
May 16 16:21:56 raspberrypi hass[18950]: 2019-05-16 16:21:56 INFO (MainThread) [aioesphomeapi.connection] smart_pet_feeder.local: Error while reading incoming messages: Error while receiving data: 0 bytes read on a total of 1 expected bytes
May 16 16:21:56 raspberrypi hass[18950]: 2019-05-16 16:21:56 INFO (MainThread) [homeassistant.components.esphome] Disconnected from ESPHome API for smart_pet_feeder.local
May 16 16:21:56 raspberrypi hass[18950]: 2019-05-16 16:21:56 INFO (MainThread) [homeassistant.components.esphome] Successfully connected to smart_pet_feeder.local
May 16 16:22:01 raspberrypi hass[18950]: 2019-05-16 16:22:01 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
May 16 16:22:01 raspberrypi hass[18950]: 2019-05-16 16:22:01 INFO (MainThread) [aioesphomeapi.connection] smart_pet_feeder.local: Error while reading incoming messages: Error while receiving data: 0 bytes read on a total of 1 expected  [... cut ...]
2019-05-16 15:17:14 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.esphome entity: sensor.weight_sensor
2019-05-16 15:17:14 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.esphome entity: sensor.current_hours_since_start
2019-05-16 15:17:14 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.esphome entity: sensor.pet_feeder_interval
2019-05-16 15:17:14 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.esphome entity: sensor.pet_feeder_target_weight
```

yeralin, 2019-05-16:
as you can see both Hass instances are of the same version, but the main difference is what's happening at the end.

On my raspberry pi, it complains about a timeout:
`Error getting initial data: Timeout while waiting for API response!`
However, everything preceding it was fine and exactly the same as in Docker's hass.

In Docker's hass logs, we can see that at the very end after log: 
`Successfully connected to smart_pet_feeder.local`
It starts registering sensors just fine i.e. not getting that Timeout

yeralin, 2019-05-20:
Sorry for bothering @OttoWinter 
Any idea what might be causing it? Worst case I’ll just reinstall OS

yeralin, 2019-05-22:
Nah I gave up. Just started using Docker

yeralin, 2019-05-23:
I tried using Docker on my raspberry. It seems like a hardware issue since docker didn't help.
To clarify, when I installed HA using Docker on my PC, ESP connected just fine.
HA and Docker HA on Raspberry PI does not work (I seriously considering just reinstall OS)

Also it seems related to #17

yeralin, 2019-05-23:
More news, after I switched to Wi-Fi from Ethernet on my raspberry pi, the issue went away!

I'll play around with changing ethernet's interface mac address and see if it helps

yeralin, 2019-05-23:
Nah, spoofing mac did not help. What I ended up doing:
1) Turn on my Wi-Fi module
2) Disable ethernet
3) Start Hass
4) Access Hass using Wi-Fi assigned address
5) Add ESPHome with all sensors resolving correctly
Then you can disable wi-fi, and go back to ethernet.

It's extremely annoying, but that's what I ended up dong.

OttoWinter, 2019-05-24:
Ok, from the logs one of two things is happening:

 - The `.local` address is looked up incorrectly on your network with your DNS settings - for example a DNS server that resolves .local addresses to some random address but not the correct one (I've seen this happen before).
 - The TCP connection is not getting any data through - though SYN/ACK appear to work (unlikely).

Please try with a static IP as seen in the FAQ/wifi page and re-add the devices with their static IPs.

Otherwise you can also enable debug HA logs - there you will also see what address they're resolved to.

yeralin, 2019-05-24:
Nope, `.local` does resolve correctly.

I turned on DEBUG logging, and grepped for `-e esp` and `-e pet`. Here is what I got:
https://pastebin.com/bRD867Kk

yeralin, 2019-05-24:
I believe the important chunk is this:
```
[aioesphomeapi.connection] smart_feeder.local: Connecting to smart_feeder.local:6053 (('192.168.1.32', 6053))
[aioesphomeapi.connection] smart_feeder.local: Opened socket for
[aioesphomeapi.connection] smart_feeder.local: Sending <class 'api_pb2.HelloRequest'>: client_info: "Home Assistant 0.92.2"
[aioesphomeapi.connection] smart_feeder.local: Write: 00 17 01 0A 15 48 6F 6D 65 20 41 73 73 69 73 74 61 6E 74 20 30 2E 39 32 2E 32
[aioesphomeapi.connection] smart_feeder.local: Got message of type <class 'api_pb2.HelloResponse'>: api_version_major: 1
    May 24 19:31:28 raspberrypi hass[13307]: server_info: "smart_feeder (esphome v1.12.2)"
[aioesphomeapi.connection] smart_feeder.local: Successfully connected ('smart_feeder (esphome v1.12.2)' API=1.0)
[aioesphomeapi.connection] smart_feeder.local: Sending <class 'api_pb2.ConnectRequest'>:
[aioesphomeapi.connection] smart_feeder.local: Write: 00 00 03
[aioesphomeapi.connection] smart_feeder.local: Got message of type <class 'api_pb2.ConnectResponse'>:
[homeassistant.components.esphome] Successfully connected to smart_feeder.local
[aioesphomeapi.connection] smart_feeder.local: Sending <class 'api_pb2.DeviceInfoRequest'>:
[aioesphomeapi.connection] smart_feeder.local: Write: 00 00 09
[aioesphomeapi.connection] smart_feeder.local: Got message of type <class 'api_pb2.DeviceInfoResponse'>: name: "smart_feeder"
    May 24 19:31:28 raspberrypi hass[13307]: esphome_core_version: "1.12.2"
[aioesphomeapi.connection] smart_feeder.local: Sending <class 'api_pb2.ListEntitiesRequest'>:
[aioesphomeapi.connection] smart_feeder.local: Write: 00 00 0B
[homeassistant.core] Bus:Handling <Event state_changed[L]: new_state=<state sensor.pet_feeder_interval=unknown; icon=, unit_of_measurement=, friendly_name=Pet feeder interval @ 2019-05-24T14:31:28.493526-04:00>, entity_id=sensor.pet_feeder_interval, old_state=<state sensor.pet_feeder_interval=unavailable; icon=, unit_of_measurement=, friendly_name=Pet feeder interval @ 2019-05-24T14:31:27.272917-04:00>>
[homeassistant.core] Bus:Handling <Event state_changed[L]: new_state=<state sensor.pet_feeder_time=unknown; icon=, unit_of_measurement=, friendly_name=Pet feeder time @ 2019-05-24T14:31:28.505127-04:00>, entity_id=sensor.pet_feeder_time, old_state=<state sensor.pet_feeder_time=unavailable; icon=, unit_of_measurement=, friendly_name=Pet feeder time @ 2019-05-24T14:31:27.305496-04:00>>
[homeassistant.core] Bus:Handling <Event state_changed[L]: new_state=<state sensor.pet_feeder_target_weight=unknown; icon=, unit_of_measurement=, friendly_name=Pet feeder target weight @ 2019-05-24T14:31:28.516267-04:00>, entity_id=sensor.pet_feeder_target_weight, old_state=<state sensor.pet_feeder_target_weight=unavailable; icon=, unit_of_measurement=, friendly_name=Pet feeder target weight @ 2019-05-24T14:31:27.321363-04:00>>
[aioesphomeapi.connection] smart_feeder.local: Got message of type <class 'api_pb2.ListEntitiesSensorResponse'>: object_id: "weight_sensor"
[homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
[aioesphomeapi.connection] smart_feeder.local: Sending <class 'api_pb2.DisconnectRequest'>:
```

yeralin, 2019-05-24:
Here is what I see on esp side:
```
[14:28:55][D][api:546]: Client 'Home Assistant 0.92.2 (192.168.1.5)' connected successfully!
[14:29:00][D][api:073]: Disconnecting Home Assistant 0.92.2 (192.168.1.5)
[14:29:00][D][api:546]: Client 'Home Assistant 0.92.2 (192.168.1.5)' connected successfully!
[14:29:04][D][api:546]: Client 'Home Assistant 0.93.2 (192.168.1.3)' connected successfully!
[14:29:05][D][api:073]: Disconnecting Home Assistant 0.92.2 (192.168.1.5)
[14:29:05][D][api:546]: Client 'Home Assistant 0.92.2 (192.168.1.5)' connected successfully!
[14:29:10][D][api:073]: Disconnecting Home Assistant 0.92.2 (192.168.1.5)
[14:29:10][D][api:546]: Client 'Home Assistant 0.92.2 (192.168.1.5)' connected successfully!
[14:29:15][D][api:073]: Disconnecting Home Assistant 0.92.2 (192.168.1.5)
```

Where `.3` is my Mac's docker-hass instance and `.5` is my raspberry pi's hass instance after the hackery I explained in the [earlier comment](https://github.com/esphome/issues/issues/312#issuecomment-495425102).

OttoWinter, 2019-05-25:
I don't know how to help here further. If the connection does work (and data is exchanged successfully) AND everything works fine for another computer, then it sounds a lot like there's some deeper networking issue somewhere.

One thing in the pastebin that might also be a clue is around line 130: There's a stacktrace (that is unfortunately cut off due to the grepping). Maybe something is indeed in an invalid state. But if you removed all the esphome integrations and re-added them and this is still a problem then the stacktrace is not the problem.

yeralin, 2019-05-25:
Thank you Otto! I'll post a solution if I find one.

closed 2019-05-25 by yeralin

yeralin, 2019-05-27:
Ok, I restored my backup I made couple months ago, and everything is working now!!!
I think you are right, now I remember that I ran `rpi-update` i.e. updated the firmware few weeks ago, and right after that I started experiencing issues. I think the new firmware has buggy networking layer. I'm not a networking expert to find out what was the actual problem.