esphome/issues #314: PCF8574 don't working  with INPUT

IvanHorban, 2019-05-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp8266_03
  platform: ESP8266
  board: esp01_1m
wifi:
  ssid: "****"
  password: "****************"
logger:
api:
  password: '1234'
ota:
  password: '1234'
i2c:
  sda: 2
  scl: 0
  scan: True
pcf8574:
  - id: 'pcf8574_hub'
    address: 0x20
    pcf8575: False
binary_sensor:
  - platform: gpio
    name: "Input_1"
    pin:
      pcf8574: pcf8574_hub
      # Use pin number 0
      number: 0
      # One of INPUT, INPUT_PULLUP or OUTPUT
      mode: INPUT
      inverted: True`

```

**Traceback (if applicable):**

```
INFO Reading configuration...
INFO Starting log output from esp8266_03.local using esphome API
INFO Connecting to esp8266_03.local:6053 (192.168.1.109)
INFO Successfully connected to esp8266_03.local
[22:57:50][I][application:097]: esphome-core version 1.12.2 compiled on May 14 2019, 22:38:44
[22:57:50][C][wifi:371]: WiFi:
[22:57:50][C][wifi:253]: SSID: [redacted]
[22:57:50][C][wifi:254]: IP Address: 192.168.1.109
[22:57:50][C][wifi:256]: BSSID: [redacted]
[22:57:50][C][wifi:257]: Hostname: 'esp8266_03'
[22:57:50][C][wifi:261]: Signal strength: -69 dB ▂▄▆█
[22:57:50][C][wifi:262]: Channel: 9
[22:57:50][C][wifi:263]: Subnet: 255.255.255.0
[22:57:50][C][wifi:264]: Gateway: 192.168.1.1
[22:57:50][C][wifi:265]: DNS1: 192.168.1.1
[22:57:50][C][wifi:266]: DNS2: 172.16.250.10
[22:57:50][C][i2c:035]: I2C Bus:
[22:57:50][C][i2c:036]: SDA Pin: GPIO2
[22:57:50][C][i2c:037]: SCL Pin: GPIO0
[22:57:50][C][i2c:038]: Frequency: 50000 Hz
[22:57:50][I][i2c:040]: Scanning i2c bus for active devices...
[22:57:50][I][i2c:047]: Found i2c device at address 0x20
[22:57:51][C][io.pcf8574:035]: PCF8574:
[22:57:51][C][io.pcf8574:036]: Address: 0x20
[22:57:51][C][io.pcf8574:037]: Is PCF8575: NO
[22:57:51][C][binary_sensor.gpio:022]: GPIO Binary Sensor 'Input_1'
[22:57:51][C][binary_sensor.gpio:023]: Pin: GPIO0 (Mode: INPUT, INVERTED)
[22:57:51][C][logger:142]: Logger:
[22:57:51][C][logger:143]: Level: DEBUG
[22:57:51][C][logger:144]: Log Baud Rate: 115200
[22:57:51][C][logger:145]: Hardware UART: UART0
[22:57:51][C][api:101]: API Server:
[22:57:51][C][api:102]: Address: esp8266_03.local:6053
[22:57:51][C][ota:127]: Over-The-Air Updates:
[22:57:51][C][ota:128]: Address: esp8266_03.local:8266
[22:57:51][C][ota:130]: Using Password.
```

**I have a problem with PCB8574 in the type of INPUT. My code was validated and uploaded to the ESP8266. i2c working, the address of module is visible, but there is no reaction to the change state of the GPIO.**

OttoWinter, 2019-05-15:
Please do these two things:

 - Use the latest dev image of ESPHome (contains a small fix) - see FAQ
 - Enable VERY_VERBOSE logs - see logger page

Thanks!

IvanHorban, 2019-05-15:
I installed the latest version ESPHome 1.13.0-dev and enabled the log level VERY_VERBOSE. After the upgrade the outputs also stopped working.

```
esphome:
  name: esp8266_03
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "********"
  password: "*****************"

logger:
  level: VERY_VERBOSE

api:
  password: '1234'

ota:
  password: '1234'

i2c:
  sda: 2
  scl: 0
  scan: True

pcf8574:
  - id: 'pcf8574_1'
    address: 0x20
    pcf8575: False
  - id: 'pcf8574_2'
    address: 0x21
    pcf8575: False

binary_sensor:
  - platform: gpio
    name: "Input_1"
    pin:
      pcf8574: pcf8574_1
      number: 0
      mode: INPUT
      inverted: False

switch:
  - platform: gpio
    name: "Output_1"
    pin:
      pcf8574: pcf8574_2
      number: 0
      mode: OUTPUT
      inverted: False

```
```
INFO Reading configuration...
INFO Starting log output from esp8266_03.local using esphome API
INFO Connecting to esp8266_03.local:6053 (192.168.1.110)
INFO Successfully connected to esp8266_03.local
[20:18:44][I][app:063]: esphome version 1.13.0-dev compiled on May 15 2019, 19:49:47
[20:18:44][C][wifi:368]: WiFi:
[20:18:44][C][wifi:250]:   SSID: [redacted]
[20:18:44][C][wifi:251]:   IP Address: 192.168.1.110
[20:18:44][C][wifi:253]:   BSSID: [redacted]
[20:18:44][C][wifi:254]:   Hostname: 'esp8266_03'
[20:18:44][C][wifi:258]:   Signal strength: -67 dB ▂▄▆█
[20:18:44][C][wifi:259]:   Channel: 9
[20:18:44][C][wifi:260]:   Subnet: 255.255.255.0
[20:18:44][C][wifi:261]:   Gateway: 192.168.1.1
[20:18:44][C][wifi:262]:   DNS1: 192.168.1.1
[20:18:44][C][wifi:263]:   DNS2: 172.16.250.10
[20:18:44][C][pcf8574:021]: PCF8574:
[20:18:44][C][pcf8574:022]:   Address: 0x20
[20:18:44][C][pcf8574:023]:   Is PCF8575: NO
[20:18:44][E][pcf8574:025]: Communication with PCF8574 failed!
[20:18:44][C][pcf8574:021]: PCF8574:
[20:18:44][C][pcf8574:022]:   Address: 0x21
[20:18:44](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[20:18:44][E][pcf8574:025]: Communication with PCF8574 failed!
[20:18:44][C][i2c:028]: I2C Bus:
[20:18:44][C][i2c:029]:   SDA Pin: GPIO2
[20:18:44][C][i2c:030]:   SCL Pin: GPIO0
[20:18:44][C][i2c:031]:   Frequency: 50000 Hz
[20:18:44][I][i2c:033]: Scanning i2c bus for active devices...
[20:18:44][I][i2c:040]: Found i2c device at address 0x19
[20:18:44][I][i2c:040]: Found i2c device at address 0x20
[20:18:44][I][i2c:040]: Found i2c device at address 0x21
[20:18:44][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Input_1'
[20:18:44][C][gpio.binary_sensor:016]:   Pin: GPIO0 (Mode: INPUT)
[20:18:44][C][switch.gpio:042]: GPIO Switch 'Output_1'
[20:18:44][C][switch.gpio:043]:   Pin: GPIO0 (Mode: OUTPUT)
[20:18:44][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[20:18:44][C][logger:137]: Logger:
[20:18:44][C][logger:138]:   Level: VERY_VERBOSE
[20:18:44][C][logger:139]:   Log Baud Rate: 115200
[20:18:44][C][logger:140]:   Hardware UART: UART0
[20:18:44][C][ota:030]: Over-The-Air Updates:
[20:18:44][C][ota:031]:   Address: esp8266_03.local:8266
[20:18:44][C][ota:033]:   Using Password.
[20:18:44][C][api:102]: API Server:
[20:18:44][C][api:103]:   Address: esp8266_03.local:6053
[20:18:49][VV][api:597]: on_ping_request_
[20:18:54][VV][api:597]: on_ping_request_
[20:18:59][VV][api:597]: on_ping_request_
[20:19:04][VV][api:597]: on_ping_request_
[20:19:09][VV][api:597]: on_ping_request_
[20:19:14][VV][api:597]: on_ping_request_
[20:19:19][VV][api:597]: on_ping_request_
[20:19:24][VV][api:597]: on_ping_request_
[20:19:29][VV][api:597]: on_ping_request_
[20:19:34][VV][api:597]: on_ping_request_
```

IvanHorban, 2019-05-17:
I noticed that if ESP8266 is loaded without the connected PCF8574, and then on the hot connect PCF8574, then it works!
```
INFO Reading configuration...
INFO Starting log output from esp8266_work_01.local using esphome API
INFO Connecting to esp8266_work_01.local:6053 (10.31.31.250)
INFO Successfully connected to esp8266_work_01.local
[15:22:00][I][app:063]: esphome version 1.13.0-dev compiled on May 17 2019, 15:14:14
[15:22:00][C][status_led:019]: Status LED:
[15:22:00][C][status_led:020]:   Pin: GPIO2 (Mode: OUTPUT, INVERTED)
[15:22:00][C][wifi:368]: WiFi:
[15:22:00][C][wifi:250]:   SSID: [redacted]
[15:22:00][C][wifi:251]:   IP Address: 10.31.31.250
[15:22:00][C][wifi:253]:   BSSID: [redacted]
[15:22:00][C][wifi:254]:   Hostname: 'esp8266_work_01'
[15:22:00][C][wifi:258]:   Signal strength: -62 dB ▂▄▆█
[15:22:00][C][wifi:259]:   Channel: 7
[15:22:00][C][wifi:260]:   Subnet: 255.255.255.0
[15:22:00][C][wifi:261]:   Gateway: 10.31.31.1
[15:22:00][C][wifi:262]:   DNS1: 10.66.68.70
[15:22:00][C][wifi:263]:   DNS2: 10.61.131.62
[15:22:00][C][pcf8574:021]: PCF8574:
[15:22:00][C][pcf8574:022]:   Address: 0x20
[15:22:00][C][pcf8574:023]:   Is PCF8575: NO
[15:22:00][C][i2c:028]: I2C Bus:
[15:22:00](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[15:22:00](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[15:22:00][C][i2c:031]:   Frequency: 50000 Hz
[15:22:00][I][i2c:033]: Scanning i2c bus for active devices...
[15:22:00][I][i2c:040]: Found i2c device at address 0x20
[15:22:01][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Input_01'
[15:22:01][C][gpio.binary_sensor:015]:   Device Class: 'window'
[15:22:01][C][gpio.binary_sensor:016]:   Pin: GPIO0 (Mode: INPUT)
[15:22:01][C][logger:137]: Logger:
[15:22:01][C][logger:138]:   Level: VERY_VERBOSE
[15:22:01][C][logger:139]:   Log Baud Rate: 115200
[15:22:01][C][logger:140]:   Hardware UART: UART0
[15:22:01][C][status:036]: Status Binary Sensor 'Office Temperature'
[15:22:01][C][status:036]:   Device Class: 'connectivity'
[15:22:01][C][dallas.sensor:070]: DallasComponent:
[15:22:01][C][dallas.sensor:071]:   Pin: GPIO13 (Mode: INPUT)
[15:22:01][C][dallas.sensor:072]:   Update Interval: 60.0s
[15:22:01][D][dallas.sensor:077]:   Found sensors:
[15:22:01][D][dallas.sensor:080]:     0x610316643E3FFF28
[15:22:01][C][dallas.sensor:085]:   Device 'Office Temperature'
[15:22:01][C][dallas.sensor:085]:     Unit of Measurement: '°C'
[15:22:01](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[15:22:01][C][dallas.sensor:085]:     Icon: 'mdi:thermometer'
[15:22:01](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[15:22:01](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[15:22:01](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[15:22:01][C][ota:030]: Over-The-Air Updates:
[15:22:01][C][ota:031]:   Address: esp8266_work_01.local:8266
[15:22:01][C][ota:033]:   Using Password.
[15:22:01][C][api:102]: API Server:
[15:22:01][C][api:103]:   Address: esp8266_work_01.local:6053
[15:22:01][VV][i2c:083]: Requesting 1 bytes from 0x20:
[15:22:01][VV][i2c:113]:     Received 0b11111111 (0xFF)
[15:22:01][VV][i2c:083]: Requesting 1 bytes from 0x20:
[15:22:01][VV][i2c:113]:     Received 0b11111111 (0xFF)
[15:22:01][VV][i2c:083]: Requesting 1 bytes from 0x20:
[15:22:01][VV][i2c:113]:     Received 0b11111111 (0xFF)
[15:22:01][VV][i2c:083]: Requesting 1 bytes from 0x20:
[15:22:01][VV][i2c:113]:     Received 0b11111111 (0xFF)
[15:22:01][VV][i2c:083]: Requesting 1 bytes from 0x20:
[15:22:01][VV][i2c:113]:     Received 0b11111111 (0xFF)
[15:22:01][VV][i2c:083]: Requesting 1 bytes from 0x20:
[15:22:01][VV][i2c:113]:     Received 0b11111111 (0xFF)
[15:22:01][VV][i2c:083]: Requesting 1 bytes from 0x20:
[15:22:01][VV][i2c:113]:     Received 0b11111111 (0xFF)
[15:22:01][VV][i2c:083]: Requesting 1 bytes from 0x20:
[... 48 lines cut ...]
[15:22:01][VV][i2c:083]: Requesting 1 bytes from 0x20:
[15:22:01][VV][i2c:113]:     Received 0b11111111 (0xFF)
[15:22:01][VV][i2c:083]: Requesting 1 bytes from 0x20:
[15:22:01][VV][i2c:113]:     Received 0b11111111 (0xFF)
[15:22:01][VV][i2c:083]: Requesting 1 bytes from 0x20:
[15:22:01][VV][i2c:113]:     Received 0b11111111 (0xFF)
[15:22:01][VV][i2c:083]: Requesting 1 bytes from 0x20:
[15:22:01][VV][i2c:113]:     Received 0b11111111 (0xFF)
[15:22:01][VV][i2c:083]: Requesting 1 bytes from 0x20:
[15:22:01][VV][i2c:113]:     Received 0b11111111 (0xFF)
[15:22:01][VV][i2c:083]: Requesting 1 bytes from 0x20:
[15:22:02][VV][i2c:113]:     Received 0b11111111 (0xFF)
[15:22:02][VV][i2c:083]: Requesting 1 bytes from 0x20:
[15:22:02][VV][i2c:113]:     Received 0b11111111 (0xFF)
[15:22:02][VV][i2c:083]: Requesting 1 bytes from 0x20:
[15:22:02][VV][i2c:113]:     Received 0b11111111 (0xFF)
[15:22:02][VV][i2c:083]: Requesting 1 bytes from 0x20:
[15:22:02][VV][i2c:113]:     Received 0b11111111 (0xFF)
[15:22:02][VV][i2c:083]: Requesting 1 bytes from 0x20:
[15:22:02][VV][i2c:113]:     Received 0b11011111 (0xDF)
[15:22:02][VV][i2c:083]: Requesting 1 bytes from 0x20:
[15:22:02][VV][i2c:113]:     Received 0b11111111 (0xFF)
[15:22:02][VV][i2c:083]: Requesting 1 bytes from 0x20:
[15:22:02][VV][i2c:113]:     Received 0b11111111 (0xFF)
```

apeeters, 2019-05-31:
I ran into the same issue after upgrading to 1.13/1.13.1. Unfortunately I can only log remotely and the VV logs are discarded before I can read them.

 @IvanHorban how have you been able to hot connect the pcf8574? At what moment do you connect it?

apeeters, 2019-05-31:
It seems a matter of initialisation order: on 1.12 I see i2c logging before pcf8574, on 1.13 it is the other way around.

OttoWinter, 2019-05-31:
@apeeters Ah thanks! that was the mistake, will be fixed in 1.13.2

closed 2019-05-31 by OttoWinter

apeeters, 2019-05-31:
Works like a charm again, thanks!