esphome/issues #315: OTA error message refers to non-existent entry in faq

david-sawatzke, 2019-05-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

esp01(esp8266)

**Affected component:**

[ota](https://esphome.io/components/ota.html) or [faq](https://esphome.io/guides/faq.html)

**Description of problem:**
When uploading a file that's too large, the upload process ends in an error that refers to a minimal firmare mentioned in the FAQ:
```
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (see FAQ)
```

Unfortunately there's no mention of this in the faq
**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: foo
  platform: ESP8266
  board: esp01
ota:
// More stuff to bloat the binary
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
- I guess minimal firmware refers to just including the bare minimum, thus having a small enough image to update successfully and then being able to use the gained free space to properly update
  
  Unfortunately that's still too large

OttoWinter, 2019-05-16:
There should be lines at the end of the compile process showing the size of the binary - pleas post those here.

Also please post the ESPHome version used here, and logs from the ESP during the upload process.

Thanks!

dskw, 2019-05-16:
Same problem for me and ESP-01 chips. 
Upload output:
```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/node1
Processing node1 (platform: espressif8266@1.8.0; framework: arduino; board: esp01)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 512k
HARDWARE: ESP8266 80MHz 80KB RAM (512KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <esphome-core> 1.12.2
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
Retrieving maximum program size /data/node1/.pioenvs/node1/firmware.elf
Checking size /data/node1/.pioenvs/node1/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  38.3% (used 31396 bytes from 81920 bytes)
PROGRAM: [=======   ]  65.6% (used 327892 bytes from 499696 bytes)
========================= [SUCCESS] Took 10.44 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of node1.local
INFO  -> 192.168.23.112
INFO Uploading /data/node1/.pioenvs/node1/firmware.bin (332032 bytes)
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (see FAQ)
```

Config:
```
esphome:
  name: node1
  platform: ESP8266
  board: esp01

wifi:
  ssid: "skynet"
  password: "FingerWegVonSkyNet"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

sensor:
  - platform: dht
    pin: GPIO2
    temperature:
      name: "temperature1"
    humidity:
      name: "humidity1"
    update_interval: 60s
    model: AM2302
```

I've also tried to compile a minimal firmware with just esphome, wifi and ota. No difference:
```

Linking /data/node1/.pioenvs/node1/firmware.elf
Building /data/node1/.pioenvs/node1/firmware.bin
Retrieving maximum program size /data/node1/.pioenvs/node1/firmware.elf
Checking size /data/node1/.pioenvs/node1/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  37.3% (used 30596 bytes from 81920 bytes)
PROGRAM: [======    ]  60.5% (used 302116 bytes from 499696 bytes)
======================== [SUCCESS] Took 242.25 seconds ========================
INFO Successfully compiled program.
INFO Resolving IP address of node1.local
INFO  -> 192.168.23.112
INFO Uploading /data/node1/.pioenvs/node1/firmware.bin (306256 bytes)
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (see FAQ)

dskw, 2019-05-16:
Same problem for me and ESP-01 chips. 
Upload output:
```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/node1
Processing node1 (platform: espressif8266@1.8.0; framework: arduino; board: esp01)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01.html
PLATFORM: Espressif 8266 > Espressif Generic ESP8266 ESP-01 512k
HARDWARE: ESP8266 80MHz 80KB RAM (512KB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.0
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <esphome-core> 1.12.2
|   |-- <ArduinoJson-esphomelib> 5.13.3
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
Retrieving maximum program size /data/node1/.pioenvs/node1/firmware.elf
Checking size /data/node1/.pioenvs/node1/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  38.3% (used 31396 bytes from 81920 bytes)
PROGRAM: [=======   ]  65.6% (used 327892 bytes from 499696 bytes)
========================= [SUCCESS] Took 10.44 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of node1.local
INFO  -> 192.168.23.112
INFO Uploading /data/node1/.pioenvs/node1/firmware.bin (332032 bytes)
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (see FAQ)
```

Config:
```
esphome:
  name: node1
  platform: ESP8266
  board: esp01

wifi:
  ssid: "skynet"
  password: "***"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

sensor:
  - platform: dht
    pin: GPIO2
    temperature:
      name: "temperature1"
    humidity:
      name: "humidity1"
    update_interval: 60s
    model: AM2302
```

I've also tried to compile a minimal firmware with just esphome, wifi and ota. No difference:
```

Linking /data/node1/.pioenvs/node1/firmware.elf
Building /data/node1/.pioenvs/node1/firmware.bin
Retrieving maximum program size /data/node1/.pioenvs/node1/firmware.elf
Checking size /data/node1/.pioenvs/node1/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  37.3% (used 30596 bytes from 81920 bytes)
PROGRAM: [======    ]  60.5% (used 302116 bytes from 499696 bytes)
======================== [SUCCESS] Took 242.25 seconds ========================
INFO Successfully compiled program.
INFO Resolving IP address of node1.local
INFO  -> 192.168.23.112
INFO Uploading /data/node1/.pioenvs/node1/firmware.bin (306256 bytes)
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (see FAQ)

HFsi, 2019-05-16:
Check the other chip on the board, if it says 80H in the end then means the ESP01 has 1 Mb of memory.
I've mines with esp01-1m board and works fine

dskw, 2019-05-16:
It writes FM25Q08 1821JOG (or J0G hard to read)
Google tells me it's an 8M-Bit Serial Flash Memory but can that be true? I thought esp8266 comes with 1m or 4m usually and I would not expect the 01 to have even more.
Anyway, will try a different board definition with higher mem.

Still: If the compiler tells me as a (new) user i should check the FAQ and I cannot find any mention of "minimal firmware" in it, then maybe we can add some note in there? :)

david-sawatzke, 2019-05-17:
So, just to add on: I'm using ESPHome 1.12.2 as an Hassio extension.
But the issue is primarily the reference to FAQ entry, I don't expect OTA to work with 512K.
Using this config:
```
esphome:
  name: …
  platform: ESP8266
  board: esp01

wifi:
  ssid: …
  password: …
# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:

ota:
```
Upload output:
```
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [====      ]  38.3% (used 31380 bytes from 81920 bytes)
PROGRAM: [======    ]  64.5% (used 322076 bytes from 499696 bytes)
======================== [SUCCESS] Took 271.99 seconds ========================
INFO Successfully compiled program.
INFO Resolving IP address of nyc.local
INFO  -> 10.30.0.171
INFO Uploading /data/nyc/.pioenvs/foo/firmware.bin (326224 bytes)
ERROR Error binary size: Error: ESP does not have enough space to store OTA file. Please try flashing a minimal firmware (see FAQ)
```
Logs during upload:
```
[16:25:11][D][ota:179]: Starting OTA Update from 10.30.0.226...
```

OttoWinter, 2019-05-27:
The outdated reference has/will been fixed in 1.13 👍

closed 2019-05-27 by OttoWinter