esphome/issues #317: NodeMCU ESP32 stops tracking Xiaomi Mijia

roybosch, 2019-05-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESP-32S (https://www.ebay.com/itm/ESP-32S-ESP32-NodeMCU-Development-Board-2-4GHz-WIFI-Bluetooth-Dual-Mode/272656534488?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649)

**Affected component:**

https://esphome.io/components/sensor/xiaomi_mijia.html

**Description of problem:**
I'm using a Xiaomi Mijia to track temperature and humidity. After a while, sometimes 24 hours, sometimes 72 hours, it stops tracking the temperature and humidity. My graph in Hassio then shows a straight line on the temp and hum that was last measured. 

After a manual reboot it starts working again. After one, two or three days it stops tracking the temperature and humidity again.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ble_tracker
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "SSID"
  password: "PASS"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'PASS'

ota:
  password: 'PASS'

esp32_ble_tracker:

sensor:
  - platform: xiaomi_mijia
    mac_address: 4C:65:A8:DE:D5:7F
    temperature:
      name: "Xiaomi MiJia Temperature"
    humidity:
      name: "Xiaomi MiJia Humidity"
    battery_level:
      name: "Xiaomi MiJia Battery Level"

```

**Traceback (if applicable):**

```
INFO Reading configuration...
INFO Starting log output from ble_tracker.local using esphome API
INFO Connecting to ble_tracker.local:6053 (192.168.178.33)
INFO Successfully connected to ble_tracker.local
[07:40:02][I][application:097]: esphome-core version 1.12.2 compiled on May  7 2019, 17:57:09
[07:40:02][C][wifi:371]: WiFi:
[07:40:02][C][wifi:253]:   SSID: [redacted]
[07:40:02][C][wifi:254]:   IP Address: 192.168.178.33
[07:40:02][C][wifi:256]:   BSSID: [redacted]
[07:40:02][C][wifi:257]:   Hostname: 'ble_tracker'
[07:40:02][C][wifi:261]:   Signal strength: -57 dB ▂▄▆█
[07:40:02][C][wifi:262]:   Channel: 1
[07:40:02][C][wifi:263]:   Subnet: 255.255.255.0
[07:40:02][C][wifi:264]:   Gateway: 192.168.178.1
[07:40:02][C][wifi:265]:   DNS1: 84.116.46.23
[07:40:02][C][wifi:266]:   DNS2: 84.116.46.22
[07:40:02][C][logger:142]: Logger:
[07:40:03][C][logger:143]:   Level: DEBUG
[07:40:03][C][logger:144]:   Log Baud Rate: 115200
[07:40:03][C][logger:145]:   Hardware UART: UART0
[07:40:03][C][esp32_ble_tracker:703]: BLE Tracker:
[07:40:03][C][esp32_ble_tracker:704]:   Scan Interval: 300 s
[07:40:03][C][esp32_ble_tracker:712]:   Xiaomi 00004C65A8DED57F
[07:40:03][C][esp32_ble_tracker:713]:     Temperature  'Xiaomi MiJia Temperature'
[07:40:03][C][esp32_ble_tracker:713]:       Unit of Measurement: '°C'
[07:40:03][C][esp32_ble_tracker:713]:       Accuracy Decimals: 1
[07:40:03][C][esp32_ble_tracker:714]:     Humidity  'Xiaomi MiJia Humidity'
[07:40:03][C][esp32_ble_tracker:714]:       Unit of Measurement: '%'
[07:40:03][C][esp32_ble_tracker:714]:       Accuracy Decimals: 1
[07:40:03][C][esp32_ble_tracker:714]:       Icon: 'mdi:water-percent'
[07:40:03][C][esp32_ble_tracker:718]:     Battery Level  'Xiaomi MiJia Battery Level'
[07:40:03][C][esp32_ble_tracker:718]:       Unit of Measurement: '%'
[07:40:03][C][esp32_ble_tracker:718]:       Accuracy Decimals: 0
[07:40:03][C][esp32_ble_tracker:718]:       Icon: 'mdi:battery'
[07:40:03][C][api:101]: API Server:
[07:40:03][C][api:102]:   Address: ble_tracker.local:6053
[07:40:03][C][ota:127]: Over-The-Air Updates:
[07:40:03][C][ota:128]:   Address: ble_tracker.local:3232
[07:40:03][C][ota:130]:   Using Password.
```

**Additional information and things you've tried:**
I tried uploading the config again. That results in a reboot, which forces it to track again, temporarily. 

I checked the log for error messages when it stalls. I've pasted it above, everything looks normal there.

jesusrop, 2019-05-25:
I have the same issue with another ESP32 board based on ESP32-WROOM. In my case, I also use the BLE tracker sensor and that also gets stuck. Maybe the whole Bluetooth component gets blocked.

My workaround is to use the restart switch in esphome: https://esphome.io/components/switch/restart.html Currently, i just reboot them when I nottice this, but I might try to automate this in the future.

OttoWinter, 2019-05-25:
The most likely cause is probably that the ESP-IDF has a bug somewhere and does not create a `ESP_GAP_SEARCH_INQ_CMPL_EVT` event after the scan is done.

Some logs of the final successful scan could help - also some logs after could help. I see you attached logs of the problem occuring, but how long did you let the log window open? Was it more than 5 minutes?

roybosch, 2019-05-26:
> me logs of the final successful scan could help - also some logs after could help. I see you attached logs of the problem occuring, but how long did you let the log window open? W

I opened the log after the problem occurred. I can't open the log and wait for the problem to happen, last time the problem happened after 7 days or so.

OttoWinter, 2019-05-27:
> I opened the log after the problem occurred. I can't open the log and wait for the problem to happen, last time the problem happened after 7 days or so.

I understand, that's why I also asked the second question about how long you left the logs open in the initial issue logs - if it's more than 5 minutes and no messages are seen that confirms my suspicion

roybosch, 2019-05-27:
> I understand, that's why I also asked the second question about how long you left the logs open in the initial issue logs - if it's more than 5 minutes and no messages are seen that confirms my suspicion

I left the log open for a couple of minutes, after I discovered my graphs showing a straight line. Indeed there were no messages at all.

I hope this gives you insight in the problem.

jesusrop, 2019-05-29:
I left the log running for ~1h when I detected a failing device (hence the device was already failing when I opened the log page). Short version: nothing appeared in the logs during that hour.

`
INFO Reading configuration...
INFO Starting log output from 10.0.27.70 using esphome API
INFO Connecting to 10.0.27.70:6053 (10.0.27.70)
INFO Successfully connected to 10.0.27.70
[21:50:38][I][application:097]: esphome-core version 1.12.2 compiled on May 25 2019, 15:33:39
[21:50:38][C][wifi:371]: WiFi:
[21:50:38][C][wifi:253]:   SSID: [redacted]
[21:50:38][C][wifi:254]:   IP Address: 10.0.27.70
[21:50:38][C][wifi:256]:   BSSID: [redacted]
[21:50:38][C][wifi:257]:   Hostname: 'hab_sala'
[21:50:38][C][wifi:261]:   Signal strength: -35 dB ▂▄▆█
[21:50:38][C][wifi:262]:   Channel: 4
[21:50:38][C][wifi:263]:   Subnet: 255.255.255.0
[21:50:38][C][wifi:264]:   Gateway: 10.0.27.1
[21:50:38][C][wifi:265]:   DNS1: 1.1.1.1
[21:50:38][C][wifi:266]:   DNS2: 1.0.0.1
[21:50:38][C][logger:142]: Logger:
[21:50:38][C][logger:143]:   Level: DEBUG
[21:50:38][C][logger:144]:   Log Baud Rate: 115200
[21:50:38][C][logger:145]:   Hardware UART: UART0
[21:50:38][C][esp32_ble_tracker:703]: BLE Tracker:
[21:50:38][C][esp32_ble_tracker:704]:   Scan Interval: 180 s
[21:50:38][C][esp32_ble_tracker:706]:   Presence 'MiFit ### Sala'
[21:50:38][C][esp32_ble_tracker:706]:     Device Class: 'presence'
[21:50:38][C][esp32_ble_tracker:706]:   Presence 'MiFit ### Sala'
[21:50:38][C][esp32_ble_tracker:706]:     Device Class: 'presence'
[21:50:38][C][esp32_ble_tracker:712]:   Xiaomi #################
[21:50:38][C][esp32_ble_tracker:713]:     Temperature  'Sala Temperatura'
[21:50:38][C][esp32_ble_tracker:713]:       Unit of Measurement: '°C'
[21:50:38][C][esp32_ble_tracker:713]:       Accuracy Decimals: 1
[21:50:38][C][esp32_ble_tracker:714]:     Humidity  'Sala Humitat'
[21:50:38][C][esp32_ble_tracker:714]:       Unit of Measurement: '%'
[21:50:38][C][esp32_ble_tracker:714]:       Accuracy Decimals: 1
[21:50:38][C][esp32_ble_tracker:714]:       Icon: 'mdi:water-percent'
[21:50:38][C][esp32_ble_tracker:718]:     Battery Level  'Sala Bateria'
[21:50:38][C][esp32_ble_tracker:718]:       Unit of Measurement: '%'
[21:50:38][C][esp32_ble_tracker:718]:       Accuracy Decimals: 0
[21:50:38][C][esp32_ble_tracker:718]:       Icon: 'mdi:battery'
[21:50:38][C][switch.restart:028]: Restart Switch 'hab_sala Restart'
[21:50:38][C][switch.restart:028]:   Icon: 'mdi:restart'
[21:50:38][C][api:101]: API Server:
[21:50:38][C][api:102]:   Address: 10.0.27.70:6053
[21:50:38][C][ota:127]: Over-The-Air Updates:
[21:50:38][C][ota:128]:   Address: 10.0.27.70:3232
[21:50:38][C][ota:130]:   Using Password.
`

OttoWinter, 2019-06-06:
@jesusrop Try with latest ESPHome version please - in the 1.13.4 version I cannot replicate this issue.

roybosch, 2019-06-06:
Since the last update, I don't face this problem anymore.

ESP32 camera does have the problem however...

jesusrop, 2019-06-10:
> @jesusrop Try with latest ESPHome version please - in the 1.13.4 version I cannot replicate this issue.

No strange behavior since update to 1.13.4. i waited a couple of days to make sure the issue is gone. I don't think that with the previous versions it would have lasted so long without loosing connection.

Will try 1.13.5 today.

Thanks!!

Koenkk, 2019-06-27:
With ESPHome 1.13.6 I still have regular disconnect (sensor becoming `unavailable`) issues. This can be seen well in the picture below, where every 'break' of the line is a disconnect:

[image]

Please let me know how I can help to solve this issue.

**Context**
This is problematic for me because I created a separate Home Assistant sensor which checks if the plant needs water. To avoid flickering of the sensor state, my sensor has an `delay_on` (see below), because of the regular unavailable states the sensor is never triggered. 

```yaml
living_room_plant:
    delay_on:
        hours: 5
    icon_template: >
        {% if is_state('binary_sensor.living_room_plant' ,'on') %} mdi:flower
        {% else %} mdi:flower-outline
        {% endif %}
    value_template: >-
        {{ states('sensor.living_room_plant_moisture')|float < 20 or states('sensor.living_room_plant_conductivity')|float < 350 }}

```

OttoWinter, 2019-07-03:
@Koenkk Ok, interesting - at least I don't see any of these issues with my devices.

Please post your ESPHome config (specifically anything related to arduino_version, and scan_interval if set)

Koenkk, 2019-07-05:
@OttoWinter log while this happens: https://pastebin.com/8kg5d5ng

**Configuration**

```yaml
esphome:
  name: living_room_plant
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "--"
  password: "--"

logger:

api:
  password: "--"

ota:
  password: "--"

esp32_ble_tracker:

sensor:
  - platform: xiaomi_miflora
    mac_address: 'C4:7C:8D:6A:3F:E4'
    temperature:
      name: living_room_plant_temperature
    moisture:
      name: living_room_plant_moisture
    illuminance:
      name: living_room_plant_illuminance
    conductivity:
      name: living_room_plant_conductivity
    battery_level:
      name: living_room_plant_battery
```

brandond, 2019-07-05:
Your device drops off the wifi twice in the period you retrieved logs. If it's not on the network, that would probably explain why the sensors are showing as unavailable in HA.

Koenkk, 2019-07-05:
@brandond yes I know, I wonder if this issue has something to do with it. Initially I thought this because this is the only of my +- 15 devices running esphome that has connectivity issues, the others are rock solid.

brandond, 2019-07-05:
Do you have the ability to leave it connected over serial for a while? When you're fighting connectivity issues, being able to pull the logs locally is pretty important; you obviously can't get the logs remotely to figure out what's going on when it drops off the network.

Koenkk, 2019-07-07:
Here is the log: https://pastebin.com/5zQxZJsf

A few observations:
- Log is +- 30 minutes long
- During this time home assistant disconnects twice (search for `Disconnecting Home Assistant 0.94.4`)
- I don't see a reason why home assistant disconnects (e.g. nothing is logged about losing wifi connection).
- Both disconnects happens around the scan (not sure if it is related though)

Note: debug loggin was activated using

```yaml
logger:
  level: debug

debug:
```

brandond, 2019-07-08:
You wouldn't see logs about wifi errors since you can't connect to it to pull the logs while wifi is down - there's no buffering so you only see what's logged while you're cinnected. That's why I asked about getting logs via local serial output.

That said, the wifi and ble stack on these devices use the same phy and are known to be kinda flakey so I wouldn't be surprised if the scans are causing wifi to drop.

Koenkk, 2019-07-08:
@brandond the log from https://github.com/esphome/issues/issues/317#issuecomment-508990380 is via serial output (see first line: `INFO Starting log output from /dev/cu.SLAB_USBtoUART with baud rate 115200
`)

OttoWinter, 2019-07-10:
@Koenkk If the error is not logged in the ESPHome logs it's because the error occured on the other end (HA) or somewhere in the TCP/IP stack - all that could be logged would be a generic "connection dropped" because more info is not available. Maybe the HA logs contain more info (enable debug logs in logger), but otherwise the error is probably in TCP/IP stack and thus hard to track down.

Koenkk, 2019-07-11:
@OttoWinter I don't think HA is the error, as also esphome logs is disconnected, left terminal shows log output via serial, right via wifi:

[image]

But as this may indeed be hard to solve I currently use the following Home Assistant configuration to workaround it:

```yaml
binary_sensor:
  - platform: template
    sensors:
      living_room_plant:
        delay_on:
          minutes: 60
        value_template: >-
          {% if states('sensor.living_room_plant_moisture')|float < 20 or states('sensor.living_room_plant_conductivity')|float < 350 %}
            true
          {% elif is_state('sensor.living_room_plant_moisture', 'unavailable') or is_state('sensor.living_room_plant_conductivity', 'unavailable') %}
            {% if is_state('binary_sensor.living_room_plant', 'on') %}
              true
            {% else %}
              false
            {% endif %}
          {% else %}
            false
          {% endif %}
```

glmnet, 2019-07-11:
What are all those three different ip addressees?

y0ast, 2019-08-28:
@Koenkk Did you ever resolve this problem? The workaround is not ideal.

I am facing the same issues you describe with an ESP-WROOM-32 (ESP32 DEVKIT V1).

Koenkk, 2019-08-29:
@y0ast no

neographikal, 2019-09-15:
Seem to have this problem with the latest 1.14.0-dev version as well.

closed 2019-10-21 by OttoWinter