esphome/issues #318: ESP32Cam rec

leoMehlig, 2019-05-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

brandond, 2019-05-18:
Can you update to show the actual error message that you're getting?

leoMehlig, 2019-05-18:
Here is the error log from an Automation that calls the service
```
Error while executing automation automation.start_camera_on_front_door_opened. Unknown error for call_service at pos 1:
Traceback (most recent call last): File "/usr/local/lib/python3.7/site-packages/homeassistant/components/automation/__init__.py", line 378, in action await script_obj.async_run(variables, context) File "/usr/local/lib/python3.7/site-packages/homeassistant/helpers/script.py", line 131, in async_run await self._handle_action(action, variables, context) File "/usr/local/lib/python3.7/site-packages/homeassistant/helpers/script.py", line 210, in _handle_action action, variables, context) File "/usr/local/lib/python3.7/site-packages/homeassistant/helpers/script.py", line 299, in _async_call_service context=context File "/usr/local/lib/python3.7/site-packages/homeassistant/helpers/service.py", line 89, in async_call_from_config domain, service_name, service_data, blocking=blocking, context=context) File "/usr/local/lib/python3.7/site-packages/homeassistant/core.py", line 1141, in async_call self._execute_service(handler, service_call)) File "/usr/local/lib/python3.7/site-packages/homeassistant/core.py", line 1163, in _execute_service await handler.func(service_call) File "/usr/local/lib/python3.7/site-packages/homeassistant/helpers/entity_component.py", line 194, in handle_service required_features File "/usr/local/lib/python3.7/site-packages/homeassistant/helpers/service.py", line 316, in entity_service_call future.result()  # pop exception if have File "/usr/local/lib/python3.7/site-packages/homeassistant/helpers/service.py", line 339, in _handle_service_platform_call await func(entity, data) File "/usr/local/lib/python3.7/site-packages/homeassistant/components/camera/__init__.py", line 653, in async_handle_record_service .format(camera.entity_id)) homeassistant.exceptions.HomeAssistantError: camera.hallway_camera does not support record service
```

nickrout, 2019-05-20:
I shall test mine tonight.

pmaciocia, 2019-05-20:
Also seeing this issue:

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/homeassistant/components/websocket_api/commands.py", line 121, in handle_call_service
    connection.context(msg))
  File "/usr/local/lib/python3.7/site-packages/homeassistant/core.py", line 1141, in async_call
    self._execute_service(handler, service_call))
  File "/usr/local/lib/python3.7/site-packages/homeassistant/core.py", line 1163, in _execute_service
    await handler.func(service_call)
  File "/usr/local/lib/python3.7/site-packages/homeassistant/helpers/entity_component.py", line 194, in handle_service
    required_features
  File "/usr/local/lib/python3.7/site-packages/homeassistant/helpers/service.py", line 316, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/local/lib/python3.7/site-packages/homeassistant/helpers/service.py", line 339, in _handle_service_platform_call
    await func(entity, data)
  File "/usr/local/lib/python3.7/site-packages/homeassistant/components/camera/__init__.py", line 653, in async_handle_record_service
    .format(camera.entity_id))
homeassistant.exceptions.HomeAssistantError: camera.my_camera does not support record service
```

pmaciocia, 2019-05-20:
@leoMehlig you might want to remove your wifi & api credentials from the yaml file you have posted there

leoMehlig, 2019-05-20:
Oh... thanks @pmaciocia

OttoWinter, 2019-05-25:
Yes this is correct, ESPHome uses something similar to MJPEG, and thus does not support the HA stream component. That includes HA's record service.

There was a longer discussion about this (I think HA's discord, somewhere in #beta maybe).

Anyway, this would need to be solved on the HA side (namely an MJPEG to RTSP converter)

closed 2019-05-25 by OttoWinter