esphome/issues #321: Heltec wifi 8 kit no data on screen

MakkeLeon, 2019-05-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

MakkeLeon, 2019-05-17:
hassio/esphome
esp8266 heltec wifi 8 kit
`esphome:
  name: oled
  platform: ESP8266
  board: heltec_wifi_kit_8

i cant get any text or data on the oled screen.
i tried diverent pins and diverent fonts but nothing seems to work.
i upload wiht esphomeflasher.
if i run a program from platformio or arduino ide wiht u8g2 it works fine.
Help me please 

wifi:
  ssid: "*******"
  password: "*******"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

font:
  - file: "ComicSansMSRegular.ttf"
    id: ComicSansMSRegular
    size: 10

spi:
  clk_pin: GPIO14
  mosi_pin: GPIO13

display:
  - platform: ssd1306_spi
    model: "SSD1306 128x32"
    cs_pin: GPIO4
    dc_pin: GPIO5
    reset_pin: GPIO16
    lambda: |-
      it.print(0, 0, id(ComicSansMSRegular), "Hello World!");
  `

OttoWinter, 2019-05-27:
Please provide more information:

- Is only the display not working? Or is nothing running (no logs etc)
- What display is this specifically, are you sure it's the SSD1306 with 128x32?

wischwien, 2019-06-09:
i use also a heletc wifi kit 8  - in my opinion it uses i2c

this yaml is working:

i2c:
# nothig - esphome find all pins itself

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x32"
    reset_pin: D0
    address: 0x3C
    id: my_display

OttoWinter, 2019-07-11:
Closing due to inactivity - and see @wischwien 's solution, you probably have set the wrong `model` parameter

closed 2019-07-11 by OttoWinter

anygenerator, 2019-07-16:
I am having the same issue using the example sketches for the Wifi 8 kit.