esphome/issues #323: Status topic message not sent after OTA

brandond, 2019-05-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
virtualenv + pip

**ESP (ESP32/ESP8266, Board/Sonoff):**
esp8266

**Affected component:**
MQTT Status + OTA Upgrade

**Description of problem:**
I've been going through and upgrading nodes from 1.11 and 1.12 to 1.13, and so far all of them have required a second reboot after updating to appear online in Home Assistant. From watching MQTT, everything looks fine, except it appears that the 'online' payload is not getting sent to the status topic. All the debug logs, output state messages, etc are fine. After a second boot (power cycling the node, doing another OTA to the same build, etc) the message comes through and home assistant updates properly.

**Problem-relevant YAML-configuration entries:**
n/a

brandond, 2019-05-18:
As a data point, I am using a mosquitto MQTT broker, not the one built in to hass. I wonder if maybe the broker is not noticing the dropped connection until later, and the 'offline' LWT message is getting sent to the status topic, and that overwrites the 'online' message that was published during startup?

OttoWinter, 2019-05-27:
@brandond Hmm not sure - at least after the MQTT spec that should not happen:

The ESP device has a fixed client_id, so if a new client connects with the same client_id, the broker should immediately close the connection to the old one before doing anything else.

The birth message should also be re-sent as many times as necessary until the broker has accepted it:

https://github.com/esphome/esphome/blob/72a80f559a0c1ef905fd4db9e164ebedbd3b1a2d/esphome/components/mqtt/mqtt_client.cpp#L268-L270

Probably this is some sort of timing problem where the broker does not accept packets until after some period of time. On the other hand many implementations probably rely on quick messaging so that's kinda unlikely.

brandond, 2019-05-28:
I'll try to grab a packet capture the next time I upgrade some devices.

brandond, 2019-05-30:
So, the issue here is that the mqtt clientid changed between 1.12 and 1.13. This means that the new connection post-upgrade doesn't kick off the old one, and the LWT message from the old client gets sent to the topic AFTER the device has finished booting, and replaces it.

```
mqtt_1   | 1559249172: New client connected from 172.20.0.4 as plug00 (p2, c1, k15).
mqtt_1   | 1559249172:  plug00/status
mqtt_1   | 1559249172: Sending CONNACK to plug00 (0, 0)
mqtt_1   | 1559249173: Received SUBSCRIBE from plug00
mqtt_1   | 1559249173:  plug00/switch/smart_plug_00_power/command (QoS 0)
mqtt_1   | 1559249173: plug00 0 plug00/switch/smart_plug_00_power/command
mqtt_1   | 1559249173: Sending SUBACK to plug00
mqtt_1   | 1559249173: Received PUBLISH from plug00 (d0, q0, r1, m0, 'plug00/status', ... (6 bytes))
mqtt_1   | 1559249173: Sending PUBLISH to 4Ippztiw3SZLoYt9dJ0dye (d0, q0, r0, m0, 'plug00/status', ... (6 bytes))
mqtt_1   | 1559249180: Client plug00-840d8e5648ef has exceeded timeout, disconnecting.
mqtt_1   | 1559249180: Sending PUBLISH to 4Ippztiw3SZLoYt9dJ0dye (d0, q0, r0, m0, 'plug00/status', ... (7 bytes))
```

OttoWinter, 2019-05-31:
@brandond Oh the client ID changed? Hmm that was not intentional. I think the best option is to just fix the client id - there probably aren't too many 1.13 users yet (weekend probably will be more).

closed 2019-05-31 by OttoWinter