esphome/issues #326: 'Delay' automation in mqtt on_message/on_json_message results in compilation errors

nirkons, 2019-05-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HassOS image on intel NUC
**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU
**Affected component:**

MQTT automation
https://esphome.io/components/mqtt.html
**Description of problem:**
Inserting a "delay" automation anywhere in mqtt automation results in the following compilation error.
Removing the delay, everything (including the automation itself) works fine.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
on_json_message: - topic: esp/lightservodirection qos: 0 then: - if: condition: lambda: 'return (x["direction"] == "up");' then: - logger.log: "Going up!" - delay: 2s - servo.write: id: lightservo level: -100% else: - logger.log: "Going down!" - servo.write: id: lightservo level: 100% - topic: esp/lightservostop qos: 0 then: - logger.log: "Stopping!" - servo.write: id: lightservo level: 0%
```

**Traceback (if applicable):**

```
Compiling /data/lightmotor/.pioenvs/lightmotor/src/main.cpp.o In file included from /data/lightmotor/.piolibdeps/esphome-core/src/esphome/automation.h:302:0, from /data/lightmotor/.piolibdeps/esphome-core/src/esphome/application.h:7, from /data/lightmotor/.piolibdeps/esphome-core/src/esphome.h:1, from src/main.cpp:3: /data/lightmotor/.piolibdeps/esphome-core/src/esphome/automation.tcc: In instantiation of 'void esphome::DelayAction<Ts>::play(Ts ...) [with Ts = {const ArduinoJson::JsonObject&}]': src/main.cpp:67:1: required from here /data/lightmotor/.piolibdeps/esphome-core/src/esphome/automation.tcc:74:3: error: use of deleted function 'std::_Bind<_Functor(_Bound_args ...)>::_Bind(const std::_Bind<_Functor(_Bound_args ...)>&) [with _Functor = std::_Mem_fn<void (esphome::Action<const ArduinoJson::JsonObject&>::*)(const ArduinoJson::JsonObject&)>; _Bound_args = {esphome::DelayAction<const ArduinoJson::JsonObject&>*, ArduinoJson::JsonObject}]' this->set_timeout(this->delay_.value(x...), f); ^ In file included from /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:66:0, from /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/algorithm:62, from /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:255, from /data/lightmotor/.piolibdeps/esphome-core/src/esphome/espmath.h:4, from /data/lightmotor/.piolibdeps/esphome-core/src/esphome/automation.h:5, from /data/lightmotor/.piolibdeps/esphome-core/src/esphome/application.h:7, from /data/lightmotor/.piolibdeps/esphome-core/src/esphome.h:1, from src/main.cpp:3: /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/functional:1339:7: note: 'std::_Bind<_Functor(_Bound_args ...)>::_Bind(const std::_Bind<_Functor(_Bound_args ...)>&) [with _Functor = std::_Mem_fn<void (esphome::Action<const ArduinoJson::JsonObject&>::*)(const ArduinoJson::JsonObject&)>; _Bound_args = {esphome::DelayAction<const ArduinoJson::JsonObject&>*, ArduinoJson::JsonObject}]' is implicitly deleted because the default definition would be ill-formed: _Bind(const _Bind&) = default; ^ /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/functional:1339:7: error: use of deleted function 'constexpr std::tuple<_T1, _T2>::tuple(const std::tuple<_T1, _T2>&) [with _T1 = esphome::DelayAction<const ArduinoJson::JsonObject&>*; _T2 = ArduinoJson::JsonObject]' In file included from /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/functional:55:0, from /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:66, from /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/algorithm:62, from /root/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/Arduino.h:255, from /data/lightmotor/.piolibdeps/esphome-core/src/esphome/espmath.h:4, from /data/lightmotor/.piolibdeps/esphome-core/src/esphome/automation.h:5, from /data/lightmotor/.piolibdeps/esphome-core/src/esphome/application.h:7, from /data/lightmotor/.piolibdeps/esphome-core/src/esphome.h:1, from src/main.cpp:3: /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/tuple:540:17: note: 'constexpr std::tuple<_T1, _T2>::tuple(const std::tuple<_T1, _T2>&) [with _T1 = esphome::DelayAction<const ArduinoJson::JsonObject&>*; _T2 = ArduinoJson::JsonObject]' is implicitly deleted because the default definition would be ill-formed: constexpr tuple(const tuple&) = default; ^ /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/tuple:540:17: error: use of deleted function 'constexpr std::_Tuple_impl<_Idx, _Head, _Tail ...>::_Tuple_impl(const std::_Tuple_impl<_Idx, _Head, _Tail ...>&) [with unsigned int _Idx = 0u; _Head = esphome::DelayAction<const ArduinoJson::JsonObject&>*; _Tail = {ArduinoJson::JsonObject}]' /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/tuple:264:17: note: 'constexpr std::_Tuple_impl<_Idx, _Head, _Tail ...>::_Tuple_impl(const std::_Tuple_impl<_Idx, _Head, _Tail ...>&) [with unsigned int _Idx = 0u; _Head = esphome::DelayAction<const ArduinoJson::JsonObject&>*; _Tail = {ArduinoJson::JsonObject}]' is implicitly deleted because the default definition would be ill-formed: constexpr _Tuple_impl(const _Tuple_impl&) = default; ^ /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/tuple:264:17: error: use of deleted function 'constexpr std::_Tuple_impl<_Idx, _Head, _Tail ...>::_Tuple_impl(const std::_Tuple_impl<_Idx, _Head, _Tail ...>&) [with unsigned int _Idx = 1u; _Head = ArduinoJson::JsonObject; _Tail = {}]' /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/tuple:264:17: note: 'constexpr std::_Tuple_impl<_Idx, _Head, _Tail ...>::_Tuple_impl(const std::_Tuple_impl<_Idx, _Head, _Tail ...>&) [with unsigned int _Idx = 1u; _Head = ArduinoJson::JsonObject; _Tail = {}]' is implicitly deleted because the default definition would be ill-formed: /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/tuple:264:17: error: use of deleted function 'std::_Head_base<1u, ArduinoJson::JsonObject, false>::_Head_base(const std::_Head_base<1u, ArduinoJson::JsonObject, false>&)' /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/tuple:128:12: note: 'std::_Head_base<1u, ArduinoJson::JsonObject, false>::_Head_base(const std::_Head_base<1u, ArduinoJson::JsonObject, false>&)' is implicitly deleted because the default definition would be ill-formed: struct _Head_base<_Idx, _Head, false> ^ /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/tuple:128:12: error: use of deleted function 'ArduinoJson::JsonObject::JsonObject(const ArduinoJson::JsonObject&)' In file included from /data/lightmotor/.piolibdeps/ArduinoJson-esphomelib_ID3837/src/ArduinoJson.hpp:9:0, from /data/lightmotor/.piolibdeps/ArduinoJson-esphomelib_ID3837/src/ArduinoJson.h:9, from /data/lightmotor/.piolibdeps/esphome-core/src/esphome/helpers.h:9, from /data/lightmotor/.piolibdeps/esphome-core/src/esphome/component.h:7, from /data/lightmotor/.piolibdeps/esphome-core/src/esphome/automation.h:6, from /data/lightmotor/.piolibdeps/esphome-core/src/esphome/application.h:7, from /data/lightmotor/.piolibdeps/esphome-core/src/esphome.h:1, from src/main.cpp:3: /data/lightmotor/.piolibdeps/ArduinoJson-esphomelib_ID3837/src/ArduinoJson/JsonObject.hpp:40:7: note: 'ArduinoJson::JsonObject::JsonObject(const ArduinoJson::JsonObject&)' is implicitly deleted because the default definition would be ill-formed: class JsonObject : public Internals::JsonPrintable<JsonObject>, ^ In file included from /data/lightmotor/.piolibdeps/ArduinoJson-esphomelib_ID3837/src/ArduinoJson/Deserialization/../JsonBuffer.hpp:11:0, from /data/lightmotor/.piolibdeps/ArduinoJson-esphomelib_ID3837/src/ArduinoJson/Deserialization/JsonParser.hpp:7, from /data/lightmotor/.piolibdeps/ArduinoJson-esphomelib_ID3837/src/ArduinoJson/JsonBufferBase.hpp:7, from /data/lightmotor/.piolibdeps/ArduinoJson-esphomelib_ID3837/src/ArduinoJson/DynamicJsonBuffer.hpp:7, from /data/lightmotor/.piolibdeps/ArduinoJson-esphomelib_ID3837/src/ArduinoJson.hpp:7, from /data/lightmotor/.piolibdeps/ArduinoJson-esphomelib_ID3837/src/ArduinoJson.h:9, from /data/lightmotor/.piolibdeps/esphome-core/src/esphome/helpers.h:9, from /data/lightmotor/.piolibdeps/esphome-core/src/esphome/component.h:7, from /data/lightmotor/.piolibdeps/esphome-core/src [... cut ...] nst ArduinoJson::JsonObject&}]' src/main.cpp:67:1: required from here /root/.platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/tuple:134:25: error: use of deleted function 'ArduinoJson::JsonObject::JsonObject(const ArduinoJson::JsonObject&)' : _M_head_impl(__h) { } ^ *** [/data/lightmotor/.pioenvs/lightmotor/src/main.cpp.o] Error 1 ========================== [ERROR] Took 8.92 seconds ==========================
```

**Additional information and things you've tried:**

thamks

Johboh, 2019-05-25:
In the docs it states that `delay` is not supported https://esphome.io/components/mqtt.html?highlight=delay
`Due to the way this trigger works internally it is incompatible with certain actions and will trigger a compile failure. For example with the delay action.`

nirkons, 2019-05-25:
Ah I must have missed that, then I guess this should be moved to feature request:)

OttoWinter, 2019-05-25:
Yes the warning is correct, delay cannot be used with JSON messages - and it will stay that way for complicated reasons (JsonObjects are references to a global buffer that cannot be passed around). The workaround would be to store every needed value into global variables and then start a script that uses the global vars.

I will close this as the issue is a won't fix/documented.

closed 2019-05-25 by OttoWinter