esphome/issues #327: Counter overflow in pulse component

J4nsen, 2019-05-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Ubuntu, pip, esphome-core version 1.12.2 compiled on May 19 2019, 19:07:32

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32, Wemos Lolin32

**Affected component:**

https://esphome.io/components/sensor/pulse_counter.html

**Description of problem:**
I need to track a high pulse rate (>1kHz). I think the counter (short int) of the ESP32 overflows. 

Would it be possible to use the [max value interrupt](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/pcnt.html#using-interrupts) to add (overflow count)*SHORT_MAX to the result?

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: pulse_counter
    pin: 25
    name: "Wassertonne Pulse Counter"
    update_interval: 60s
    internal_filter: 1us
    count_mode:
            rising_edge: DISABLE
            falling_edge: INCREMENT
    filters:
    - multiply: 0.01666666666
```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
I've measured the pulse count with my oscilloscope: ~2.5 to 2.7 kHz.

Reported pulses with an update interval of 1s:
```
[19:06:56][D][sensor.pulse_counter:178]: 'Wassertonne Pulse Counter': Retrieved counter: 163200.00 pulses/min
[19:06:56][D][sensor.sensor:100]: 'Wassertonne Pulse Counter': Sending state 2720.05420 pulses/min with 2 decimals of accuracy
```

Reported pulses with an update interval of 60s:
```
[19:17:47][D][sensor.pulse_counter:178]: 'Wassertonne Pulse Counter': Retrieved counter: 31785.00 pulses/min
[19:17:47][D][sensor.sensor:100]: 'Wassertonne Pulse Counter': Sending state 529.76056 pulses/min with 2 decimals of accuracy
```

Johboh, 2019-05-26:
So it overflows for longer update intervals, but is OK for the 1s update interval?

Maybe you can use a sliding window to update value every 1s, but report only every 60s:
```yaml
sensor:
  - platform: pulse_counter
    pin: 25
    name: "Wassertonne Pulse Counter"
    update_interval: 1s
    internal_filter: 1us
    count_mode:
            rising_edge: DISABLE
            falling_edge: INCREMENT
    filters:
    - multiply: 0.01666666666
    - sliding_window_moving_average:
            window_size: 60
            send_every: 60
```

OttoWinter, 2019-05-26:
Exactly, the counter is read out with the given update interval, and on the ESP32 the pulse counter module has a fixed width of 16-bit signed int.

Fix is using a shorter update_interval as @Johboh suggested. Optionally with an average filter to recreate the longer update interval.

Closing as this is a limitation of the hardware and workaround exists.

closed 2019-05-26 by OttoWinter

J4nsen, 2019-05-29:
> So it overflows for longer update intervals, but is OK for the 1s update interval?
> 
> Maybe you can use a sliding window to update value every 1s, but report only every 60s:

 Thanks, this workaround works fine.