esphome/issues #328: on_multi_click doesn't work

pavolholes, 2019-05-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Affected component:**

https://beta.esphome.io/components/binary_sensor/index.html#on-multi-click

**Description of problem:**
I just updated my ESP8266 with latest DEV version and the `on_multi_click` Automation doesn't work anymore. I simplified it to basic single click for testing purposes and it doesn't work. 

`esphome version 1.13.0-dev compiled on May 20 2019, 09:42:53`

**Problem-relevant YAML-configuration entries:**
```yaml
binary_sensor:
  - platform: gpio
    id: 'buttonWallRgb'
    name: '$upperDeviceName RGB Wall Button'
    pin:
      number: D2 #GPIO4
      inverted: True
      mode: INPUT_PULLUP
    on_multi_click:
      # --- Single short click begin
      - timing:
          - ON for at most 0.5s
          - OFF for at least 0.5s
        then:
          - logger.log: "[Light Wall Button RGB] Single Short Clicked"
          - light.toggle: lightRgb
      # --- Single short click end
```

**Traceback (if applicable):**

```
[11:04:15][I][app:063]: esphome version 1.13.0-dev compiled on May 20 2019, 09:42:53
...
[11:04:15][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Livingroom RGB Wall Button'
[11:04:15][C][gpio.binary_sensor:016]:   Pin: GPIO4 (Mode: INPUT_PULLUP, INVERTED)
...
[11:04:15][C][logger:137]: Logger:
[11:04:15][C][logger:138]:   Level: VERBOSE
[11:04:15][C][logger:139]:   Log Baud Rate: 115200
[11:04:15][C][logger:140]:   Hardware UART: UART0
...
[11:16:55][D][binary_sensor:033]: 'Livingroom RGB Wall Button': Sending state ON
[11:16:55][D][binary_sensor:033]: 'Livingroom RGB Wall Button': Sending state OFF
```

**Additional information and things you've tried:**
When I change it to `on_click` Automation it works fine.

pavolholes, 2019-05-23:
I just installed stable version and the `on_multi_click` work fine.
```esphome-core version 1.12.2 compiled on May 23 2019, 08:53:03```

```
[08:59:29][D][binary_sensor:037]: 'Livingroom RGB Wall Button': Sending state ON
[08:59:29][V][binary_sensor:168]: START min=0 max=500
[08:59:29][V][binary_sensor:169]: Multi Click: Starting multi click action!
[08:59:29][V][binary_sensor:229]: Multi Click: You can now RELEASE the button.
[08:59:29][D][binary_sensor:037]: 'Livingroom RGB Wall Button': Sending state OFF
[08:59:29][V][binary_sensor:205]: C i=1 min=500
[08:59:29][V][binary_sensor:241]: Multi Click: Hooray, multi click is valid. Triggering!
[08:59:29][D][main:090]: [Light Wall Button RGB] Single Short Clicked
[08:59:29][D][light.state:418]: 'Livingroom RGB light' Turning OFF
[08:59:29][D][light.state:459]:   Transition Length: 1000 ms
```

OttoWinter, 2019-05-24:
Fixed in https://github.com/esphome/esphome/commit/999c1a5357196b88b7c57ea581b36f72fa96f746

closed 2019-05-24 by OttoWinter