esphome/issues #330: Wemos D1 Mini with DS18B20 no sensors found

JBS5, 2019-05-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
ESPHome 1.12.2 add-on on Hass.io in Docker on a NUC running Ubuntu.

**ESP (ESP32/ESP8266, Board/Sonoff):**
Wemos D1 Mini

**Affected component:**
dallas

**Description of problem:**
On a Wemos D1 Mini v3.1 I'm trying to get a DS18B20 to work following this diagram (not a Wemos D1 on the picture, I know :-))

[image]

Black on Wemos GND
Red on Wemos 3V3
Yellow on Wemos D2
4.7K resistor as on the picture

Any clue what I am missing?

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: projectname
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: 'xxxx'
  password: 'xxxx'
  manual_ip:
# Set this to the IP of the ESP
    static_ip: 192.168.178.229
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.178.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0  

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:

ota:

dallas:
  - pin: GPIO4

```

**Traceback (if applicable):**
Log:
[14:40:43][C][sensor.dallas:081]: DallasComponent:
[14:40:43][C][sensor.dallas:082]:   Pin: GPIO4 (Mode: OUTPUT)
[14:40:43](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[14:40:43][W][sensor.dallas:086]:   Found no sensors!
```

```

**Additional information and things you've tried:**
I have tried some other pins for the yellow data wire.

OttoWinter, 2019-05-24:
Have seen this before, but I think every time it was either a wrong connection or faulty sensor - the dallas integration has been working quite well for a longer period of time, so I don't think it's an issue there.

I'd try with another sensor.

sandos, 2019-05-28:
I had some trouble with certain wemos d1 minis and dalls, but that manifested itself with watchdog reboot loops. A small test program in arduino worked on the same hw with dallas.

Also, at one point, chaning the logging level fixed my problem! I suspect its something of a timing problem related to power draw somehow. If I unplugged the sensor during times near wifi startup and then re-connected it, it started working better, but not good, ie. it still crashed later on randomly.

JBS5, 2019-06-17:
Just tried another Dallas sensor on another Wemos D1 Mini. 
Same issue...

lubbertkramer, 2019-06-17:
Same setup on a D1 mini Pro

esphome:
  name: zwembadd1
  platform: ESP8266
  board: d1_mini_pro

wifi:
  ssid: "Wifi"
  password: "Wifipassword"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "password"

ota:
  password: "password"

# Example configuration entry
dallas:
  - pin: GPIO13

# Individual sensors
sensor:
  - platform: dallas
    index: 0
    name: "Pool"

[image]

I followed this blog -> https://neon.ninja/2017/10/smarter-swimming-pool-2-water-temperature/

JBS5, 2019-06-17:
Had no luck, unfortunately.

In the logging I see now (when using esphomeflasher on windows to flash the image)

'[E][dallas.sensor:123]: Requesting conversion failed'

JBS5, 2019-06-19:
Tried again without the resistor.
Connected the Yellow wire to D7 (GPIO13) and the sensor is detected for several times. But now, after trying some other pins. D7 is not working anymore, but D2 is.

Kind of strange

brandond, 2019-06-19:
It looks like you're powering the sensor via the 3.3v output on the D1 Mini? Note that the Dallas sensors draw 1.5 mA at 3.0 to 5.5 volts on Vdd when taking a measurement, and the built-in voltage regulators on these development boards are frequently underpowered and cannot supply this without browning out.

JBS5, 2019-06-19:
@brandond Thanks for your reply. What kind of external power supply or capacitor do you suggest?

I have almost no knowledge of electrical engineering so just wondering: although no external power supply or capacitor is mentioned in all connection diagrams, can this be a problem?

brandond, 2019-06-19:
It really depends on what's on your board. The official Wemos appears to use a [TLV1117](http://www.ti.com/lit/ds/symlink/tlv1117.pdf) which is rated at up to 800mA, which is more than enough to power the esp8266 at about 400mA when transmitting, plus a few extra peripherals. Knockoff boards will frequently include different or fake voltage regulators that may not work as well.

Another thing to try would be to simply connect the Vdd directly to the 5v pin and bypass the voltage regulator entirely.

OttoWinter, 2019-07-11:
Closing because this is very likely a hardware connection problem (due to the fact that it only appears on certain pins like @JBS5 confirmed). Could be power supply issue, not a good solder connection etc.

closed 2019-07-11 by OttoWinter