esphome/issues #333: Travis build error: ha_hello_world

T3m3z, 2019-05-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
-

**ESP (ESP32/ESP8266, Board/Sonoff):**
-

**Affected component:**
-

**Description of problem:**
I was creating a pull request (esphome/esphome#554) for AM2320 component. During the Travis checks there were errors stating:
`ValueError: Component ID ha_hello_world was not declared to inherit from Component, or was registered twice. Please create a bug report with your configuration.`

Errors happen both with test2.yaml and test3.yaml even though I just edited test3.yaml.

[Direct link](https://travis-ci.org/esphome/esphome/builds/536349420) to travis build 1741.

**Problem-relevant YAML-configuration entries:**
```yaml

```

**Traceback (if applicable):**

```
Worker information
hostname: 0fc7f5f1-bed2-4242-814f-557f76203caf@1.production-2-worker-org-gce-2zlz
version: v6.2.0 https://github.com/travis-ci/worker/tree/5e5476e01646095f48eec13196fdb3faf8f5cbf7
instance: travis-job-54392ee2-af59-48fd-aa5b-4076efe5d057 travis-ci-garnet-trusty-1512502259-986baf0 (via amqp)
startup: 6.779552266s
system_info
Build system information
Build language: python
Build group: stable
Build dist: trusty
Build id: 536349420
Job id: 536349423
Runtime kernel version: 4.4.0-101-generic
travis-build version: 1ea7e2622
Build image provisioning date and time
Tue Dec  5 19:58:13 UTC 2017
Operating System Details
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.5 LTS
Release:	14.04
Codename:	trusty
Cookbooks Version
7c2c6a6 https://github.com/travis-ci/travis-cookbooks/tree/7c2c6a6
git version
git version 2.15.1
bash version
GNU bash, version 4.3.11(1)-release (x86_64-pc-linux-gnu)
gcc version
gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
docker version
Client:
 Version:      17.09.0-ce
 API version:  1.32
 Go version:   go1.8.3
 Git commit:   afdb6d4
 Built:        Tue Sep 26 22:42:38 2017
 OS/Arch:      linux/amd64
Server:
 Version:      17.09.0-ce
 API version:  1.32 (minimum version 1.12)
 Go version:   go1.8.3
 Git commit:   afdb6d4
[... 987 lines cut ...]
  File "/home/travis/build/esphome/esphome/esphome/core.py", line 421, in _wrapper_generator
    for x in var:
  File "/home/travis/build/esphome/esphome/esphome/core.py", line 417, in _wrapper_generator
    var = gen.send(var)
  File "/home/travis/build/esphome/esphome/esphome/cpp_helpers.py", line 42, in register_component
    u"configuration.".format(id_))
ValueError: Component ID ha_hello_world was not declared to inherit from Component, or was registered twice. Please create a bug report with your configuration.
The command "esphome tests/test3.yaml compile" exited with 1.
cache.2
store build cache
0.00s36.02schange detected (content changed, file is created, or file is deleted):
/home/travis/.platformio/.cache/db.data
changes detected, packing new archive
.
.
.
.
.
.
uploading PR.554/cache-linux-trusty-037f70e24431b2e805dbcf061b149514c4e393d5eaff743b1e8b8d8b0dc19721--python-2.7.tgz
cache uploaded
Done. Your build exited with 1.
```

**Additional information and things you've tried:**

mtl010957, 2019-05-23:
I'm seeing the same problem with a completely different branch. It seems the new component registration checks are catching bugs that were there all along but not noticed.

OttoWinter, 2019-05-24:
Fixed in dev

closed 2019-05-24 by OttoWinter