esphome/issues #334: Not waking from Deep Sleep

eflemingtra, 2019-05-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassbian

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 / D1mini

**Affected component:**

Ultrasonic Sensor

**Description of problem:**

I'm running an ultrasonic sensor to measure my water tank's level. While at the moment it is powered by USB, it will eventaully be powered by solar and battery, hence the deep_sleep component being used. It is not however waking from this state. I note on the logs that no matter what I put down for the sleep_duration, it shows as `Sleep Duration: lu ms`. Full logs below 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: water_tank_sensor
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: ****
  password: ****

# Enable logging
logger:

mqtt:
  broker: 192.168.1.39
  username: ****
  password: ****
  birth_message:
    topic: water_tank_level_sensor/state
    payload: 
  will_message:
    topic: water_tank_level_sensor/state
    payload: 

ota:

sensor:
  - platform: ultrasonic
    trigger_pin: D1
    echo_pin: D2
    name: "Water Tank Level Sensor"
    filters:
      - offset: -0.15
      - lambda: return (1.0 - (x/2.0)) * 100.0;
    unit_of_measurement: "%"

deep_sleep:
  run_duration: '0:01'
  sleep_duration: '0:05'

```

**Traceback (if applicable):**

```
[10:31:30][C][wifi:253]:   SSID: [redacted]
[10:31:30][C][wifi:254]:   IP Address: 192.168.1.55
[10:31:30][C][wifi:256]:   BSSID: [redacted]
[10:31:30][C][wifi:257]:   Hostname: 'water_tank_sensor'
[10:31:30][C][wifi:261]:   Signal strength: -66 dB ▂▄▆█
[10:31:30][C][wifi:262]:   Channel: 6
[10:31:30][C][wifi:263]:   Subnet: 255.255.255.0
[10:31:30][C][wifi:264]:   Gateway: 192.168.1.1
[10:31:30][C][wifi:265]:   DNS1: 192.168.1.1
[10:31:30][C][wifi:266]:   DNS2: 0.0.0.0
[10:31:30][C][logger:142]: Logger:
[10:31:30][C][logger:143]:   Level: DEBUG
[10:31:30][C][logger:144]:   Log Baud Rate: 115200
[10:31:30][C][logger:145]:   Hardware UART: UART0
[10:31:30][C][sensor.ultrasonic:045]: Ultrasonic Sensor 'Water Tank Level Sensor'
[10:31:30][C][sensor.ultrasonic:045]:   Unit of Measurement: '%'
[10:31:30][C][sensor.ultrasonic:045]:   Accuracy Decimals: 2
[10:31:30][C][sensor.ultrasonic:045]:   Icon: 'mdi:arrow-expand-vertical'
[10:31:30][C][sensor.ultrasonic:046]:   Echo Pin: GPIO4 (Mode: INPUT)
[10:31:30][C][sensor.ultrasonic:047]:   Trigger Pin: GPIO5 (Mode: OUTPUT)
[10:31:30][C][sensor.ultrasonic:048]:   Pulse time: 10 µs
[10:31:30][C][sensor.ultrasonic:049]:   Timeout: 11662 µs
[10:31:30][C][sensor.ultrasonic:050]:   Update Interval: 60000 ms
[10:31:30][C][ota:127]: Over-The-Air Updates:
[10:31:30][C][ota:128]:   Address: water_tank_sensor.local:8266
[10:31:30][C][mqtt.client:062]: MQTT:
[10:31:30][C][mqtt.client:064]:   Server Address: 192.168.1.39:1883 (192.168.1.39)
[10:31:30][C][mqtt.client:065]:   Username: [redacted]
[10:31:30][C][mqtt.client:066]:   Client ID: [redacted]
[10:31:30][C][mqtt.client:068]:   Discovery prefix: 'homeassistant'
[10:31:30][C][mqtt.client:069]:   Discovery retain: YES
[10:31:30][C][mqtt.client:071]:   Topic Prefix: 'water_tank_sensor'
[10:31:30][C][mqtt.client:073]:   Log Topic: 'water_tank_sensor/debug'
[10:31:30][C][mqtt.client:076]:   Availability: 'water_tank_level_sensor/state'
[10:31:30][C][sensor.mqtt:025]: MQTT Sensor 'Water Tank Level Sensor':
[10:31:30][C][sensor.mqtt:029]:   State Topic: 'water_tank_sensor/sensor/water_tank_level_sensor/state'
[10:31:31][C][deep_sleep:025]: Setting up Deep Sleep...
[10:31:31][C][deep_sleep:027]:   Sleep Duration: lu ms
[10:31:31][C][deep_sleep:030]:   Run Duration: 60000 ms
[10:31:31][I][application:109]: Running through first loop()
[10:31:31][I][application:136]: First loop finished successfully!
[10:32:10][D][sensor.sensor:100]: 'Water Tank Level Sensor': Sending state 79.33112 % with 2 decimals of accuracy
[10:32:10][D][sensor.ultrasonic:040]: 'Water Tank Level Sensor' - Got distance: 0.56 m
[10:32:10][D][sensor.sensor:100]: 'Water Tank Level Sensor': Sending state 79.33112 % with 2 decimals of accuracy
[10:32:30][I][deep_sleep:072]: Beginning Deep Sleep
```

**Additional information and things you've tried:**

dr1rrb, 2019-05-24:
It's normal that your devices appears as unavailable in HA, it's due to the availability topic (birth and will message of MQTT, cf. https://esphome.io/components/mqtt.html).

However I'm facing the same issue with a ESP01 and a DHT11: goes in deep sleep and never wakes up, with the same strange `Sleep duration: 1u ms` in logs... If anybody has an idea?

r-jordan, 2019-05-24:
Hi, you have to connect D0 pin with the RST pin for the ESP to be waken up.

Have a good day playing with ESPhome! ;-D

middelink, 2019-05-24:
@eflemingtra: I think you're setting the times wrong. (https://esphome.io/components/deep_sleep.html)
 Try
  ```
deep_sleep:
   run_duration: 1m
   sleep_duration: 5min
```
and since it is not waking up, I guess the same answer below applies here as well.

@dr1rrb: For deepsleep to work on an esp01 (esp8266), you need to connect gpio 16 to reset. Virtually impossible to do as the pin is not brought out.
(its documented in 3rd pagaraph in https://esphome.io/components/deep_sleep.html)

eflemingtra, 2019-05-24:
> @dr1rrb: For deepsleep to work on an esp01 (esp8266), you need to connect gpio 16 to reset. Virtually impossible to do as the pin is not brought out.

Ahh, okay, I missed that. So physically connect the two pins? Any additional code required or is that part of the ESP core?

r-jordan, 2019-05-24:
Yes, connect the two pins GPIO16 (D0) and RST with a wire and it's all. You will have to remove the wire if you want to reflash the device throught USB so it's better to do it with a jumper or a short jump wire.

eflemingtra, 2019-05-24:
Thanks for the replies, guys.

A big help - i'm off to try it now :)

eflemingtra, 2019-05-24:
> Yes, connect the two pins GPIO16 (D0) and RST with a wire and it's all. You will have to remove the wire if you want to reflash the device throught USB so it's better to do it with a jumper or a short jump wire.

Now, can I push things and ask another question? :)

I thought I had the unavailability in HA sorted with how I've got blank will and birth payloads. I believe my topic is correct as per the logs:

`Availability: 'water_tank_level_sensor/state'`

Which coincides with what I have in my yaml:

```
mqtt:
  broker: 192.168.1.39
  username: ****
  password: ****
  birth_message:
    topic: water_tank_level_sensor/state
    payload: 
  will_message:
    topic: water_tank_level_sensor/state
    payload: 

```
Thoughts?

dr1rrb, 2019-05-24:
To disable the availability, you only have to publish `online` in all cases :)

```yaml
mqtt:
  broker: 192.168.1.39
  username: ****
  password: ****
  birth_message:
    topic: water_tank_level_sensor/state
    payload: online
  will_message:
    topic: water_tank_level_sensor/state
    payload: online
```

OttoWinter, 2019-05-24:
Can this issue be closed? The initial issue (wake up not working looks to be fixed/user error).

eflemingtra, 2019-05-24:
Thanks for all your help!

closed 2019-05-24 by eflemingtra