esphome/issues #338: LEDC

pavolholes, 2019-05-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**Affected component:**

https://beta.esphome.io/components/output/ledc.html

**Description of problem:**
`esphome version 1.13.0-dev compiled on May 25 2019, 00:06:18`
LEDC Output at ESP32 doesn't work at all, the light can't be turned on. I thing the problem might be in frequency which is initially set to 0 Hz and can't be changed via yaml.

**Problem-relevant YAML-configuration entries:**
```yaml
output:
  # Enable one pin as outputs with LEDC PWM output of the ESP32 for the main white light.
  - platform: ledc
    pin: GPIO23
    frequency: 1000 Hz #bug note: this doesn't have any effect if it's set or not
    id: 'outputWhite'
```

**Traceback (if applicable):**

```
[00:15:06][I][app:063]: esphome version 1.13.0-dev compiled on May 25 2019, 00:06:18
...
[00:15:06][C][ledc.output:032]: LEDC Output:
[00:15:06][C][ledc.output:033]:   PinGPIO23 (Mode: OUTPUT)
[00:15:06][C][ledc.output:034]:   LEDC Channel: 3
[00:15:06][C][ledc.output:035]:   Frequency: 0.0 Hz
[00:15:06][C][ledc.output:036]:   Bit Depth: 0
...
[00:15:06][C][logger:137]: Logger:
[00:15:06][C][logger:138]:   Level: VERBOSE
[00:15:06][C][logger:139]:   Log Baud Rate: 115200
[00:15:06][C][logger:140]:   Hardware UART: UART0

```

**Additional information and things you've tried:**
It works at `esphome-core version 1.12.2` without any problems

```
[00:51:36][I][application:097]: esphome-core version 1.12.2 compiled on May 25 2019, 00:45:53
...
[00:51:36][C][output.ledc:032]: LEDC Output:
[00:51:36][C][output.ledc:033]:   Pin: 23
[00:51:36][C][output.ledc:034]:   LEDC Channel: 4
[00:51:36][C][output.ledc:035]:   Frequency: 1000.0 Hz
[00:51:36][C][output.ledc:036]:   Bit Depth: 12
```

pavolholes, 2019-05-24:
this issue might be related: https://github.com/esphome/issues/issues/258

OttoWinter, 2019-05-25:
Whoops, thanks. Somehow that must have been lost in the refactor.

closed 2019-05-25 by OttoWinter

davericher, 2019-05-26:
This was the cause of my issue posted last month