esphome/issues #339: Reboot using a simple user-defined service

diraimondo, 2019-05-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on RPi for HA 0.93.2

**ESP (ESP32/ESP8266, Board/Sonoff):**

nodemcu esp8266-12E lolin v3

**Affected component:**

Maybe https://esphome.io/components/api.html or basic automation building blocks.

**Description of problem:**
I'm trying to use a ESP8266 to send infrared commands to my AC unit. I had success using the transmitter component in esphome. Now I want to expose such commands to HA by means of user-defined services (https://esphome.io/components/api.html#user-defined-services). I started to face reboots of the unit upon service requests. I started to strip off all the not-involved components so now I got a simple user-defined service (with some nested if actions and comparisons on service variables) that reboot upon some specific HA requests. Maybe I'm doing something wrong with strings but it could be an esphome bug. I hope you are able to reproduce the problem.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp_test
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "hassio"
  password: "XXXXXXXXX"
  fast_connect: true

logger:
  level: "VERBOSE"

#api:
#  password: 'XXXXXXXXX'

ota:
  password: 'XXXXXXXXX'

api:
  password: 'XXXXXXXXX'
  services:
    - service: ac_set_state
      variables:
        op_mode: string
        temp: int
        fan_mode: string
        swing_mode: string
      then:
        - logger.log: "Setting up AC..."
        - if:
            condition:
              lambda: 'return op_mode.compare("cool") == 0;'
            then:
              - logger.log: "COOL"
              - if:
                  condition:
                    lambda: 'return fan_mode.compare("auto") == 0;'
                  then:
                    - if:
                        condition:
                          lambda: 'return temp == 16;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_auto_swing_manual_temp_16"
              - if:
                  condition:
                    lambda: 'return fan_mode.compare("low") == 0;'
                  then:
                    - if:
                        condition:
                          lambda: 'return temp == 16;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_16"
                    - if:
                        condition:
                          lambda: 'return temp == 17;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_17"
              - if:
                  condition:
                    lambda: 'return fan_mode.compare("medium") == 0;'
                  then:
                    - if:
                        condition:
                          lambda: 'return temp == 16;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_16"
                    - if:
                        condition:
                          lambda: 'return temp == 17;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_17"
                    - if:
                        condition:
                          lambda: 'return temp == 18;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_16"
                    - if:
                        condition:
                          lambda: 'return temp == 19;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_17"
              - if:
                  condition:
                    lambda: 'return fan_mode.compare("high") == 0;'
                  then:
                    - if:
                        condition:
                          lambda: 'return temp == 16;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_16"
                    - if:
                        condition:
                          lambda: 'return temp == 17;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_17"
                    - if:
                        condition:
                          lambda: 'return temp == 18;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_16"
                    - if:
                        condition:
                          lambda: 'return temp == 19;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_17"
                    - if:
                        condition:
                          lambda: 'return temp == 20;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_16"
                    - if:
                        condition:
                          lambda: 'return temp == 21;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_17"
                    - if:
                        condition:
                          lambda: 'return temp == 22;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_16"
                    - if:
                        condition:
                          lambda: 'return temp == 23;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_17"
                    - if:
                        condition:
                          lambda: 'return temp == 24;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_16"
                    - if:
                        condition:
                          lambda: 'return temp == 25;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_17"
                    - if:
                        condition:
                          lambda: 'return temp == 26;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_16"
                    - if:
                        condition:
                          lambda: 'return temp == 27;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_17"
                    - if:
                        condition:
                          lambda: 'return temp == 28;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_17"
                    - if:
                        condition:
                          lambda: 'return temp == 29;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_17"
                    - if:
                        condition:
                          lambda: 'return temp == 30;'
                        then:
                          - logger.log: "switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_17"

        - if:
            condition:
              lambda: 'return op_mode.compare("off") == 0;'
            then:
              - logger.log: "OFF"
              - logger.log: "switch.turn_on: ac_mode_off"

time:
  - platform: homeassistant
    id: homeassistant_time
```

**Traceback (if applicable):**

```
```
The compilation goes fine. On the output (network or serial) I can't spot anything of interest.

**Additional information and things you've tried:**
From HA I can invoke (multiple time) the service without any problem using the following params:
- `{ "op_mode": "cool", "temp": 16, "fan_mode": "medium", "swing_mode": "manual"}`
- `{ "op_mode": "off", "temp": 16, "fan_mode": "medium", "swing_mode": "manual"}`

I get a reboot on something related the long if statements queue in "fan_mode = high":
- `{ "op_mode": "cool", "temp": 16, "fan_mode": "high", "swing_mode": "manual"}`

It could be related to the string variables and their comparisons. I did a quick test replacing all with integer variables and integer comparisons: no reboots.

OttoWinter, 2019-05-25:
Please attach the device via USB and post the stacktrace you get when getting the error.

My first guess would be that the issue is the stack size on the ESP (something like `Stack canary watchpoint triggered`) - that would be because the automation is a bit too long.

Please check if it's that - I don't think a fix would be easy (some big core changes would be needed). So let's first see if that's actually it.

diraimondo, 2019-05-25:
Sorry but I can't see any stacktrace on the serial terminal on reboot:
```
[D][main:031]: Setting up AC...
[D][main:038]: COOL
[D][main:062]: switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_16 <== SAFE PARAMS
[D][main:031]: Setting up AC...
[D][main:038]: COOL
[D][main:120]: switch.turn_on: ac_mode_cool_fan_low_swing_manual_temp_16 <== CRITICAL REQUEST

 ets Jan  8 2013,rst cause:4, boot mode:(3,2)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

```
I've tried to use "VERY_VERBOSE" level for logging but the resulting firmware doesn't boot.

Any suggestion for a smarter way to manage with one (or few) user-defined service the MANY switches related to the AC control?

diraimondo, 2019-05-25:
Looking for a work-around in the meanwhile: it looks I could expose a user-defined service that get as string the id of switch (one of the many) and it should trigger the 'turn_on' action on such switch. Is it possible? I can't get the right syntax: could you help?

diraimondo, 2019-05-25:
I was able to create the single user-defined service to access all the switches but now the firmware doesn't boot, I guess, because of too many remote_transmitter switches...

This is the complete yaml file: https://pastebin.com/3jHP8HRQ

This is what I get on the serial connection: 
```
Exception (29):
epc1=0x4022048d epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont 
sp: 3ffff8f0 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffa90:  00000028 3ffffb7c 00000035 401004e4  
3ffffaa0:  3fff313e 3fff30bc 00000035 40212c84  
3ffffab0:  3ffffb60 3ffffb64 00000028 00009470  
3ffffac0:  3ffffb70 3ffffb74 00000035 401004e4  
3ffffad0:  3ffffc54 3ffeb5c4 00000035 40212c84  
3ffffae0:  00000093 3ffeb5c4 00000028 40220488  
3ffffaf0:  3fff239c 000012a4 3fffbf4c 402207a9  
3ffffb00:  3ffffb70 3ffffb74 3ffffb70 40220970  
3ffffb10:  3fff239c 0000019e 3ffeb604 40220990  
3ffffb20:  3fffbf8c 3ffffb74 3fffbf3c 40220624  
3ffffb30:  00000000 00000001 3fffbeb4 4020cada  
3ffffb40:  00000093 3ffffb74 3ffeb5c4 40220a76  
3ffffb50:  3ffffb80 3fff812c 3fffbe38 00009470  
3ffffb60:  00000093 3fffbebc 3fffbeb4 4020bc06  
3ffffb70:  3fff23d0 3fffbf98 3fffbf58 3fff23d0  
3ffffb80:  3fffbe1c 3fff36a0 00000028 402203af  
3ffffb90:  3fff7db4 40257180 3fffbeb4 4020bc5a  
3ffffba0:  3fffbe74 3fff3984 3fffbeb4 4020ec33  
3ffffbb0:  12a4124f 5a595857 3fffbeb4 4020e504  
3ffffbc0:  00009470 3fff3a64 3fffbeb4 4020e308  
3ffffbd0:  3ffffc54 3fff3070 00000054 3fffbeb4  
3ffffbe0:  00009470 3fff3a64 3fff20a4 40205abd  
3ffffbf0:  3fff3560 00000000 00000000 3fff3578  
3ffffc00:  00000000 00000000 40217930 00000000  
3ffffc10:  40217930 00000000 40217930 00000000  
3ffffc20:  40217930 00000000 40217930 00000000  
3ffffc30:  00000000 3fffbe80 0000010c 60000318  
3ffffc40:  00001000 3fff3700 00000003 3fff37d0  
3ffffc50:  00000003 3fff23d0 3fff3960 3fff37d0  
3ffffc60:  3fff37b0 3fff3798 3fff3780 feefeffe  
3ffffc70:  feefeffe feefeffe feefeffe feefeffe  
3ffffc80:  3fff3aec 3fff3bdc 3fff3d0c 3fff3e3c  
3ffffc90:  3fff3ed4 3fff4044 3fff4174 3fff420c  
3ffffca0:  3fff437c 3fff44ac 3fff45dc 3fff46cc  
3ffffcb0:  3fff4764 3fff48d4 3fff496c 3fff4adc  
3ffffcc0:  3fff4b74 3fff4d6c 3fff4e5c 3fff4ef4  
3ffffcd0:  3fff5064 3fff50fc 3fff526c 3fff5304  
3ffffce0:  3fff5474 3fff550c 3fff567c 3fff5714  
3ffffcf0:  3fff5884 3fff591c 3fff5a8c 3fff5b24  
3ffffd00:  3fff5c94 3fff4c0c 3fff5cec 3fff60ac  
3ffffd10:  3fff6144 3fff62b4 3fff634c 3fff64bc  
3ffffd20:  3fff6554 3fff66c4 3fff675c 3fff68cc  
3ffffd30:  3fff6964 3fff6ad4 3fff6b6c 3fff6cdc  
3ffffd40:  3fff6d74 3fff6ee4 3fff6f7c 3fff70ec  
3ffffd50:  3fff7184 3fff72f4 3fff738c 3fff74fc  
3ffffd60:  3fff7594 3fff7704 3fff779c 3fff790c  
3ffffd70:  3fff79a4 3fff7b14 3fff7bac 3fff7d1c  
3ffffd80:  3fff7db4 3fff5dc4 3fff5e5c 3fff8334  
3ffffd90:  3fff83cc 3fff853c 3fff85d4 3fff8744  
3ffffda0:  3fff87dc 3fff894c 3fff89e4 3fff8b54  
3ffffdb0:  3fff8bec 3fff8d5c 3fff8df4 3fff8f64  
3ffffdc0:  3fff8ffc 3fff916c 3fff9204 3fff9374  
3ffffdd0:  3fff940c 3fff957c 3fff9614 3fff9784  
3ffffde0:  3fff981c 3fff998c 3fff9a24 3fff9b94  
3ffffdf0:  3fff9c2c 3fff9d9c 3fff9e34 3fff9fa4  
3ffffe00:  3fffa03c 3fffa1ac 3fffa244 3fffa3b4  
3ffffe10:  3fffa44c 3fffa5bc 3fffa654 3fffa7c4  
3ffffe20:  3fffa85c 3fffa9cc 3fffaa64 3fffabd4  
3ffffe30:  3fffac6c 3fffaddc 3fffae74 3fffafe4  
3ffffe40:  3fffb07c 3fffb1ec 3fffb284 3fffb3f4  
3ffffe50:  3fffb48c 3fffb5fc 3fffb694 3fffb804  
3ffffe60:  3fffb89c 3fffba0c 3fffbaa4 3fffbc14  
3ffffe70:  3fffbcac 3fffbe1c feefeffe feefeffe  
3ffffe80:  feefeffe feefeffe feefeffe feefeffe  
3ffffe90:  feefeffe feefeffe feefeffe feefeffe  
3ffffea0:  feefeffe feefeffe feefeffe feefeffe  
3ffffeb0:  feefeffe feefeffe feefeffe feefeffe  
3ffffec0:  feefeffe feefeffe feefeffe feefeffe  
3ffffed0:  feefeffe feefeffe feefeffe feefeffe  
3ffffee0:  feefeffe feefeffe feefeffe feefeffe  
3ffffef0:  feefeffe feefeffe feefeffe feefeffe  
3fffff00:  feefeffe feefeffe feefeffe feefeffe  
3fffff10:  feefeffe feefeffe feefeffe feefeffe  
3fffff20:  feefeffe feefeffe feefeffe feefeffe  
3fffff30:  feefeffe feefeffe feefeffe feefeffe  
3fffff40:  feefeffe feefeffe feefeffe feefeffe  
3fffff50:  feefeffe feefeffe feefeffe feefeffe  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  feefeffe feefeffe 3fff380c 3fff3644  
3fffff90:  00000000 3fff3934 feefeffe feefeffe  
3fffffa0:  feefeffe feefeffe feefeffe 3fff22f8  
3fffffb0:  3fffdad0 00000000 3fff22f0 40212e78  
3fffffc0:  feefeffe feefeffe 3ffe8580 40100721  
<<<stack<<<

last failed alloc call: 3FFFFC54(53)
```

I understand that this is a non-conventional way to interface with an AC (but the SmartIR project do this) but my unit is unknown by major list.

closed 2020-12-06 by stale[bot]