esphome/issues #340: mqtt on_message compilation error when logger added

Johboh, 2019-05-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Windows 10, pip.
`$ python -m esphome test.yaml version`
`Version: 1.12.2`

**ESP (ESP32/ESP8266, Board/Sonoff):**

```
  platform: ESP8266
  board: d1_mini
```

**Affected component:**

https://esphome.io/components/mqtt.html or https://esphome.io/components/logger.html or something else.

**Description of problem:**
Compilation error when `- logger.log: "hello"` is added to the mqtt `on_json_message`. If no action is added or a different action, it compiles and runs.
It only happens when using a custom component.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test
  platform: ESP8266
  board: d1_mini
  includes:
    - BME280Spi.h
  libraries:
    - SPI
    - Wire
    - "Adafruit Unified Sensor"
    - "Adafruit BME280 Library"

wifi:
  ssid: !secret ssid
  password: !secret password

sensor:
  - platform: custom
    lambda: |-
      auto bme280_spi = new BME280Spi(D8);
      App.register_component(bme280_spi);
      return {bme280_spi->temperature_sensor, bme280_spi->humidity_sensor, bme280_spi->pressure_sensor};
    sensors:
      - name: "Temperature"
      - name: "Humidity"
      - name: "Bedroom Pressure"

mqtt:
  broker: !secret mqtt_broker
  port: 1883
  username: !secret mqtt_username
  password: !secret mqtt_password
  on_json_message:
    - topic: bed/calibration
      then:
        - logger.log: "hello"

logger:

ota:
```

**Traceback (if applicable):**

```
In file included from C:\Users\johboh\.platformio\packages\framework-arduinoespressif8266@2.20402.4\cores\esp8266/Arduino.h:259:0,
from .piolibdeps\esphome-core\src/esphome/espmath.h:4,
from .piolibdeps\esphome-core\src/esphome/automation.h:5,
from .piolibdeps\esphome-core\src/esphome/application.h:7,
from .piolibdeps\esphome-core\src/esphome.h:1,
from src\main.cpp:3:
C:\Users\johboh\.platformio\packages\framework-arduinoespressif8266@2.20402.4\cores\esp8266/pgmspace.h:16:51: error: __c causes a section type conflict with __c
#define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))
^
C:\Users\johboh\.platformio\packages\framework-arduinoespressif8266@2.20402.4\cores\esp8266/WString.h:38:76: note: in definition of macro 'FPSTR'
#define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
^
C:\Users\johboh\.platformio\packages\framework-arduinoespressif8266@2.20402.4\cores\esp8266/WString.h:39:34: note: in expansion of macro 'PSTR'
#define F(string_literal) (FPSTR(PSTR(string_literal)))
^
.piolibdeps\esphome-core\src/esphome/log.h:80:3: note: in expansion of macro 'F'
F(ESPHOME_LOG_COLOR_##letter "[" #letter "][%s:%03u]: " format ESPHOME_LOG_RESET_COLOR), tag, __LINE__
^
.piolibdeps\esphome-core\src/esphome/log.h:106:49: note: in expansion of macro 'ESPHOME_LOG_FORMAT'
esp_log_printf_(ESPHOME_LOG_LEVEL_DEBUG, tag, ESPHOME_LOG_FORMAT(tag, D, format), ##__VA_ARGS__)
^
.piolibdeps\esphome-core\src/esphome/log.h:168:28: note: in expansion of macro 'esph_log_d'
#define ESP_LOGD(tag, ...) esph_log_d(tag, __VA_ARGS__)
^
src\..\..\BME280Spi.h:31:5: note: in expansion of macro 'ESP_LOGD'
ESP_LOGD("bme280", "Temperature: %f C, Humidity: %f %%, Pressure: %f",
^
C:\Users\johboh\.platformio\packages\framework-arduinoespressif8266@2.20402.4\cores\esp8266/pgmspace.h:16:51: note: '__c' was declared here
#define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))
^
C:\Users\johboh\.platformio\packages\framework-arduinoespressif8266@2.20402.4\cores\esp8266/WString.h:38:76: note: in definition of macro 'FPSTR'
#define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))
^
C:\Users\johboh\.platformio\packages\framework-arduinoespressif8266@2.20402.4\cores\esp8266/WString.h:39:34: note: in expansion of macro 'PSTR'
#define F(string_literal) (FPSTR(PSTR(string_literal)))
^
.piolibdeps\esphome-core\src/esphome/log.h:80:3: note: in expansion of macro 'F'
F(ESPHOME_LOG_COLOR_##letter "[" #letter "][%s:%03u]: " format ESPHOME_LOG_RESET_COLOR), tag, __LINE__
^
.piolibdeps\esphome-core\src/esphome/log.h:106:49: note: in expansion of macro 'ESPHOME_LOG_FORMAT'
esp_log_printf_(ESPHOME_LOG_LEVEL_DEBUG, tag, ESPHOME_LOG_FORMAT(tag, D, format), ##__VA_ARGS__)
^
.piolibdeps\esphome-core\src/esphome/log.h:168:28: note: in expansion of macro 'esph_log_d'
#define ESP_LOGD(tag, ...) esph_log_d(tag, __VA_ARGS__)
^
src\main.cpp:40:7: note: in expansion of macro 'ESP_LOGD'
ESP_LOGD("main", "hello");
```
And the content of the custom sensor `BME280Spi.h`
```cpp
#include "esphome.h"
#include "Adafruit_BME280.h"
using namespace esphome;

class BME280Spi : public PollingComponent {
 public:
  sensor::Sensor *temperature_sensor = new sensor::Sensor();
  sensor::Sensor *humidity_sensor = new sensor::Sensor();
  sensor::Sensor *pressure_sensor = new sensor::Sensor();

  BME280Spi(uint8_t cs)
    : PollingComponent(30000), _cs(cs) { }

  void setup() override {
    bme = new Adafruit_BME280(_cs);
    bool status = bme->begin();
    if (!status) {
      ESP_LOGD("bme280", "Failed to init!");
    } else {
      ESP_LOGD("bme280", "Init ok!");
    }
  }

  void update() override {
    float temperature = bme->readTemperature();
    float humidity = bme->readHumidity();
    float pressure = bme->readPressure() / 100.0F;
    temperature_sensor->publish_state(temperature);
    humidity_sensor->publish_state(humidity);
    pressure_sensor->publish_state(pressure);
    ESP_LOGD("bme280", "Temperature: %f C, Humidity: %f %%, Pressure: %f",
      temperature, humidity, pressure);
  }

private:
  Adafruit_BME280 *bme;
  uint8_t _cs;
};
```

**Additional information and things you've tried:**
Logging in the custom sensor works fine if the mqtt action is removed, but there is a clash if both the sensor and logger.log is added.

Johboh, 2019-05-25:
If I remove the `ESP_LOGD` from the custom component, it compiles with the action `logger.log`

OttoWinter, 2019-05-25:
Yes that is a known problem, and not really possible to solve.

The problem is that log statements are stored in flash - and that conflicts when used in class member functions.

But there's a workaround in dev already: see bottom of custom sensor page on https://next.esphome.io/ (basically allows you to disable storing log format strings in flash)

closed 2019-05-25 by OttoWinter

Johboh, 2019-05-26:
Ah, got it!
Will give the dev-version a go.