esphome/issues #342: Adding SI1145 as Custom Sensor fails reading values

Lefuneste83, 2019-05-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-05-25:
From a quick look the code looks ok.

Can you try publishing a fixed value like `42` on one of the sensors in the update block? If that works that means the hardware is not wired up correctly or the library does not work.

Lefuneste83, 2019-05-25:
Thanks a lot Otto for your quick reply.

Ok loading variables with fixed values in the update block works. I find the values published in HA so the rest of the code is fine indeed.

Wiring issue is not likely as I can upload the equivalenty Arduino code and it works (same calls to the same library).

I have a very dumb question. The library I am calling  "Adafruit_SI1145.h", do I need to put it myself inside some directory (and which one) or does the compiler downloads it directly from some repository ? It could be this stupid from me...

Lefuneste83, 2019-05-25:
Ok Otto I got it working... I modified slightly the custom sensor code to match the one in my Arduino sketch.

I changed the alias fonction declaration to match the one in my IDE code known to work.

From "Adafruit_SI1145 uv;" changed to "Adafruit_SI1145 uv = Adafruit_SI1145();"

Declared my variables before using them

Make a reset() call before begin() in the setup block

```
#include "esphome.h"
#include "Adafruit_SI1145.h"

using namespace esphome;

class SI1145_sensor : public PollingComponent {
 public:

  Adafruit_SI1145 uv = Adafruit_SI1145();

  sensor::Sensor *visible_sensor = new sensor::Sensor();
  sensor::Sensor *ir_sensor = new sensor::Sensor();
  sensor::Sensor *uvindex_sensor = new sensor::Sensor();

  SI1145_sensor() : PollingComponent(1000) { }

float visible;
float irlight;
float uvindex;

  void setup() override {
    uv.reset();
    uv.begin();
  }

  void update() override {
    visible = uv.readVisible();
    visible_sensor->publish_state(visible);

    irlight = uv.readIR();
    ir_sensor->publish_state(irlight);

    uvindex = uv.readUV();
    uvindex_sensor->publish_state(uvindex / 100.0);
  }
};
```

closed 2019-05-25 by Lefuneste83