esphome/issues #345: API Service Call from binary sensor

Rodrigoah, 2019-05-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 NodeMcu
**Affected component:**

Binary Sensor

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: led_strip
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "*****"
  password: "*****"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'Diferente1!'

ota:

light:
  - platform: monochromatic
    name: "Ledes Cama"
    output: output_component1
    id: Ledes_Cama

output:
  - platform: esp8266_pwm
    id: output_component1
    pin: D1
    

binary_sensor:
  - platform: gpio
    pin:
      number: D5
      mode: INPUT_PULLUP
      inverted: True
    name: "Switch D5"
    on_click:
      min_length: 50ms
      max_length: 350ms
      then:
      - homeassistant.service:
        service: light.toggle
        entity_id: light.ledes_cama
      
    
  - platform: gpio
    pin:
      number: D6
      mode: INPUT_PULLUP
      inverted: True
    name: "Switch D6"
    on_click:
      min_length: 50ms
      max_length: 350ms
      then:
        - light.toggle: Ledes_Cama
    
  - platform: gpio
    pin:
      number: D7
      mode: INPUT_PULLUP
      inverted: True
    name: "Switch D7"

```

**Traceback (if applicable):**

```
INFO Reading configuration...
Failed config

binary_sensor.gpio: [source /config/esphome/LED_strip.yaml:31]
  platform: gpio
  pin: 
    number: D5
    mode: INPUT_PULLUP
    inverted: True
  name: Switch D5
  on_click:  [source /config/esphome/LED_strip.yaml:38]
    min_length: 50ms
    max_length: 350ms
    then:  [source /config/esphome/LED_strip.yaml:41]
      
      Cannot have two actions in one item. Key 'homeassistant.service' overrides 'service'! Did you forget to indent the block inside the action?.
      - [source /config/esphome/LED_strip.yaml:41]
        homeassistant.service: 
        service: light.toggle
        entity_id: light.ledes_cama
```

**Additional information and things you've tried:**

OttoWinter, 2019-05-26:
Read the error message. You need to indent the block with the service options (service & data)

closed 2019-05-26 by OttoWinter

Rodrigoah, 2019-05-26:
When indenting one more level I get `[entity_id] is an invalid option for [0]. Please check the indentation.`

OttoWinter, 2019-05-26:
That is correct. entity_id goes in the data block

Rodrigoah, 2019-05-26:
Amazing Thanks!!