esphome/issues #348: Esphome stops tracking Xiaomi MiFlora devices - not sending updates to HA

grzeg8102, 2019-05-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HA in Docker. Esphome installed in Windows 10 with pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 WROOM DEVKIT V1
https://www.aliexpress.com/item/32849567377.html?spm=a2g0s.9042311.0.0.579e4c4dpgaE7U

**Affected component:**

https://esphome.io/components/sensor/xiaomi_miflora.html?highlight=miflora

**Description of problem:**
Yesterday suddenly the ESP32 BLE tracker stopped tracking my MiFlora device. Home Assitant shows its sensors as 'Unavailable' and the history graph shows just a horizontal line as no new values are pushed by the tracker. The MiFlora device is fully functional as I'm able to connect to it with my smartphone and see current readings. 
The ESP32 board is also fine I guess - it tracks MiJia temperature and humidity without any problems. The BLE signal range is not a problem either. I put it next to the ESP board and still no luck getting it work again.
I restarted the ESP32 - didn't help. I did a new OTA upgrade - didn't help either. I added this workaround with the restart switch - didn't help. Restarting doesn't help in general. Today I switched from MQTT to the native API - didn't help.

In the logs I see that the tracker receives the mesages from MiFlora, but does not push it to HA either via MQTT or the API.

If this matters - my ESP32 board is a tracker for 1 MiFlora device and 1 MiJia device + reads 3 ADC inputs. I wanted to add more MiFlora devices but now I'm in doubts if this makes sense as the 1st one stopped working suddenly and noting helps.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: sensors_tracker
  platform: ESP32
  board: esp32doit-devkit-v1
          
wifi:
  ssid: 'XXXX'
  password: 'XXXXX'

ota:
  password: 'xxxxxxxxxx'

# Enable logging
logger:

esp32_ble_tracker:

sensor:
  - platform: xiaomi_miflora
    mac_address: C4:7C:8D:65:FC:33
    temperature:
      name: "MiFlora Trawa Temperature"
    moisture:
      name: "MiFlora Trawa Moisture"
    illuminance:
      name: "MiFlora Trawa Illuminance"
    conductivity:
      name: "MiFlora Trawa Soil Conductivity"
    battery_level:
      name: "MiFlora Trawa Battery Level"

  - platform: xiaomi_miflora
    mac_address: C4:7C:8D:65:FB:D5
    temperature:
      name: "MiFlora Tuje Temperature"
    moisture:
      name: "MiFlora Tuje Moisture"
    illuminance:
      name: "MiFlora Tuje Illuminance"
    conductivity:
      name: "MiFlora Tuje Soil Conductivity"
      
  - platform: xiaomi_mijia
    mac_address: 4C:65:A8:DB:DB:A6
    temperature:
      name: "Garaz Temperature"
[... 560 lines cut ...]
[10:51:43][10:51:43][D][esp32_ble_tracker:375]: Xiaomi MiFlora C4:7C:8D:65:FC:33 Got soil conductivity=106┬ÁS/cm
[10:51:46][D][sensor.adc:087]: 'esp32_gleba': Got voltage=3.90V
[10:51:51][D][sensor.adc:087]: 'esp32_gleba': Got voltage=3.90V
[10:51:56][D][sensor.adc:087]: 'esp32_gleba': Got voltage=3.90V
[10:52:01][D][sensor.adc:087]: 'esp32_gleba': Got voltage=3.90V
[10:52:01][D][sensor.sensor:100]: 'esp32_gleba': Sending state -0.00000 V with 2 decimals of accuracy
[10:52:06][D][sensor.adc:087]: 'esp32_gleba': Got voltage=3.90V
[10:52:07][D][esp32_ble_tracker:372]: Xiaomi MiFlora C4:7C:8D:65:FC:33 Got illuminance=8258lx
[10:52:08][D][esp32_ble_tracker:369]: Xiaomi MiFlora C4:7C:8D:65:FC:33 Got moisture=28%
[10:52:11][D][sensor.adc:087]: 'esp32_gleba': Got voltage=3.90V
[10:52:16][D][sensor.adc:087]: 'esp32_gleba': Got voltage=3.90V
[10:52:21][D][sensor.adc:087]: 'esp32_gleba': Got voltage=3.90V
[10:52:26][D][sensor.adc:087]: 'esp32_gleba': Got voltage=3.90V
[10:52:31][D][sensor.adc:087]: 'esp32_gleba': Got voltage=3.90V
[10:52:31][D][esp32_ble_tracker:360]: Xiaomi MiFlora C4:7C:8D:65:FC:33 Got temperature=18.9°C
[10:52:35][D][sensor.adc:087]: 'esp32_deszcz': Got voltage=0.00V
[10:52:36][D][sensor.adc:087]: 'esp32_gleba': Got voltage=3.90V
[10:52:41][D][sensor.adc:087]: 'esp32_gleba': Got voltage=3.90V
[10:52:41][D][sensor.adc:087]: 'esp32_swiatlo': Got voltage=2.20V
[10:52:46][D][sensor.adc:087]: 'esp32_gleba': Got voltage=3.90V
[10:52:51][D][sensor.adc:087]: 'esp32_gleba': Got voltage=3.90V
[10:52:56][D][sensor.adc:087]: 'esp32_gleba': Got voltage=3.90V
```

**Additional information and things you've tried:**

OttoWinter, 2019-05-28:
Is it really "Unavailable" ? Or just "unknown". If it's unavailable that means that HA can't connect to the ESP. For that please look in the HA logs (and paste them here).

Also try removing the ESP from the integrations screen and re-adding it.

grzeg8102, 2019-05-28:
Yes, all sensors related to this MiFlora device are shown as 'Unavailable' in HA. The ESP surly **does** connect to HA as it sends readings of the ADC inputs and the MiJia device. I also added my 2nd MiFlora and its readings are sent to HA. No problem here. It's just this one MiFlora that is not being pushed to HA. 
Before I switched from MQTT to the API I observed the MQTT debug messages from the ESP and I observed no status updates from this device despite the fact that the ESP does receive the messages from MiFlora. I can see that in the logs (attched in the original post).
So all I can think of is not a problem in HA <-> ESP communication, but with the ESP software.

grzeg8102, 2019-05-28:
Update: I removed all sensors from the config leaving only the failing MiFlora and got it working again. Also noticed in the integrations screen that the entity name was changed. Now I'm not sure any longer if this is esphome or HA bug. Perhaps something screwed up in the entity registry in HA. But this does not explain no messages observed in the MQTT client. Sort of a "X-Files" case ;)
Well, let me observe it for a couple of days perhaps and if it'll be running find - I'll close this issue.
Thanks for the immediate reply.

zbuh, 2019-06-05:
I'm facing the same situation, it keep running for about 24h, then it stops updating  Mifloras. 
I placed a restart switch on ESP32 yaml, and remotely restart it... it then starts updating again...

grzeg8102, 2019-06-06:
Yesterday I was about to close this issue, but it happened to me again. Something's wrong with Esphome. This time the entities did not appear as 'Unavailable', so HA was ok. It must have been something wrong with esphome. The weird thing is that I setup automatic restart every 24 hours, which happens at noon (12:00) and the time when the tracker stopped working was 2 hours later. Only 2 hours after the automatic restart. The rest of the entities (ADC inputs) were tracked fine all the time, no interruptions, so ESP32 as a whole was working fine. 
Only BLE devices were not updated - neither Mi Flora, nor MiJia temp sensor. So there is something wrong with the BLE tracker.
Restarting the ESP32 programatically ( a switch wuth latform: restart) solved it.

zbuh, 2019-06-06:
Thats exactly what I'm facing. I also setup a restart at 00:00 to minimize this issue.

OttoWinter, 2019-06-06:
In that case I'm marking this as a duplicate of https://github.com/esphome/issues/issues/317 and merging the discussions there. - `unavailable` would mean HA cannot connect to the ESP, `unknown` is the thing described in #317

closed 2019-06-06 by OttoWinter