esphome/issues #349: esphome inflates snapshots

frog32, 2019-05-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** hassio

**ESP (ESP32/ESP8266, Board/Sonoff):** all of them

**Affected component:** all of them

**Description of problem:** esphome stores build artifacts in /config which inflates snapshots in hassio.

As an example every one of my many esphome projects has a .pioenvs folder that consumes around 9MB and an .piolibdeps folder that is ~3MB in size. Issuing `rm -r esphome/*/.pioenvs esphome/*/.piolibdeps` gets rid of those files and reduces my overall snapshot from over 100MB to ~13MB (I store all the databases offsite).

I see 3 possible solutions here with a strong preference on 1:

1. store those two build artifacts folders outside of /config since they are not real config
2. enable timed cash deletion so one can have esphome clear its build cache before automatic backups take place or as a minimal solution document this so users can build their own cache clearing into their backup solution
3. get hassio to allow for exclude patterns to not include those folders in snapshots 

**Problem-relevant YAML-configuration entries:**
```yaml

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-05-28:
1 is already the case with esphome >= 1.12 - the pioenvs etc is stored in a /data folder (which can be un-selected in the backup form)

So:

 - Users coming from pre-1.12 need to manually remove those folders in /config (can't be done automatically)
 - When creating a backup, deselect esphome (only cache data is stored there, no important data)

OttoWinter, 2019-05-28:
Closing this issue as already done - if there's a bug with the solution then please reopen.

closed 2019-05-28 by OttoWinter