esphome/issues #350: No api connection to HA after a while using Deep Sleep

Lefuneste83, 2019-05-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio on Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 Wemos D1 Mini R2

**Affected component:**

https://esphome.io/components/deep_sleep.html?highlight=api

**Description of problem:**
Hi there. I have what seems a very simple issue here.

I have setup a ESP8266 to monitor various environmental sensors as a wireless weather stations. I have implemented deep sleep mode with target values of 10s wake time every 10min. I have noticed that after starting the board, everything runs fine for a few hours and HA gets updated on a regular basis through native api calls, and my sensor values are refreshed on every wake period. But after some random time, the wake/sleep loop while, proceeding normaly, seems to fail establishing a connection to HA in order to update values. Sensors still get read locally though during the wake period, but the ESP does not connect to HA in the wake time slot.

It is quite strange as it seems to work fine at first, but nothing seems to repair this behaviour until I disconnect the board completely from its PS and restart it (or flash firmware again).

**Is there a way to conditionaly force the connection to HA (with api mode) on resume from wake or before going to sleep.**

I have tried Wifi fast connect, forced IP, increasing wake time, but the issue seems deeper than allowing more time for the board to establish a connection.
Or maybe I am doing someting fondamentaly wrong here... 
As a sidenote I would perfer to use api mode to connect to HA rather than MQTT if possible as I find it very elegant.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: meteo
  platform: ESP8266
  board: d1_mini
  
  includes:
    - SI1145.h
  libraries:
    - "Adafruit SI1145 Library"

wifi:
  ssid: "myssid"
  password: "mypwd"
  fast_connect: on

  manual_ip:
    static_ip: 192.168.0.XX
    gateway: 192.168.0.XX
    subnet: 255.255.255.0
    dns1: 192.168.0.XX
    dns2: 8.8.8.8
    
# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'mypwd'

ota:
  password: 'mypwd'

i2c:
  sda: D2
  scl: D1
  scan: True
  frequency: 200kHz

sensor:
  - platform: wifi_signal
    name: "Meteo WiFi Signal Sensor"
    update_interval: 1s
    
  - platform: bme280
    temperature:
      name: "BME280 Temperature"
      id: bme280_temperature
      oversampling: 16x
    pressure:
      name: "BME280 Pressure"
      id: bme280_pressure
      oversampling: 16x
    humidity:
      name: "BME280 Relative Humidity"
      id: bme280_humidity
    address: 0x76
    update_interval: 1s
  
  - platform: template
    name: "Absolute Humidity"
    lambda: |-
      const float mw = 18.01534;    // molar mass of water g/mol
      const float r = 8.31447215;   // Universal gas constant J/mol/K
      return (6.112 * powf(2.718281828, (17.67 * id(bme280_temperature).state) /
        (id(bme280_temperature).state + 243.5)) * id(bme280_humidity).state * mw) /
        ((273.15 + id(bme280_temperature).state) * r); // in grams/m^3
    accuracy_decimals: 2
    update_interval: 1s
  
  - platform: template
    name: "Equivalent sea level pressure"
    lambda: |-
      const float STANDARD_ALTITUDE = 260; // in meters, see note
      return id(bme280_pressure).state / powf(1 - ((0.0065 * STANDARD_ALTITUDE) /
        (id(bme280_temperature).state + (0.0065 * STANDARD_ALTITUDE) + 273.15)), 5.257); // in hPa
    update_interval: 1s

  - platform: custom
    lambda: |-
      auto UV_sensor = new SI1145_sensor();
      App.register_component(UV_sensor);
      return {UV_sensor->visible_sensor, UV_sensor->ir_sensor, UV_sensor->uvindex_sensor};

    sensors:
    - name: "Visible Light"
      unit_of_measurement: lux
    - name: "IR Light"
      unit_of_measurement: lux
    - name: "UV Index"
      unit_of_measurement: uvindex
      accuracy_decimals: 2
      
  - platform: tsl2561
    name: "TSL2561 Ambient Light"
    address: 0x39
    update_interval: 1s
      
deep_sleep:
  run_duration: 10s
  sleep_duration: 10min

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-05-28:
No, deep sleep is not really compatible with native API (https://github.com/esphome/issues/issues/174)

HA only checks if a given ESP is connectable every 60s (with an exponential backoff).

So you need to a) increase deep sleep run duration (and use more energy) or use MQTT.

See also https://github.com/esphome/feature-requests/issues/46

closed 2019-05-28 by OttoWinter

Lefuneste83, 2019-05-28:
Thank you for the quick feedback. This makes perfect sense now. I will try to move to MQTT based messenging then and will follow the progress of this feature.