esphome/issues #351: Can't compile with script name "open"

noxhirsch, 2019-05-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on Ubuntu
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 board
**Affected component:**

https://esphome.io/guides/automations.html?highlight=scripts#script-execute-action

**Description of problem:**
I created window blinds with esphome, which work great with 1.12.2, but didn't compile with dev and beta now. I tried different things and noticed, that the problem was my script name "open" (maybe "close" too - I renamed that in the same step).
It's not a problem for me anymore, but might be a problem beginners could come across, when they use easy names for their scripts (or other stuff like variable names? I don't know where such problems could appear, too). Maybe it makes sense to either validate these names, so people get a readable and easy to understand feedback - or do something like adding random letters to the names in the background.
If you think that this case is so rare that it can be ignored, please close this issue :) 

**Problem-relevant YAML-configuration entries:**
```yaml
script:
  - id: open
    then:
      - ...

```

**Traceback (if applicable):**

```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/sleep_rollo
Processing sleep_rollo (platform: espressif32@1.6.0; framework: arduino; board: nodemcu-32s)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 > NodeMCU-32S
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP> 1.0.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <WiFi> 1.0
|-- <Update> 1.0
Compiling /data/sleep_rollo/.pioenvs/sleep_rollo/src/main.cpp.o
Archiving /data/sleep_rollo/.pioenvs/sleep_rollo/lib789/libWiFi.a
Archiving /data/sleep_rollo/.pioenvs/sleep_rollo/libe94/libESPmDNS.a
Archiving /data/sleep_rollo/.pioenvs/sleep_rollo/libf2c/libPreferences.a
Compiling /data/sleep_rollo/.pioenvs/sleep_rollo/liba61/Update/Updater.cpp.o
Archiving /data/sleep_rollo/.pioenvs/sleep_rollo/libFrameworkArduinoVariant.a
Compiling /data/sleep_rollo/.pioenvs/sleep_rollo/FrameworkArduino/Esp.cpp.o
src/main.cpp:32:17: error: 'esphome::script::Script* open' redeclared as different kind of symbol
script::Script *open;
^
In file included from /root/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/newlib/sys/fcntl.h:3:0,
from /root/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/lwip/lwipopts.h:39,
from /root/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/lwip/lwip/opt.h:51,
from /root/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/lwip/lwip/pbuf.h:41,
from /data/sleep_rollo/.piolibdeps/AsyncTCP_ID1826/src/AsyncTCP.h:29,
from src/esphome/components/api/api_server.h:18,
from src/esphome.h:4,
from src/main.cpp:3:
/root/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/newlib/sys/_default_fcntl.h:186:12: note: previous declaration 'int open(const char*, int, ...)'
extern int open _PARAMS ((const char *, int, ...));
^
[... 114 lines cut ...]
^
src/esphome/components/script/script.h:32:32: note:   no known conversion for argument 1 from 'int(const char*, int, ...)' to 'const esphome::script::ScriptStopAction<>&'
src/esphome/components/script/script.h:32:32: note: candidate: constexpr esphome::script::ScriptStopAction<>::ScriptStopAction(esphome::script::ScriptStopAction<>&&)
src/esphome/components/script/script.h:32:32: note:   no known conversion for argument 1 from 'int(const char*, int, ...)' to 'esphome::script::ScriptStopAction<>&&'
src/main.cpp:522:67: error: no matching function for call to 'esphome::script::ScriptStopAction<>::ScriptStopAction(int (&)(int))'
script_scriptstopaction_3 = new script::ScriptStopAction<>(close);
^
In file included from src/esphome.h:23:0,
from src/main.cpp:3:
src/esphome/components/script/script.h:34:3: note: candidate: esphome::script::ScriptStopAction<Ts>::ScriptStopAction(esphome::script::Script*) [with Ts = {}]
ScriptStopAction(Script *script) : script_(script) {}
^
src/esphome/components/script/script.h:34:3: note:   no known conversion for argument 1 from 'int(int)' to 'esphome::script::Script*'
src/esphome/components/script/script.h:32:32: note: candidate: constexpr esphome::script::ScriptStopAction<>::ScriptStopAction(const esphome::script::ScriptStopAction<>&)
template<typename... Ts> class ScriptStopAction : public Action<Ts...> {
^
src/esphome/components/script/script.h:32:32: note:   no known conversion for argument 1 from 'int(int)' to 'const esphome::script::ScriptStopAction<>&'
src/esphome/components/script/script.h:32:32: note: candidate: constexpr esphome::script::ScriptStopAction<>::ScriptStopAction(esphome::script::ScriptStopAction<>&&)
src/esphome/components/script/script.h:32:32: note:   no known conversion for argument 1 from 'int(int)' to 'esphome::script::ScriptStopAction<>&&'
Compiling /data/sleep_rollo/.pioenvs/sleep_rollo/FrameworkArduino/FunctionalInterrupt.cpp.o
*** [/data/sleep_rollo/.pioenvs/sleep_rollo/src/main.cpp.o] Error 1
========================== [ERROR] Took 2.89 seconds ==========================
```

**Additional information and things you've tried:**

OttoWinter, 2019-05-29:
Hmm yeah that won't be solvable - the ID name is intentionally the one specified by the user because otherwise things like custom components can't access the values.

Also it's impossible to keep a complete list of these values because there are a lot of identifiers that are already used by the stdlib. I'll add open, close and some other ones I could find to the exclude list but the answer is just: don't use those IDs.

OttoWinter, 2019-05-29:
Closed in https://github.com/esphome/esphome/commit/b115b5c90d7769c2f52005e25ed786047021da94

closed 2019-05-29 by OttoWinter