esphome/issues #353: 1.13 Bug Report Thread

OttoWinter, 2019-05-29:
ESPHome v1.13 is currently in beta phase: https://beta.esphome.io/ 🎉 

The version also brings a big refactor on the C++ generator side. However, that also means some components could have stopped working.

This thread is for quickly finding and reporting issues in the current beta release so that the 1.13 release can be as stable as possible. You can post it as a new issue, or use this thread with the form provided below (this thread is here so that creating a bug report is quicker)

Most errors will be on the "this config does not compile" side. I don't expect any errors that would limit an ESPs ability to roll back if there's an issue.

If you find an issue, please do report it! Thanks!

````md
Component: <COMPONENT THAT HAS ISSUE>
Description:
DESCRIPTION OF PROBLEM

Configuration:
```yaml
# PASTE YAML CONFIG HERE

```

Additional Info:

````

oohlaf, 2019-05-30:
Component: Platform io compilation issue on hassio esphome 1.13
Description:
See build log below config.

Configuration:
```yaml
esphome:
  name: werkbank
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret esphome_password

ota:
  password: !secret esphome_password

output:
  - platform: ledc
    pin: 17
    id: gpio_17_red
  - platform: ledc
    pin: 5
    id: gpio_5_green
  - platform: ledc
    pin: 18
    id: gpio_18_blue
  - platform: ledc
    pin: 19
    id: gpio_19_white

light:
  - platform: rgbw
    name: "Werkbank"
    red: gpio_17_red
    green: gpio_5_green
    blue: gpio_18_blue
    white: gpio_19_white
```

Additional Info:
```
INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/werkbank
Processing werkbank (platform: espressif32@1.6.0; framework: arduino; board: nodemcu-32s)
--------------------------------------------------------------------------------
LibraryManager: Installing id=1826 @ 1.0.3
Downloading...
Unpacking  [####################################]  100%
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 > NodeMCU-32S
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP> 1.0.3
[... 72 lines cut ...]
from src/esphome/core/helpers.h:6,
from src/esphome/components/api/util.h:3,
from src/esphome/components/api/api_message.h:4,
from src/esphome/components/api/basic_messages.h:3,
from src/esphome/components/api/basic_messages.cpp:1:
/root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/uses_allocator.h:25:0: error: unterminated #ifndef
#ifndef _USES_ALLOCATOR_H
^
In file included from /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/memory:77:0,
from src/esphome/core/helpers.h:6,
from src/esphome/components/light/light_color_values.h:3,
from src/esphome/components/light/light_effect.h:4,
from src/esphome/components/light/light_state.h:6,
from src/esphome/core/controller.h:11,
from src/esphome/components/api/api_server.h:4,
from src/esphome/components/api/api_server.cpp:3:
/root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/uses_allocator.h:25:0: error: unterminated #ifndef
#ifndef _USES_ALLOCATOR_H
^
*** [/data/werkbank/.pioenvs/werkbank/src/esphome/components/api/api_message.cpp.o] Error 1
*** [/data/werkbank/.pioenvs/werkbank/src/esphome/components/api/command_messages.cpp.o] Error 1
*** [/data/werkbank/.pioenvs/werkbank/src/esphome/components/api/basic_messages.cpp.o] Error 1
*** [/data/werkbank/.pioenvs/werkbank/src/esphome/components/api/api_server.cpp.o] Error 1
========================== [ERROR] Took 11.57 seconds ==========================
```

OttoWinter, 2019-05-30:
@oohlaf Looks like a filesystem in your setup or in the build system truncated a file somehow. Please post more information about your setup, specifically what host is running Hass.io and it's architecture.

oohlaf, 2019-05-30:
It's a Raspberry Pi 3 Model B with hassos 2.12 and ha 0.93.2

tatunts, 2019-05-30:
Hello @OttoWinter.

After updating to version 1.13, I can no longer compile this project:

```yaml
esphome:
  name: ${project_name}
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: ${wifi_ssid}
  password: ${wifi_psw}
  fast_connect: false
  ap:
    ssid: ${friendly_name}
    password: ${ap_psw}

logger:

web_server:

api:
  password: ${api_psw}

ota:

light:
  - platform: fastled_clockless
    id: magic_led
    name: "Móvel da sala"
    rgb_order: GRB
    pin: D1
#    max_refresh_rate: 16ms
    num_leds: 108
    chipset: WS2812B
    effects:
      - random:
      - addressable_rainbow:
          name: Rainbow
          speed: 5
          width: 128
      - addressable_color_wipe:
      - addressable_scan:
          name: Scan
          move_interval: 10ms
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:
      - random:
          name: "RGB transition slow"
          transition_length: 30s
          update_interval: 30s
      - random:
          name: "RGB transition fast"
          transition_length: 4s
          update_interval: 5s              

binary_sensor:
  - platform: remote_receiver
    name: "IR TV sala volume mais"
    id: 'tv_lg_sala_vol_plus'
    filters:
      - delayed_off: 800ms
    lg:
      data: 0xC10040BF
      nbits: 32
#    raw:
#      data: [147, -315, 882, -421, 232, -164, 6792, -4553, 505, -1755, 532, -1711, 500, -707, 219, -840, 474, -610, 523, -650, 455, -655, 451, -1787, 478, -642, 489, -874, 234, -609, 523, -605, 504, -630, 499, -758, 375, -607, 499, -610, 523, -630, 475]

  - platform: remote_receiver
    name: "IR TV sala volume menos"
    id: 'tv_lg_sala_vol_minus'
    filters:
      - delayed_off: 800ms
    lg:
      data: 0xC100C03F
      nbits: 32
#    raw:
#      data: [152, -378, 868, -473, 186, -162, 442, -528, 734, -422, 816, -397, 3291, -4636, 527, -1716, 548, -1737, 474, -877, 233, -648, 483, -608, 523, -605, 502, -634, 498, -1735, 503, -607, 523, -627, 618, -573, 268, -783, 505, -701, 405, -656, 502]

  - platform: remote_receiver
    name: "IR TV sala energia"
    id: 'tv_lg_sala_vol_power'
    filters:
      - delayed_off: 800ms
    lg:
      data: 0x20DF10EF
      nbits: 32

sensor:
  - platform: wifi_signal
    id: wifi_measure
    update_interval: 60s

  - platform: template
    name: ${friendly_name} - Sinal Wifi
    unit_of_measurement: '%'
    accuracy_decimals: 0
    icon: "mdi:wifi"
    lambda: |-
        if (id(wifi_measure).state < -92.0)
          return 100.0;
        if (id(wifi_measure).state > -21.0)
          return 1.0;
        else
          return round(( -0.0154 * id(wifi_measure).state * id(wifi_measure).state ) - ( 0.3794 * id(wifi_measure).state ) + 98.182 );

  - platform: adc
    pin: A0
    name: 'planta_casa'
    id: 'planta_casa'
    update_interval: 3600s
    unit_of_measurement: '%'
    accuracy_decimals: 0
    icon: "mdi:water-percent"
    filters:
      - lambda: return (( ( x * 100.0 ) - 80.0 ) / ( 44.0 - 80.0 )) * 100.0;

remote_receiver:
  pin:
    number: GPIO2
    inverted: True
  dump: all
```

Here is the error log:

```
PS C:\Users\anton\Desktop\Casa> esphome casa_d1mini_sensor_planta.yaml config
[32mINFO Reading configuration...[0m
Failed config

binary_sensor.remote_receiver: [source casa_d1mini_sensor_planta.yaml:70]
  platform: remote_receiver
  name: IR TV sala volume mais
  id: tv_lg_sala_vol_plus
  filters:
    - [source casa_d1mini_sensor_planta.yaml:74]

      Expected string for time period with unit.
      delayed_off: 800ms [source casa_d1mini_sensor_planta.yaml:74]
  lg:
    data: 3238019263
    nbits: 32
binary_sensor.remote_receiver: [source casa_d1mini_sensor_planta.yaml:81]
  platform: remote_receiver
  name: IR TV sala volume menos
  id: tv_lg_sala_vol_minus
  filters:
    - [source casa_d1mini_sensor_planta.yaml:85]

      Expected string for time period with unit.
      delayed_off: 800ms [source casa_d1mini_sensor_planta.yaml:85]
  lg:
    data: 3238051903
    nbits: 32
binary_sensor.remote_receiver: [source casa_d1mini_sensor_planta.yaml:92]
  platform: remote_receiver
  name: IR TV sala energia
  id: tv_lg_sala_vol_power
  filters:
    - [source casa_d1mini_sensor_planta.yaml:96]

      Expected string for time period with unit.
      delayed_off: 800ms [source casa_d1mini_sensor_planta.yaml:96]
  lg:
    data: 551489775
    nbits: 32
```

Something changed when use the time syntax? 

Thanks!

brandond, 2019-05-30:
This issue will affect MQTT users coming from 1.12 or earlier: https://github.com/esphome/issues/issues/323

seanomat, 2019-05-31:
Component: api_message.cpp.o
Description:
Error while compile.

Configuration:
`substitutions:
  device_name: balkon_01
  api_pwd: !secret balkon_01_pwd

esphome:
  name: ${device_name}
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

logger:
  baud_rate: 0

#web_server:
#  port: 80

api:
  password: '${api_pwd}'

ota:
  password: '${api_pwd}'

switch:
  - platform: restart
    name: "${device_name}_restart"

light:
  - platform: rgb
    name: '${device_name}_Light_01'
    red: output_red
    green: output_green
    blue: output_blue
    effects:
      - random:
      - strobe:
      - flicker:
output:
  - platform: ledc
    id: output_red
    pin: GPIO16
  - platform: ledc
    id: output_green
    pin: GPIO17
  - platform: ledc
    id: output_blue
    pin: GPIO18

i2c:
  sda: 21
  scl: 22
  scan: false

sensor:
  - platform: uptime
    name: "${device_name}_Uptime"
  - platform: bh1750
    name: "${device_name}_Brightness"
    address: 0x23
    update_interval: 30s
    filters:
      - sliding_window_moving_average:
            window_size: 20
            send_every: 10
      - delta: 0.1
  - platform: bme280
    address: 0x76
    update_interval: 30s  
    temperature:
      name: "${device_name}_Temperature"
      filters:
        - offset: -1.5
        - delta: 0.1
        - sliding_window_moving_average:
            window_size: 20
            send_every: 10
      oversampling: 16x
    pressure:
      name: "${device_name}_Pressure"
      filters:
        - delta: 0.1
        - offset: -1000
        - sliding_window_moving_average:
            window_size: 20
            send_every: 10
    humidity:
      name: "${device_name}_Humidity"
      filters:
        - delta: 0.1
        - sliding_window_moving_average:
            window_size: 20
            send_every: 10

# Infrared
remote_receiver:
  pin: 
    number: GPIO26
    # Remote Receiver Signal starts with a HIGH value. Usually this means you have to invert the signal using 'inverted: True'
    inverted: yes
  dump: all

`

Compilelog

`INFO Reading configuration...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/balkon_01
Processing balkon_01 (platform: espressif32@1.6.0; framework: arduino; board: nodemcu-32s)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 > NodeMCU-32S
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP> 1.0.3
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Preferences> 1.0
|-- <Wire> 1.0.1
|-- <WiFi> 1.0
|-- <Update> 1.0
Compiling /data/balkon_01/.pioenvs/balkon_01/src/esphome/components/api/api_message.cpp.o
Compiling /data/balkon_01/.pioenvs/balkon_01/src/esphome/components/api/api_server.cpp.o
Compiling /data/balkon_01/.pioenvs/balkon_01/src/esphome/components/api/basic_messages.cpp.o
Compiling /data/balkon_01/.pioenvs/balkon_01/src/esphome/components/api/command_messages.cpp.o
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/balkon_01/.pioenvs/balkon_01/src/esphome/components/api/api_message.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/balkon_01/.pioenvs/balkon_01/src/esphome/components/api/api_server.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/balkon_01/.pioenvs/balkon_01/src/esphome/components/api/basic_messages.cpp.o] Error 127
sh: 1: xtensa-esp32-elf-g++: not found
*** [/data/balkon_01/.pioenvs/balkon_01/src/esphome/components/api/command_messages.cpp.o] Error 127
`

seanomat, 2019-05-31:
`asdfvefefverv`

seanomat, 2019-05-31:
Component:  api_message.cpp.o

Configuration:

    substitutions:
    device_name: balkon_01
    api_pwd: !secret balkon_01_pwd
    
    esphome:
      name: ${device_name}
      platform: ESP32
      board: nodemcu-32s
    
    wifi:
      ssid: !secret wifi_ssid
      password: !secret wifi_password
    
    logger:
      baud_rate: 0
    
    #web_server:
    #  port: 80
    
    api:
      password: '${api_pwd}'
    
    ota:
      password: '${api_pwd}'
    
    switch:
      - platform: restart
        name: "${device_name}_restart"
    
    light:
      - platform: rgb
        name: '${device_name}_Light_01'
        red: output_red
        green: output_green
        blue: output_blue
        effects:
          - random:
          - strobe:
          - flicker:
    output:
      - platform: ledc
        id: output_red
        pin: GPIO16
      - platform: ledc
        id: output_green
        pin: GPIO17
      - platform: ledc
        id: output_blue
        pin: GPIO18
    
    i2c:
      sda: 21
      scl: 22
      scan: false
    
    sensor:
      - platform: uptime
        name: "${device_name}_Uptime"
      - platform: bh1750
        name: "${device_name}_Brightness"
        address: 0x23
        update_interval: 30s
        filters:
          - sliding_window_moving_average:
                window_size: 20
                send_every: 10
          - delta: 0.1
      - platform: bme280
        address: 0x76
        update_interval: 30s  
        temperature:
          name: "${device_name}_Temperature"
          filters:
            - offset: -1.5
            - delta: 0.1
            - sliding_window_moving_average:
                window_size: 20
                send_every: 10
          oversampling: 16x
        pressure:
          name: "${device_name}_Pressure"
          filters:
            - delta: 0.1
            - offset: -1000
            - sliding_window_moving_average:
                window_size: 20
                send_every: 10
        humidity:
          name: "${device_name}_Humidity"
          filters:
            - delta: 0.1
            - sliding_window_moving_average:
                window_size: 20
                send_every: 10
    
    
    # Infrared
    remote_receiver:
      pin: 
        number: GPIO26
        # Remote Receiver Signal starts with a HIGH value. Usually this means you have to invert the signal using 'inverted: True'
        inverted: yes
      dump: all

Buildlog:

    INFO Reading configuration...
    INFO Generating C++ source...
    INFO Compiling app...
    INFO Running:  platformio run -d /config/esphome/balkon_01
    Processing balkon_01 (platform: espressif32@1.6.0; framework: arduino; board: nodemcu-32s)
    --------------------------------------------------------------------------------
    Verbose mode can be enabled via `-v, --verbose` option
    CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
    PLATFORM: Espressif 32 > NodeMCU-32S
    HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
    DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
    Library Dependency Finder -> http://bit.ly/configure-pio-ldf
    LDF MODES: FINDER(chain) COMPATIBILITY(soft)
    Collected 30 compatible libraries
    Scanning dependencies...
    Dependency Graph
    |-- <AsyncTCP> 1.0.3
    |-- <ESPmDNS> 1.0
    |   |-- <WiFi> 1.0
    |-- <Preferences> 1.0
    |-- <Wire> 1.0.1
    |-- <WiFi> 1.0
    |-- <Update> 1.0
    Compiling /data/balkon_01/.pioenvs/balkon_01/src/esphome/components/api/api_message.cpp.o
    Compiling /data/balkon_01/.pioenvs/balkon_01/src/esphome/components/api/api_server.cpp.o
    Compiling /data/balkon_01/.pioenvs/balkon_01/src/esphome/components/api/basic_messages.cpp.o
    Compiling /data/balkon_01/.pioenvs/balkon_01/src/esphome/components/api/command_messages.cpp.o
    sh: 1: xtensa-esp32-elf-g++: not found
    *** [/data/balkon_01/.pioenvs/balkon_01/src/esphome/components/api/api_message.cpp.o] Error 127
    sh: 1: xtensa-esp32-elf-g++: not found
    *** [/data/balkon_01/.pioenvs/balkon_01/src/esphome/components/api/api_server.cpp.o] Error 127
    sh: 1: xtensa-esp32-elf-g++: not found
    *** [/data/balkon_01/.pioenvs/balkon_01/src/esphome/components/api/basic_messages.cpp.o] Error 127
    sh: 1: xtensa-esp32-elf-g++: not found
    *** [/data/balkon_01/.pioenvs/balkon_01/src/esphome/components/api/command_messages.cpp.o] Error 127
    ========================== [ERROR] Took 7.52 seconds ==========================

OttoWinter, 2019-05-31:
@seanomat Is that the 64-bit version of RPi 3b+ (or any other device with aarch64?), I think you might be running into https://github.com/esphome/issues/issues/129

closed 2019-05-31 by OttoWinter

oohlaf, 2019-05-31:
> 
> 
> @oohlaf Looks like a filesystem in your setup or in the build system truncated a file somehow. Please post more information about your setup, specifically what host is running Hass.io and it's architecture.

I'm using 32bit btw and using docker exec -it I was able to look at the header file and indeed it got truncated on that line it just shows #en as last line. I removed /root/.platformio and starting a new build downloaded the dependencies again.

Might be nice to have an option to also clean the platformio installation from within the dashboard.

seanomat, 2019-06-01:
Switching to 32bit solved my problem.

andyinno, 2019-06-01:
I just came into #129 using  1.13.2 from the EspHome (not the beta) addon.

aarch64 hassio build of rpi3

How it is possible that a beta release comes into the stable branch?
If aarch64 is not supported by a build (or it is known it is broken) why display the upgrade button for the addon? This addon upgrade shouldn't be available if it's known it is broken at least for the archs that this is already known.

OttoWinter, 2019-06-02:
> How it is possible that a beta release comes into the stable branch?

Normal; at some point a beta is released; if no major issues are reported it gets promoted to stable; simple as that.

> If aarch64 is not supported by a build (or it is known it is broken) why display the upgrade button for the addon?

Hass.io shows the button, but has no rollback mechanism; that's Hass.io's fault; all other esphome installation methods have specific version targeting capabilities - plus it's only partially broken, the majority of ESPHome's userbase uses ESP8266 devices which do work.

Plus this is an external issue with platformio and their builds. So I'll wait for them to fix it. (If you want you can help them fix it too!)

oohlaf, 2019-06-02:
> I'm using 32bit btw and using docker exec -it I was able to look at the header file and indeed it got truncated on that line it just shows #en as last line. I removed /root/.platformio and starting [...]

BTW: When 1.13.2 was released the truncated header file issue returned again. I had to manually go into the docker container for the second time and remove the platformio folder and upon redownload of those sources I could compile again for 1.13.2.

Not sure how the code works that manages upgrading to a new release. But might that have an issue in causing truncated files?

OttoWinter, 2019-06-02:
@oohlaf I looked in the docker image and can't see the file truncated on my end - also I'm not seeing the error you've showed when using my own RPi in armv7 mode. You said it's 32-bit RPi, right?

Steps to view image:

```bash
docker run --rm --privileged multiarch/qemu-user-static:register
docker run --entrypoint "/bin/bash" -it esphome/esphome-hassio-armv7:1.13.3
> cat /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/uses_allocator.h
```

andyinno, 2019-06-02:
> Normal; at some point a beta is released; if no major issues are reported it gets promoted to stable; simple as that.
>
>
> Hass.io shows the button, but has no rollback mechanism; that's Hass.io's [...]

Otto, I can understand your reply.
But please understand also the end user of the product.
I don't have to read the changelog of everything, I don't have to check for every aspect of the product before the update.
I already saw this kind of reply in other open source project, the: "this is how it works, fix it yourself" is not a good reply. First because not everybody is a programmer, second because also if someone is a programmer it has some specific interests and want to use other software/tools does not need to be an expert.

I'm just saying... 
1) someone knows that there was a bug for some arch of the toolchain
2) the toolchain changed from beta to standard for everybody
3) somebody have a broken toolchain and cannot go back.

the reply:

> no major issue reported...

from what I see the issue #129 is from march, it is open and it is confirmed.

What is the solution migrating from the 64bit version to the 32bit version? really?

oohlaf, 2019-06-03:
> @oohlaf I looked in the docker image and can't see the file truncated on my end - also I'm not seeing the error you've showed when using my own RPi in armv7 mode. You said it's 32-bit RPi, right?

@OttoWinter  Yes it's 32bit RPi3b on HassOS. I can't reproduce the issue anymore when upgrading from 1.13.2 to 1.13.4. So maybe it was  a one off issue (or two off actually).