esphome/issues #355: End stop cover, how it works?

glmnet, 2019-05-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

beta, Endstop Cover

**Description of problem:**

I'm just reading the docs and have a question about how it behaves. I'm copying the description here for reference:

> The endstop cover platform allows you to create covers with position control that have endstops at both ends of the cover to detect the fully-open and fully-closed states. When any of these endstops are reached, the cover stops and sends out the corresponding state.

Now, my covers have 433Mhz remotes, when I press open they will open and stop by themselves when they are fully open, likewise when closing. I'm okay with that and I don't want to change. To have endstop detection by ESP I will need to add some reed switches, I would not like ESPHome to send the stop command to them as soon as the reed switch is made. 
From what I understand, ESPHome will do so, in that case. 

- Can this be stated clearly on the docs and... 
- now this becomes a feature request: can it be made configurable?

`send-endstop-commands: false ` # defaults to true

Sending the stop to an already stop cover will not damage it so I could add a delay to the sensor but that seems like weird.
I guess what makes more sense is to put the reed switches so that they are well made by the time the cover stops by itself.

Thanks

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-05-30:
The endstop platform is for DIY covers - setup is that you have endstops at either end and a motor that controls them. So yes when the endstop is reached `stop_action` is called.

I'm not sure why you want to use the endstop platform for your cover. If the cover does the stopping itself I think you should look at different platforms, for example time_based. That way you don't need to install reed switches but you also get position control.

glmnet, 2019-05-30:
Great, it makes more sense now.

The thing is I'd like to have a positive feedback when the cover are really closed. This is a security concern for anti-theft reasons, and I don't want to hack in the covers (electro-mechanical) logic with high voltages in there. 

Anyway I realize I have to write a lambda for the `stop_action` so I can check if any of my reed_switches are made or something else before sending the 433mhz stop action.

Docs look better now, except for a bit here: 

> Two endstops at either and a motor controlling the cover.

I'm an spanish language native but that seems like a word is missing there?

Thanks

ricreis394, 2019-06-10:
I'm with the same problem as @glmnet 
I tried using time_based component but at the end of the time the stop is triggered the same way as endstop component. Is that possible to just trigger the Stop command only when the user click on the stop button?

glmnet, 2019-06-10:
Actually I did not implement this neither way yet (433 transmitter is lost in the mail ether)
I was thinking that the stop_action should be executed only if any of my end stops where not made
so I could do:

say you have some reed switches:
````yaml
binary_sensor:
  - platform: ..
    id: reed_fully_open
  - platform: ..
    id: reed_fully_closed

cover:
  - platform: endstop
    # ...
    stop_action:
      - if:
          condition:
            and:
              - binary_sensor.is_off: reed_fully_open
              - binary_sensor.is_off: reed_fully_closed
          then:
            # -- do your actual stop action.
````
The previous condition actually reads: _if none of the sensors are made then execute the stop action._

Disclaimer: untested code

OttoWinter, 2019-07-11:
Closing as I would consider this more of a feature request, plus we already have https://github.com/esphome/esphome/pull/665 coming so it will be done by then anyway.

closed 2019-07-11 by OttoWinter