esphome/issues #356: hx7411 reports inaccurate results.

vamichael, 2019-05-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**D1 Mini:**

**Affected component:**

https://esphome.io/components/sensor/hx711.html?highlight=hx711

**Description of problem:**

Testing three different weights:

0 lbs reports: 8397548
8.34 lbs (gallon of water) reports: 24981314
200 lbs (me) reports: 20611480 (which is less than what a gallon of water reports)
--

--

--

**Problem-relevant YAML-configuration entries:**
```yaml

sensor:
  - platform: hx711
    name: "HX711 Value"
    dout_pin: D2
    clk_pin: D3
    gain: 128
    update_interval: 1s
    unit_of_measurement: kg
    filters:
      - lambda: |-
          auto first_mass = 0.0; 
          auto first_value = 8398500;
          auto second_mass = 3.78;
          auto second_value = 24981300;
          return x;

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

I've run the `SparkFun_HX711_Calibration` sketch using the same hardware which reports accurate values.

vamichael, 2019-05-30:
I also tried using the filter code and tested that along with the `map` function.  Not only are the results incorrect, these two different methods return quite different results. The [SparkFun code](https://github.com/sparkfun/HX711-Load-Cell-Amplifier/blob/master/firmware/SparkFun_HX711_Calibration/SparkFun_HX711_Calibration.ino) which I used on the same hardware uses the [bogde HX711 library](https://github.com/bogde/HX711) which I believe is the same one you use.  The only modification I made was to switch the DT and CLK pins, just because I soldered them that way.  Note that in my ESPHome yaml I specified `D2` and `D3`.  I tried playing around with using other values, but the log showed readings weren’t being captured unless I used what I have below.

Is there anything I can do to help narrow down the problem?

```
sensor:
  - platform: hx711
    name: "HX711 Value"
    dout_pin: D2
    clk_pin: D3
    update_interval: 1s
    unit_of_measurement: g
    filters:
      - calibrate_linear:
          - 8400000 -> 0.000
          - 24984758 -> 3938
                    
      # - lambda: |-
      #     auto first_value = 8400000;
      #     auto first_mass = 0.0; 
      #     auto second_value = 24984758;
      #     auto second_mass = 3938;
      #     return map(x, first_value, second_value, first_mass, second_mass);             
```

vamichael, 2019-05-31:
@OttoWinter 

I also just noticed something odd in the [documentation](https://github.com/esphome/esphome-docs/blob/current/components/sensor/hx711.rst).

> Connect ``GND`` to ``GND``, ``VCC`` to ``3.3V`` and the other three ``MISO`` (or ``SO`` for short),
``CS`` and ``CLOCK`` (or ``SCK``) to free GPIO pins.

Although I’m not using the Sparkfun amplifier, the one I have is compatible with all of the HX711 libraries.  And from all of the documentation available, these amplifiers should work at both 3.3v and 5v.  I’m also not sure about the MISO/SO, CS, CLOCK/SCK outputs.  All of the HX711 hookup guides mention DT and SCK outputs being connected to GPIO pins, and the Arduino sketches only mention these two outputs/pins.  Indeed the ESPHome example only mentions these two pins in the documentation.  Perhaps this is a typo, and even if it is, I doubt this is related to the issue I’m experiencing.

What I am aware of is that the output of the component does not appear to be linear.  As more weight is applied to the array of load sensors, one would expect the component output to also increase.  But in reality the output actually *decreases* somewhere between 100 lbs and 200 lbs.  I can experiment a bit more to find out more precisely where this happens, if you think this will help.

OttoWinter, 2019-07-11:
Closing due to inactivity - and HX711 has been updated some releases ago to have a better value conversion.

closed 2019-07-11 by OttoWinter