esphome/issues #360: Rotary Encoder boot loop on v1.13.1

sabeechen, 2019-05-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io
Home Assistant  0.92.2
ESPHome 1.13.1

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32 Devkit

**Affected component:**
Rotary Encoder

**Description of problem:**
I narrowed the failure down to rotary encoder component, with the minimal yaml file below that just configures the rotary encoder component.  My ESP32 boot loops before entering into safe mode.  It compiled and ran fine before upgrading to v1.13.x, so I suspect its caused by the latest rewrite.  In the console it prints out:
```
[0;32m[I][logger:116]: Log initialized[0m
[0;35m[C][ota:364]: There have been 9 suspected unsuccessful boot attempts.[0m
[0;32m[I][app:027]: Running through setup()...[0m
[0;35m[C][rotary_encoder:101]: Setting up Rotary Encoder 'Living Room LED Encoder'...[0m
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d4c1b  PS      : 0x00060730  A0      : 0x800d4c53  A1      : 0x3ffb1dc0  
A2      : 0x3ffb8b44  A3      : 0x3ffb8c0c  A4      : 0x40080f54  A5      : 0x00000003  
A6      : 0x3ffb8be4  A7      : 0x3ffb8c44  A8      : 0x800d4c18  A9      : 0x3ffb1da0  
A10     : 0x00000000  A11     : 0x40080f54  A12     : 0x3ffb8c0c  A13     : 0x00000003  
A14     : 0x00000065  A15     : 0x3ffb8c6c  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffe9  

Backtrace: 0x400d4c1b:0x3ffb1dc0 0x400d4c50:0x3ffb1de0 0x400d894e:0x3ffb1e00 0x400d8719:0x3ffb1e20 0x400d9d4f:0x3ffb1e70 0x400ddfc3:0x3ffb1fb0 0x4008eb9d:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7944
entry 0x40080310
[0;32m[I][logger:116]: Log initialized[0m
[0;35m[C][ota:364]: There have been 10 suspected unsuccessful boot attempts.[0m
[1;31m[E][ota:369]: Boot loop detected. Proceeding to safe mode.[0m
[0;35m[C][wifi:029]: Setting up WiFi...[0m
[0;36m[D][wifi:268]: Starting scan...[0m
[0;36m[D][wifi:283]: Found networks:[0m
[0;32m[I][wifi:318]: - redacted 
[0;36m[D][wifi:319]:     Channel: 2[0m
[0;36m[D][wifi:320]:     RSSI: -47 dB[0m
some lines removed
[0;32m[I][wifi:164]: WiFi Connecting to 'because'...[0m
[0;32m[I][wifi:376]: WiFi connected![0m
[0;35m[C][wifi:250]:   SSID: redacted
[0;35m[C][wifi:251]:   IP Address: 192.168.1.245[0m
[0;35m[C][wifi:253]:   BSSID: [5m74:D0:2B:D0:A9:38[6m[0m
[0;35m[C][wifi:254]:   Hostname: 'living_room_lamp'[0m
[0;35m[C][wifi:258]:   Signal strength: -44 dB [0;32m▂▄▆█[0m[0m
[0;35m[C][wifi:259]:   Channel: 2[0m
[0;35m[C][wifi:260]:   Subnet: 255.255.255.0[0m
[0;35m[C][wifi:261]:   Gateway: 192.168.1.1[0m
[0;35m[C][wifi:262]:   DNS1: 192.168.1.1[0m
[0;35m[C][wifi:263]:   DNS2: 0.0.0.0[0m
[0;35m[C][ota:029]: Over-The-Air Updates:[0m
[0;35m[C][ota:030]:   Address: living_room_lamp.local:3232[0m
[0;35m[C][ota:032]:   Using Password.[0m
[0;33m[W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts[0m
[0;32m[I][ota:375]: Waiting for OTA attempt.[0m
```

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: living_room_lamp
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: redacted
  password: redacted

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: redacted
ota:
  password: redacted

sensor:
  - platform: rotary_encoder
    name: "Living Room LED Encoder"
    resolution: 2
    pin_a:
      number: 18
      mode: INPUT_PULLUP
    pin_b:
      number: 19
      mode: INPUT_PULLUP
```

**Traceback (if applicable):**
The config compiles and uploads without error, just doesn't run.  

**Additional information and things you've tried:**
Is there a way I can provide a decoded stacktrace for the error?  I can't figure if the esphome addon makes the .elf files accessible somewhere.

sabeechen, 2019-05-31:
BTW I just wanted to say that I think this addon is great.  It really lets you focus on writing the automation logic instead of getting frustrated trying to find the right library to work with or mucking around with network libraries.  It was the super-clean and understandable interface this add-on provides that convinced me to author [my own add-on](https://github.com/sabeechen/hassio-google-drive-backup) with a similar design aesthetic.
Hass.io needs more addon's like this with good UI's and great documentation.  I really appreciate that you publish doxygen info for your source for those of us who want to get really complicated with our lambdas.

sandro-defender, 2019-05-31:
same problem after updating, node with rotary encoder stops working, status led flashs rapidly...

closed 2019-05-31 by OttoWinter

sabeechen, 2019-05-31:
The new version today (1.13.2) works great, thanks for the quick turnaround.