esphome/issues #361: Esphome through hass.io on Ubuntu 19.04 gives nginx permission denied errors

saberone, 2019-05-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Ubuntu 19.04
Hass.io
Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**
Not relevant for issue.

**Affected component:**
Not relevant

**Description of problem:**
Esphome won't start. It doesn't seem to be able to start nginx. Esphome was installed from hass.io. It's a clean install running on Ubuntu 19.04 server.

```
-----------------------------------------------------------
 Hass.io Add-on: ESPHome
 ESPHome Hass.io add-on for intelligently managing all your ESP8266/ESP32 devices.
-----------------------------------------------------------
 Add-on version: 1.13.1
 You are running the latest version of this add-on.
 System: Ubuntu 19.04  (amd64 / qemux86-64)
 Home Assistant version: 0.93.2
 Supervisor version: 164
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
[cont-init.d] 00-banner.sh: exited 0.
[cont-init.d] 01-log-level.sh: executing... 
[cont-init.d] 01-log-level.sh: exited 0.
[cont-init.d] 10-requirements.sh: executing... 
[cont-init.d] 10-requirements.sh: exited 0.
[cont-init.d] 20-nginx.sh: executing... 
[cont-init.d] 20-nginx.sh: exited 0.
[cont-init.d] 30-esphome.sh: executing... 
[cont-init.d] 30-esphome.sh: exited 0.
[cont-init.d] 40-migrate.sh: executing... 
[cont-init.d] 40-migrate.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[01:42:08] INFO: Waiting for dashboard to come up...
[01:42:08] INFO: Starting ESPHome dashboard...
INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
[01:42:09] INFO: Starting NGINX...
nginx: [emerg] open() "/proc/1/fd/1" failed (13: Permission denied)
[01:42:09] INFO: Waiting for dashboard to come up...
[01:42:09] INFO: Starting NGINX...
nginx: [emerg] open() "/proc/1/fd/1" failed (13: Permission denied)
```

**Problem-relevant YAML-configuration entries:**
Not relevant

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-05-31:
Hmm interesting, your system must be somehow blocking docker from accessing fd1 - but that file descriptor is stdout and that should *always* be accessible. I have the feeling this is a bug somewhere in docker or your filesystem setup.

Related https://community.home-assistant.io/t/community-hass-io-add-on-influxdb/54491/182, https://github.com/moby/moby/issues/31243

saberone, 2019-05-31:
Well... after a lot of late night cursing. I figured out that it had to do with the docker version that is installed by Ubuntu. I reinstalled the system and manually installed docker, as is advised by home assistant. Now everything works fine and dandy.

So this issue can be closed.

closed 2019-05-31 by OttoWinter