esphome/issues #362: ESP Home API ERROR

Wriman, 2019-05-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22ESP8266WiFi%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x76139310>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution',)) (incremented retry: max=5, total=1)
**Affected component:**

**Description of problem:**
NO, ITS NOTE SAME!! I never used pi hole and same app, but my esp home never Compile anythingh.. 
Please HELP ME

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
-Reinstall HASSIO
-Calibrate anything my rooter (Tplink Archer C5 v4
-Reinstall ESP home many times
-read and try many things from here: https://github.com/esphome/esphome/issues/40
or here: https://github.com/esphome/issues/issues/302

the addon is never work, but i wery like it

OttoWinter, 2019-05-31:
Yep I understand, but I'd still like to continue the discussion in https://github.com/esphome/issues/issues/302, because it's still an error in the DNS setup somewhere as seen here: `Temporary failure in name resolution'`

What happens if you type `ping api.platformio.org` in some console (for example Hass.io console addon)?

Also, what system in hassio running on? RPi or so?

OttoWinter, 2019-05-31:
Moving to #302

closed 2019-05-31 by OttoWinter

Wriman, 2019-05-31:
the problem is my rooter ip adress. 192.168.8.8
is change it 192.168.1.1, and its Works
i dont understood why is problem the first address