esphome/issues #367: MQTT on_json_message result in boot loop

SebDominguez, 2019-05-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

macOS 10.14.5
Python 2.7.16 (brew)
pip 19.0.3

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 WROOM DOIT DEVKIT V1

**Affected component:**

https://esphome.io/components/mqtt.html?highlight=mqtt#on-json-message-trigger

**Description of problem:**

First of, I just tried the exemple found on the documentation but unfortunately it seems that the YAML is wrong somehow

**Problem-relevant YAML-configuration entries:**
```yaml
mqtt:
  broker: 10.0.2.3
  username: !secret mqtt_username
  password: !secret mqtt_password
  client_id: esp32_0001
  port: 1883
  birth_message:
    topic: lwt/esp
    payload: online
  will_message:
    topic: lwt/esp
    payload: offline
  log_topic: esphome
  on_json_message:
    topic: the/topic
      then:
      - light.turn_on:
          id: leds

          transition_length: !lambda |-
            int length = 1000;
            if (x.containsKey("length"))
              length = x["length"];
            return length;

          brightness: !lambda "return x["bright"];"

          effect: !lambda |-
            const char *effect = "None";
            if (x.containsKey("effect"))
              effect = x["effect"];
            return effect;
```

**Traceback (if applicable):**

```
ERROR Error while reading config: Invalid YAML syntax. Please see YAML syntax reference or use an online YAML syntax validator:

mapping values are not allowed here
  in "config/esp32_led_test.yaml", line 28, column 11:
          then:
              ^
```

**Additional information and things you've tried:**

I've changed the yaml to

```yaml
  on_json_message:
    - topic: esp_led_test/json
      then:
        - light.control:
            id: leds
            state: !lambda |-
              int state = true;
              if (x.containsKey("state"))
                state = x["state"];
              return state;
    - topic: esp_led_test/effects
      then:
        - light.control:
            id: leds
            effect: !lambda |-
              const char *effect = "None";
              if (x.containsKey("effect"))
                effect = x["effect"];
              return effect;
```

This does compile and sending the payload  `"{\"state\": false}"` `"{\"state\": true}"` toggle the light, but then I tried to enable the Rainbow effect with `"{\"effect\": \"Rainbow\"}"` and since then my esp32 is stuck on a boot loop

Light component config 

```yaml
light:
  - platform: fastled_clockless
    chipset: WS2813
    pin: GPIO23
    num_leds: 160
    rgb_order: BRG
    name: "FastLED WS2811 Light"
    id: leds
    effects:
    - addressable_rainbow:
    - addressable_color_wipe:
    - addressable_scan:
    - addressable_twinkle:
    - addressable_random_twinkle:
    - addressable_fireworks:
    - addressable_flicker:
```

logs (using the USB to UART interface since the logs is now unavailable wirelessly)
```
[20:07:30]ets Jun  8 2016 00:22:57
[20:07:30]
[20:07:30]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[20:07:30]configsip: 0, SPIWP:0xee
[20:07:30]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[20:07:30]mode:DIO, clock div:2
[20:07:30]load:0x3fff0018,len:4
[20:07:30]load:0x3fff001c,len:928
[20:07:30]ho 0 tail 12 room 4
[20:07:30]load:0x40078000,len:9280
[20:07:30]load:0x40080400,len:5860
[20:07:30]entry 0x40080698
[20:07:30][I][logger:116]: Log initialized
[20:07:30][C][ota:364]: There have been 4 suspected unsuccessful boot attempts.
[20:07:30][I][app:027]: Running through setup()...
[20:07:30][C][fastled:010]: Setting up FastLED light...
[20:07:30][C][light:094]: Setting up light 'FastLED WS2811 Light'...
[20:07:30][D][light:261]: 'FastLED WS2811 Light' Setting:
[20:07:30][D][light:270]:   State: ON
[20:07:30][D][light:274]:   Brightness: 100%
[20:07:30][D][light:283]:   Red=0%, Green=100%, Blue=100%
[20:07:30][D][light:323]:   Effect: 'Rainbow'
[20:07:30][C][wifi:029]: Setting up WiFi...
[20:07:30][D][wifi:268]: Starting scan...
[20:07:30]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[20:07:30]Core 1 register dump:
[20:07:30]PC      : 0x40160783  PS      : 0x00060430  A0      : 0x800d3755  A1      : 0x3ffb1ac0  
WARNING Decoded 0x40160783: esphome::light::AddressableLight::operator[](int) const
[20:07:31]A2      : 0x3ffb1b24  A3      : 0x00000000  A4      : 0x00000026  A5      : 0x3ffb19e0  
[20:07:31]A6      : 0x00000000  A7      : 0x3ffb0060  A8      : 0x00000020  A9      : 0x80000020  
[20:07:31]A10     : 0x00000008  A11     : 0x00000001  A12     : 0x00000005  A13     : 0x00000020  
[20:07:31]A14     : 0x00000020  A15     : 0x00000003  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
[20:07:31]EXCVADDR: 0x00000000  LBEG    : 0x40160d19  LEND    : 0x40160d32  LCOUNT  : 0x00000000  
[20:07:31]
[20:07:31]Backtrace: 0x40160783:0x3ffb1ac0 0x400d3752:0x3ffb1ae0 0x400dda8c:0x3ffb1b00 0x400dd567:0x3ffb1b60 0x400d4842:0x3ffb1b90 0x400d48f6:0x3ffb1bd0 0x400dc962:0x3ffb1bf0 0x400dc44b:0x3ffb1c10 0x400df567:0x3ffb1c60 0x400e4fc3:0x3ffb1fb0 0x4008f9dd:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40160783: esphome::light::AddressableLight::operator[](int) const
WARNING Decoded 0x400d3752: esphome::light::ESPRangeView::Iterator::operator*() const
WARNING Decoded 0x400dda8c: esphome::light::AddressableRainbowLightEffect::apply(esphome::light::AddressableLight&, esphome::light::ESPColor const&)
WARNING Decoded 0x400dd567: esphome::light::AddressableLightEffect::apply()
WARNING Decoded 0x400d4842: esphome::light::LightState::loop()
WARNING Decoded 0x400d48f6: non-virtual thunk to esphome::light::LightState::loop()
WARNING Decoded 0x400dc962: esphome::Component::call_loop()
WARNING Decoded 0x400dc44b: esphome::Application::setup()
WARNING Decoded 0x400df567: setup()
WARNING Decoded 0x400e4fc3: loopTask(void*)
WARNING Decoded 0x4008f9dd: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c:355 (discriminator 1)
[20:07:31]
[20:07:31]Rebooting...
```

trying to reflash the esp with a blank firmware or a known working one didn't stop the boot loop. For that I needed to rollback to 1.12.2 and reflash my known working yaml  : 

```yaml
mqtt:
  broker: 10.0.2.3
  username: !secret mqtt_username
  password: !secret mqtt_password
  client_id: esp32_0001
  port: 1883
  birth_message:
    topic: lwt/esp
    payload: online
  will_message:
    topic: lwt/esp
    payload: offline
  log_topic: esphome
  on_message:
    - topic: esp_led_test/TOGGLE
      then:
        - light.toggle: leds
    - topic: esp_led_test/LIGHT
      payload: "ON"
      then:
        - light.turn_on: leds
    - topic: esp_led_test/LIGHT
      payload: "OFF"
      then:
        - light.turn_off: leds
```

OttoWinter, 2019-05-31:
The problem very likely has nothing to do with the MQTT stuff (seen by the error repeating each reboot, even before MQTT is connected).

I changed the title to match that.

networkid, 2019-05-31:
I'm also experiencing issues with FastLED effects since update to 1.13.0-1-2 (Flicker with custom doesn't allow color change and gradually fades to no lights although still has ON state).  Fireworks causes the ESP8266 to crash, requires reboot.  Thanks for all the great work Otto :)

natwilson, 2019-06-01:
I'm using the Neopixelbus platform and the addressable_fireworks effect causes the ESP8266 to crash with the status LED flash rapidly.  All the logs say is "WARNING Disconnected from API."  It worked okay with version 1.12.  Currently using version 1.13.2.

closed 2019-06-01 by OttoWinter