esphome/issues #371: Hass.io can't upgrade to 1.13.0+: "This update is no longer compatible with your system."

ifol, 2019-06-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

OrangePi PC (armhf) Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

It seems the armhf architecture was removed for 1.13.0, is there any reason this is no longer supported? Is there any way to make it work please?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**

OttoWinter, 2019-06-01:
The base image (https://github.com/hassio-addons/addon-ubuntu-base) does not support armhf anymore, that's why.

Though the interesting thing is that esphome's armhf image was always really just an armv7 image (armv7 *is* a type of armhf, but not the other way around); This update just made it more clear that it really only supports armv7 and for example not armv6 (at least the base image does not). So I think the architecture detection on your system might be wrong.

ifol, 2019-06-01:
you might be right, this is what /proc/cpuinfo outputs:

model name      : ARMv7 Processor rev 5 (v7l)

This is what description of the board CPU says:

H3. Allwinner H3 (sun8iw7p1) SoC features a Quad-Core Cortex-A7 ARM CPU, and a Mali400 MP2 GPU from ARM.

Can you advise how to rectify the issue and use the correct architecture for my system?

OttoWinter, 2019-06-01:
Sorry, not my area of expertise. Please try in the Hass.io forums, I think they will be able to help.

frenck, 2019-06-01:
## The history

`armhf` was actually never supported, since the images are based on Ubuntu (which only supports `armv7`.

In the past, Hass.io used `armhf` for all `armv5/v6/v7` CPU's. Recently, Hass.io gained support for `armv7` (because Alpine Linux added native support). Hence the base images now use the "correct" label, instead of the mislabeling previously.

As a matter of fact, nothing changed in the devices supported. Only the label is now correct.

I'll take a look at the Hass.io add-on images to see if I can locate the issue.

OttoWinter, 2019-06-01:
@frenck Thanks for looking into this. As to why the error is showing up, I know why: The config.json file no longer contains `armhf` in the `archs` section.

For me the question would rather be:

- Does the armv7 image secretly also support armv6 (even if it's just random)
- Why does HA detect the arch of the device in question as armhf, even if it appears to be armv7.

Again, not in any way my area of expertise, so can't help much.

frenck, 2019-06-01:
@OttoWinter Your `config.json` is correct. Don't add `armhf`, that will cause issues.

> Does the armv7 image secretly also support armv6 (even if it's just random)

No, it does not. Ubuntu does not compile for armv6 CPU's (and has never done that). Please note, in Ubuntu's world, `armhf` == `armv7`. In the Hass.io world (based on Alpine Linux) `armhf` == `armv5/v6` and `armv7` == `armv7`. (yeah the definitions of this is all over the place, a pretty big issue in the ARM world in general... e.g., `aarch64/armv8/arm64`)
Hence, previously, people would get an error when installing the add-on on an ARMv6 CPU (e.g., Raspberry Pi Zero). It would install, but not start.

> Why does HA detect the arch of the device in question as armhf, even if it appears to be armv7.

That is a good question. More a question for Hass.io in general I guess.
Given a part of the log @ifol posted on the Base image repository:

```txt
CONTAINER ID IMAGE COMMAND CREATED STATUS PORTS NAMES
59d4db90d27e hassioaddons/node-red-armhf "/init" 2 hours ago Up 2 hours addon_a0d7b954_nodered
baf8f2fe9826 esphome/esphome-hassio-armhf "/init" 2 hours ago Up 2 hours addon_15ef4d2f_esphome
97d9ee16bc04 homeassistant/armhf-hassio-supervisor "python3 -m hassio" 2 hours ago Up 2 hours hassio_supervisor
8a6ba517f386 homeassistant/qemuarm-homeassistant "/bin/entry.sh pytho…" 12 hours ago Up 2 hours homeassistant
```

It seems like everything is running `armhf` on his end (including the supervisor). This makes me wonder if the original machine setting is correct on his end. Cause by user or Hass.io error... I'm not sure...

frenck, 2019-06-01:
Recap: Everything looks great on your end @OttoWinter.

ifol, 2019-06-01:
just to address whether my system is configured correctly, in general it appears so as it correctly reports armv7l:

root@orangepipc:~# uname -m
armv7l

I think the issue is the install script I referenced in the above report, this one:
https://github.com/home-assistant/hassio-installer/blob/master/hassio_install.sh

I've used this script to install Hass.io, you may notice that even when you specify the machine as armv7l, it still installs hassio supervisor for armhf:

# Generate hardware options
case $ARCH in
    "i386" | "i686")
        MACHINE=${MACHINE:=qemux86}
        HOMEASSISTANT_DOCKER="$DOCKER_REPO/$MACHINE-homeassistant"
        HASSIO_DOCKER="$DOCKER_REPO/i386-hassio-supervisor"
    ;;
    "x86_64")
        MACHINE=${MACHINE:=qemux86-64}
        HOMEASSISTANT_DOCKER="$DOCKER_REPO/$MACHINE-homeassistant"
        HASSIO_DOCKER="$DOCKER_REPO/amd64-hassio-supervisor"
    ;;
    "arm" | "armv7l" | "armv6l")
        if [ -z $MACHINE ]; then
            echo "[ERROR] Please set machine for $ARCH"
            exit 1
        fi
        HOMEASSISTANT_DOCKER="$DOCKER_REPO/$MACHINE-homeassistant"
        HASSIO_DOCKER="$DOCKER_REPO/armhf-hassio-supervisor"
    ;;

So this caused the armhf version being installed and perhaps all other addons inherit this architecture from the supervisor?

frenck, 2019-06-02:
@ifol I'm not sure, you really need to consult with the Hass.io project on this one.

ifol, 2019-06-02:
I did so based on your suggestion: "I really think you should create an issue with Hass.io. From my end, there is nothing I can do. The images have the correct label, according to specifications."

Anyway, the install script is part of the Hass.io project, also Hass.io itself should throw an error or at least a warning when it's being installed with incorrect architecture for the system. So there appears to be at least a room for improvement for Hass.io in this area. Also, I'm not an expert on what exactly caused this misconfiguration or what is needed to rectify it and avoid it in the future, so it's best for the devs to have a look and decide what action needs to be done from Hass.io side if any.

I suspect this is an inconsistency caused by the recent changes to armhf/armv7 labeling and some code needs to be refactored in Hass.io

Thank you for your support!

ifol, 2019-06-02:
I did not realize that by making a comment I reopen the issue, sorry for that - closing, this is not an issue with ESPHome

closed 2019-06-02 by ifol

ifol, 2019-06-04:
OK, I've managed to make it work but I was not able to make it in a way that would feel like the "right way", armv7-homeassistant docker image is useless, it disabled ALL the addons when I managed to install it. So i decided to just start from scratch:

service docker stop
rm -rf /var/lib/docker/*
rm -rf /usr/share/hassio/*
service docker start
./hassio_install.sh -m raspberrypi2

I chose machine raspberrypi2 because there is no specific image for Orange Pi PC and the generic armv7 doesn't work:

"raspberrypi2": [
        "armv7",
        "armhf"
],

just the architectures I need - this worked well, installed a new instance and I added samba addon to upload a snapshot and restore:

root@orangepipc:~# docker ps     
CONTAINER ID        IMAGE                                      COMMAND                  CREATED             STATUS              PORTS               NAMES
88d077722c50        homeassistant/armv7-addon-samba            "/run.sh"                3 minutes ago       Up 3 minutes                            addon_core_samba
3eb3108a78bd        homeassistant/raspberrypi2-homeassistant   "/bin/entry.sh pytho…"   6 minutes ago       Up 6 minutes                            homeassistant
532ea23b27eb        homeassistant/armhf-hassio-supervisor      "python3 -m hassio"      13 minutes ago      Up 13 minutes                           hassio_supervisor

Notice the architecture mix that this produced, but that's apparently "normal".

When I restored the snapshot, it pulled esphome (armhf) and node-red (armv7), the ESPHome repository has not been restored, don't know why but when I added it manually it updated ESPHome just fine but now with armv7 arch ... it's a mess if you ask me but well, it works :)