esphome/issues #372: Ingress is not working in rev 1.13.x

ppmt, 2019-06-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

 Hass.io Add-on: ESPHome
 ESPHome Hass.io add-on for intelligently managing all your ESP8266/ESP32 devices.

 Add-on version: 1.13.2
 You are running the latest version of this add-on.
 System: null  (amd64 / qemux86-64)
 Home Assistant version: 0.93.2
 Supervisor version: 165

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32 Webcam (AI-thinker)
ESP32 Dev kits

**Affected component:**
Can't access webui via ingress

**Description of problem:**
Since the upgrade from 1.12.2 to 1.13.1 and after activating ingress I can no longer see the logs or compile my ESP modules. Instead I get some errors in the addon logs:

WARNING 403 GET /ace (0.0.0.0) 2.93ms
WARNING 403 GET /ace (0.0.0.0) 1.90ms
WARNING 403 GET /ace (0.0.0.0) 1.74ms
WARNING 403 GET /ace (0.0.0.0) 1.72ms
WARNING 403 GET /ace (0.0.0.0) 1.64ms
WARNING 403 GET /ace (0.0.0.0) 1.62ms
WARNING 403 GET /ace (0.0.0.0) 2.74ms
WARNING 403 GET /ace (0.0.0.0) 8.31ms

**Problem-relevant YAML-configuration entries:**
no problem with yaml configuration

**Traceback (if applicable):**
n/a

**Additional information and things you've tried:**
The workaround is to reactivate port 6052 in the addon and to use it rather than the ingress links. Thank you to user CBRHerms from the HA community.

[Link to relevant HA community post](https://community.home-assistant.io/t/cant-upload-firmware-since-v1-13-1/119490)

OttoWinter, 2019-06-01:
Hmm, this might be a tornado security issue.

Tornado (correctly) denies all websocket connections with an `Origin` header due to WS security reasons with a 403 status code.

What's interesting to me is that I can't replicate this - I tested with firefox/chrome/safari and none have had this error.

I think this might be a more complex issue; ESPHome will probably need to add a second authentication mechanism besides secure cookies to fix this issue, could take a while.

Could you provide more information about your setup? What browser is this? How did you access ingress (HA cloud or directly over IP)?

cbrherms, 2019-06-01:
As mentioned in linked community forum I'm in the same situation with *potentially* the same configuration. 

Add-on version: 1.13.2
System: qemux86-64
Home Assistant version: 0.93.2
Supervisor version: 165

@ottowinter after reading your comment it did remind me that my reverse proxy may be a factor.
 
At present I use pretty much exclusively my personal domain name URL to access homeassistant, with internal DNS pointing to a traefik container on another server, as well as a portforward from my router. 

Tested, and ingress is working without 403's if I use http://IP:8123 to access rather than https://my.domain.com

So, in my case it's potentially a misconfiguration on my part of traefik or using a reverse proxy is the main factor at least.

HA-TB303, 2019-06-01:
Same issue here since 1.13.2 and the same with 1.13.2:

When accessing hassio on http://localip:8123 everything is working, 

but when accessing it via my domain https://mydomain.com:40443/hassio/ingress/....

This error appears when validating, updating etc:

```19-06-01 15:49:33 ERROR (MainThread) [hassio.api.ingress] Ingress error: 403, message='Invalid response status'```

cbrherms, 2019-06-01:
Are you also using a reverse proxy between internet and homeassistant or does port 40443 point directly to 8123 on you HA server?

HA-TB303, 2019-06-01:
40443 points to the proxy.
The proxy points to: http://192.168.x.x:8123

HA-TB303, 2019-06-01:
FIY, node-red uses ingress as well and that works just fine.

cbrherms, 2019-06-01:
Well the common theme here seems to be that we're all using a reverse proxy in front of it. Without, ingress is working fine, so likely how something is passed by the proxy itself. 

For node red, you don't directly open a websocket connection to it through the ui so that wouldn't be affected by this proxy issue.

ppmt, 2019-06-01:
I can confirm as well that I am using homeassistant before a proxy (traefik) and that if I use the ip address rather than the domain then all is working fine!

cbrherms, 2019-06-01:
Anything in your traefik logs when you've got the add-on open or attempting to run something?

cbrherms, 2019-06-01:
> I'm using the nginx proxy manager addon, which works fine with all other addons.

I've just tested the nginx proxy manager addon and it worked on both http and https. Had to enable "Websocket support" on the proxy host entry though. Maybe that's your issue @HA-TB303  ?
I actually couldn't intentionally break that one without just disabling websocket support, which of course wouldn't be a bug. That resulted in a `400 Bad Request` error.

Did a fair amount of checking between direct ip, nginx proxy manager, traefik, with without https etc.

It looks like traefik is dropping the `Sec-WebSocket-Extensions` header on the return which i think is messing up the handshake due to the reliance on `permessage-deflate`.
Turns out there is an issue open on their repo found here:
https://github.com/containous/traefik/issues/4843

So, i *think* the issue myself and @ppmt are having is in fact an issue with traefik itself. It should automatically manage websockets so I don't think it's a misconfiguration.

@OttoWinter I may have got the wrong end of the stick here though with my basic understanding....

ppmt, 2019-06-02:
@cbrherms , Impressive troubleshooting skill. I am definitely not at that level of understanding web application.

Hopefully the Traefik people will come up with  a solution.

kwdavidson, 2019-06-03:
Just to throw a wrench in the works, I've been having the same problems with the error 400s. I'm also running reverse proxy, which continues to work when I disable ingress. However, I'm using Apache as the reverse proxy. I've been thinking all along that the issue was probably with the reverse proxy, but those settings are a bit of a black art to me. I was able to modify settings that I found online without fully understanding what every line item and option is for.

TEfjLONjDON, 2019-06-04:
Using caddy here.. and what's stranger still is after I update. I can't reach the Esphome dashboard at all. No mater if I use domain name. Ip+port. Or NabuCasa. Not reachable via ingress or browser tab. (Chrome) hassio is on Ubuntu via quemx86_64

zvldz, 2019-08-15:
I found a solution.
https://stackoverflow.com/questions/24851207/tornado-403-get-warning-when-opening-websocket

```
--- dashboard.py.orig   2019-08-15 15:17:05.501576096 +0300
+++ dashboard.py        2019-08-15 15:18:00.872871527 +0300
@@ -178,6 +178,9 @@
 # pylint: disable=abstract-method, arguments-differ
 @websocket_class
 class EsphomeCommandWebSocket(tornado.websocket.WebSocketHandler):
+    def check_origin(self, origin):
+        return True
+
     def __init__(self, application, request, **kwargs):
         super(EsphomeCommandWebSocket, self).__init__(application, request, **kwargs)
         self._proc = None
```

cbrherms, 2019-08-16:
> I found a solution.
> https://stackoverflow.com/questions/24851207/tornado-403-get-warning-when-opening-websocket

Do you by chance have a repo with this change in so i could test whether it fixes my issue also?

cbrherms, 2019-08-16:
Can confirm that this allows actions such as validate and build to run when going through proxies that previously would not. 
Tested both Kong and Traefik and it is running as expected.

TEfjLONjDON, 2019-08-20:
Can we expect this to be patched into the next update of ESPHome to facilitate normal use of ESPHome even when run behind a proxy?

brandond, 2019-08-20:
The proposed patch appears to bypass all origin checks, which seems like it could present a security issue. Where'd you find this? What's the intended use of the check_origin function?

OttoWinter, 2019-08-27:
Correct, bypassing the origin check is not ok - See [my first post here](https://github.com/esphome/issues/issues/372#issuecomment-497930398), I know it's the check_origin function, but just returning true there would be security vulnerability (external sites could hijack the esphome session through your browser).

But, it at least confirms that it is in fact the check_origin function and not something else.

The only solution I see would be to:

 a) Move websocket auth to a packet sent at the beginning - not such a big fan because it's kind of like rolling your own auth
 b) Add a new config option for supplying the trusted origins - the user needs to manually set this for each domain they want to access their esphome dashboard from.

OttoWinter, 2019-08-27:
@brandond Ideally, yes, but I don't think we get that passed from Hass.io - and I don't think Hass.io even knows it itself (what if the user has an alternate domain they use HA under?)

OttoWinter, 2019-08-30:
@brandond Only whitelisted CORS domain is `cast.home-assistant.io` (and only appears to have been added recently)

For WS autch they use option (a) as seen here: https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/components/websocket_api/auth.py#L53 (but with an access token system via jwt).

zvldz, 2019-10-21:
Will be any fixes in the new release?

OttoWinter, 2019-10-21:
Not that I know of.

KiLLeRRaT, 2020-02-27:
I have the same issue, IIS Reverse Proxy, can open the ESPHome dashboard, but when clicking Logs, shows Process Terminated. 

Browsing to the internal IP of Home Assistant, and then ESPHome -> Logs, works as it should.

How come ESPHome has this issue, where things like Grafana, and others don't have this issue? What's the difference in handling the ingress in those applications?

Cheers,

edenhaus, 2020-03-12:
@OttoWinter I think we should implement your suggest option b. Also the "http" integration of home-assistant let you specify the cors as list.
Everyone can copy the list, if they need websockets from outside.
If in the future hassio supports a way to provide the cors, than we can adapt the addon.

KiLLeRRaT, 2020-08-01:
Keep Alive!

ppmt, 2020-08-02:
Actually it is working for me again. I can now access logs and compile firmware using the side bar icon.

Addon version is 1.14.4 (from the HA Community add-ons)

KiLLeRRaT, 2020-08-02:
> Actually it is working for me again. I can now access logs and compile firmware using the side bar icon.
> 
> Addon version is 1.14.4 (from the HA Community add-ons)

Nope I still get Terminated Process straigh away when opening Logs on any entity when browsing from the sidebar icon

ppmt, 2020-08-03:
what version of the addon do you have and which version of HA?

KiLLeRRaT, 2020-08-03:
> what version of the addon do you have and which version of HA?

I'm running 
 - HA 0.113.2, and 
- ESPHome addon 1.14.5

Works when I browser to my HA IP directly on port 8123, but NOT when I browser via my IIS Reverse Proxy URL.  I do have websockets turned on, and a few other addins/sites needing websockets are working as they should through the IIS Reverse Proxy.

Cheers,

ppmt, 2020-08-04:
Ok not sure what to say then. I use Traefik for the reverse proxy.

cbrherms, 2020-08-04:
What version of traefik? I've not looked in to this myself since last year so I might have another go when I update traefik from v1.x to v2.x

ppmt, 2020-08-04:
> What version of traefik? I've not looked in to this myself since last year so I might have another go when I update traefik from v1.x to v2.x

I use Traefik V2.3

geiseri, 2020-09-17:
I also use Traefik 2.3 and have the same issue... I am confused though that I do not see any messages in my console.  All I see is "Process Terminated."  Do you think it could be an issue with ingress AND reverse proxy together?   I tried to mess around with the `relative_url` but that resulted in 404 errors.  One interesting thing though is that when websockets "work" they come from "Python/3.8 aiohttp/3.6.2", while all other resources come from nginx.  Is there anything special that needs to be done to make the websockets portion also use the HA ingress?

ppmt, 2020-09-17:
I really don't know what to advise @geiseri . All I can say is that it wasn't working and then recently I tried again and it was working .I assume it had been fixed with an update of ESPHome.

I have down nothing special to make it work.

mario-tux, 2020-09-17:
@ppmt, is it working for you? Under `https://` using a proxy like NGINX addons? Can you describe your setup, please?

KiLLeRRaT, 2020-09-17:
Yeah I'm still having that issue with IIS reverse proxy.  @geiseri are you using IIS for reverse proxying?

ppmt, 2020-09-17:
Yes it definitely works for me @mario-tux 

I am using Traefik v2.3 for proxy  and it takes care of the TLS via Let'sEncrypt.
I am using HA 0.115 (just updated) and ESPHome 1.15.1. 

If I click on ESPHOME in the sidebar then it opens up the addons as expected with all my config. If I then click on Logs it opens up the logs and display them correctly. 
The same is true for Edit an Upload.

geiseri, 2020-09-17:
I am using traefik 2.3 as the SSL reverse proxy external to my HA setup.  Let update to 0.115 for HA and see if that makes a difference.

Dacesilian, 2020-11-07:
I have also problem with esphome -> device -> Logs. I'm using Apache as reverse proxy and I don't know, how to configure it. Error in Chrome console is 'WebSocket connection to 'wss://ha..../api/hassio_ingress/5sa4d65a46d4as6d/ace' failed: One or more reserved bits are on: reserved1 = 0, reserved2 = 1, reserved3 = 1'

NobleKangaroo, 2020-12-01:
*(Sorry this is so long - I spent *hours* on a deep dive into this)*

I'm having the exact same issue, but I came up with a workaround for those that want to do this on their own setup with relative ease. But first, some details about what's actually going on here.

As noted in Tornado's [`tornado.websocket` documentation](https://www.tornadoweb.org/en/stable/websocket.html),
> Web browsers allow any site to open a websocket connection to any other, instead of using the same-origin policy that governs other network access from JavaScript. This can be surprising and is a potential security hole, so since Tornado 4.0 `WebSocketHandler` requires applications that wish to receive cross-origin websockets to opt in by overriding the `check_origin` method (see that method’s docs for details). Failure to do so is the most likely cause of 403 errors when making a websocket connection.

And their [`WebSocketHandler.check_origin` documentation](https://www.tornadoweb.org/en/stable/websocket.html#tornado.websocket.WebSocketHandler.check_origin),
> Should return True to accept the request or False to reject it. By default, rejects all requests with an origin on a host other than this one.
>
> To accept all cross-origin traffic (which was the default prior to Tornado 4.0), simply override this method to always return True:
> ```python
> def check_origin(self, origin):
>     return True
> ```
> To allow connections from any subdomain of your site, you might do something like:
> ```python
> def check_origin(self, origin):
>     parsed_origin = urllib.parse.urlparse(origin)
>     return parsed_origin.netloc.endswith(".mydomain.com")
> ```

In short, as @OttoWinter [mentioned above](https://github.com/esphome/issues/issues/372#issuecomment-497930398), the `Origin` header being set is preventing this from working. Due to the nonexistence of a `check_origin` override, **all** cross-origin traffic is blocked. So if your Home Assistant is `hass.internal.com` and you're using `hass.external.com` to access it, you'll be denied. If however in your reverse proxy configuration you were to remove the `Origin` header, or override it to be the same as the `Host` header, you will be welcomed with milk and cookies at the front door and invited to come inside. This is because Tornado goes the full-open route and [allows full access to clients that omit this header](https://www.tornadoweb.org/en/stable/websocket.html#tornado.websocket.WebSocketHandler.check_origin):
> The `origin` argument is the value of the `Origin` HTTP header, the url responsible for initiating this request. **This method is not called for clients that do not send this header; such requests are always allowed** (because all browsers that implement WebSockets support this header, and non-browser clients do not have the same cross-site security concerns).

I found the actual check that's performed, [websocket.py:L536](https://github.com/tornadoweb/tornado/blob/master/tornado/websocket.py#L536). First, I copied `dashboard.py` to the `/data` volume so I can edit it (e.g. with vi) from my Home Assistant host:
```bash
# docker exec -it $(docker ps | awk '($(NF) ~ "^addon_.*_esphome$"){ print $1; }') bash -c 'cp /opt/esphome/esphome/dashboard/dashboard.py /data/dashboard.py'
```

Then I edited the file from my host using vi:
```bash
# vi /mnt/data/supervisor/addons/data/*_esphome/dashboard.py
```

From the actual check above I added the import line, a slightly modified version of the origin check, and a couple debug lines to `dashboard.py`:
```python
  15 | from urllib.parse import urlparse
...
 190 |     def check_origin(self, origin):
 191 |         parsed_origin = urlparse(origin)
 192 |         origin = parsed_origin.netloc
 193 |         origin = origin.lower()
 194 |
 195 |         host = self.request.headers.get("Host")
 196 |
 197 |         if origin == host:
 198 |             _LOGGER.info("[allow] Host '%s' == Origin '%s'", host, origin)
 199 |         else:
 200 |             _LOGGER.warning("[deny] Host '%s' !== Origin '%s'", host, origin)
 201 |
 202 |         return origin == host
```

Next I copied the dashboard file from the data volume to the running container, effectively the opposite from a minute ago:
```bash
# docker exec -it $(docker ps | awk '($(NF) ~ "^addon_.*_esphome$"){ print $1; }') bash -c 'cp /data/dashboard.py /opt/esphome/esphome/dashboard/dashboard.py'
```

Then I killed the ESPHome processes without restarting the container, which causes s6-supervise to restart them so that my above changes to the running volume don't get reset when the container restarts:
```bash
> login
# docker exec -it $(docker ps | awk '($(NF) ~ "^addon_.*_esphome$"){ print $1; }') bash -c 'pkill -9 esphome'
```

Now when I try to use the ESPHome validator or compiler, I get a useful log in the addon's log:
```log
WARNING Host: hass.internal.com:443 Origin: hass.external.com
```

[... 17 lines cut ...]
    #RequestHeader set Origin "http://hass.internal.com:443"
    #RequestHeader set Origin "https://hass.internal.com:443"

    ProxyPass           /   https://hass.internal.com/
    ProxyPassReverse    /   https://hass.internal.com/

    ErrorLog    logs/ssl_error_log
    TransferLog logs/ssl_access_log
    LogLevel warn
</VirtualHost>
```

All said and done, I don't like this and it shouldn't be *the* solution. What really should happen is a sane `check_origin` routine to be performed, ultimately allowing or denying cross-origin websocket traffic - perhaps a con [... cut ...] from cross-domain origins. Something like:
```yaml
trusted-origins:
  - abc.xyz.com
  - ddd.efg.hi
```

And some mockup Python which, depending on how the logic will be addressed, probably needs some love (and actual calls to read the addon's config):
```python
return origin is None or (type(cfgTrustedOrigins) is list and origin in cfgTrustedOrigins)
```

At least this way, there's a *possibility* of this working for those of us that use reverse proxy, but it doesn't open anything up by default.

NobleKangaroo, 2020-12-01:
*(Sorry this is so long - I spent *hours* on a deep dive into this)*

I'm having the exact same issue, but I came up with a workaround for those that want to do this on their own setup with relative ease. But first, some details about what's actually going on here.

As noted in Tornado's [`tornado.websocket` documentation](https://www.tornadoweb.org/en/stable/websocket.html),
> Web browsers allow any site to open a websocket connection to any other, instead of using the same-origin policy that governs other network access from JavaScript. This can be surprising and is a [...]

And their [`WebSocketHandler.check_origin` documentation](https://www.tornadoweb.org/en/stable/websocket.html#tornado.websocket.WebSocketHandler.check_origin),
> Should return True to accept the request or False to reject it. By default, rejects all requests with an origin on a host other than this one.
>
> To accept all cross-origin traffic (which was the [...]

In short, as @OttoWinter [mentioned above](https://github.com/esphome/issues/issues/372#issuecomment-497930398), the `Origin` header being set is preventing this from working. Due to the nonexistence of a `check_origin` override, **all** cross-origin traffic is blocked. So if your Home Assistant is `hass.internal.com` and you're using `hass.external.com` to access it, you'll be denied. If however in your reverse proxy configuration you were to remove the `Origin` header, or override it to be the same as the `Host` header, you will be welcomed with milk and cookies at the front door and invited to come inside. This is because Tornado goes the full-open route and [allows full access to clients that omit this header](https://www.tornadoweb.org/en/stable/websocket.html#tornado.websocket.WebSocketHandler.check_origin):
> The `origin` argument is the value of the `Origin` HTTP header, the url responsible for initiating this request. **This method is not called for clients that do not send this header; such requests [...]

I found the actual check that's performed, [websocket.py:L536](https://github.com/tornadoweb/tornado/blob/master/tornado/websocket.py#L536). First, I copied `dashboard.py` to the `/data` volume so I can edit it (e.g. with vi) from my Home Assistant host:
```bash
# docker exec -it $(docker ps | awk '($(NF) ~ "^addon_.*_esphome$"){ print $1; }') bash -c 'cp /opt/esphome/esphome/dashboard/dashboard.py /data/dashboard.py'
```

Then I edited the file from my host using vi:
```bash
# vi /mnt/data/supervisor/addons/data/*_esphome/dashboard.py
```

From the actual check above I added the import line, a slightly modified version of the origin check, and a couple debug lines to `dashboard.py`:
```python
  15 | from urllib.parse import urlparse
...
 190 |     def check_origin(self, origin):
 191 |         parsed_origin = urlparse(origin)
 192 |         origin = parsed_origin.netloc
 193 |         origin = origin.lower()
 194 |
 195 |         host = self.request.headers.get("Host")
 196 |
 197 |         if origin == host:
 198 |             _LOGGER.info("[allow] Host '%s' == Origin '%s'", host, origin)
 199 |         else:
 200 |             _LOGGER.warning("[deny] Host '%s' !== Origin '%s'", host, origin)
 201 |
 202 |         return origin == host
```

Next I copied the dashboard file from the data volume to the running container, effectively the opposite from a minute ago:
```bash
# docker exec -it $(docker ps | awk '($(NF) ~ "^addon_.*_esphome$"){ print $1; }') bash -c 'cp /data/dashboard.py /opt/esphome/esphome/dashboard/dashboard.py'
```

Then I killed the ESPHome processes without restarting the container, which causes s6-supervise to restart them so that my above changes to the running volume don't get reset when the container restarts:
```bash
# docker exec -it $(docker ps | awk '($(NF) ~ "^addon_.*_esphome$"){ print $1; }') bash -c 'pkill -9 esphome'
```

Now when I try to use the ESPHome validator or compiler, I get a useful log in the addon's log:
```log
WARNING Host: hass.internal.com:443 Origin: hass.external.com
```

Okay, okay... I won't keep you waiting any longer! 🥁 *(drumroll please...)* Here's all you need to make that happen for us Apache users. Nginx users can Google how to set headers. 😄 Take note of the **RequestHeader unset Origin** and other directives - those are the magic sauce that will make ESPHome work through a reverse proxy. The domain name and port must match exactly, but a missing or incorrect scheme (http, https, or none) is fine. So in that regard, since effectively we're achieving the same result regardless of how we get there, it's probably easiest to just delete the `Origin` header altogether which will get us what we want.
```conf
# Home Assistant
<VirtualHost *:443>
    ServerName hass.external.com

    SSLEngine On
    SSLProxyEngine On

    RewriteEngine On
    RewriteCond %{HTTP:Upgrade} websocket [NC]
[... 15 lines cut ...]
</VirtualHost>
```

All said and done, I don't like this and it shouldn't be *the* solution. What really should happen is a sane `check_origin` routine to be performed, ultimately allowing or denying cross-origin websocket traffic - perhaps a config directive that can be set in the addon, to allow requests from cross-domain origins. Something like:
```yaml
trusted-origins:
  - abc.xyz.com
  - ddd.efg.hi
```

And an idea for the Python side of things which needs some love and actual calls to read the addon's config from someone more familiar with that side of things:
```python
[... cut ...]
This way it's possible for those of us that use a reverse proxy to grant ourselves access without breaking anything for those who don't use a reverse proxy (their `origin` will be `None`) by default (e.g. if the config is empty). As far as security goes, it's still possible to send requests to the service by editing the `Origin` header, but that's not the only security check in place - If I understand things correctly, Home Assistant won't let you talk to the websocket without logging in and getting a token first anyways.

Wondering what everyone's thoughts here are? Would be nice to fix this upstream; it feels very hacky doing it this way.

NobleKangaroo, 2020-12-01:
*(Sorry this is so long - I spent *hours* on a deep dive into this)*

I'm having the exact same issue, but I came up with a workaround for those that want to do this on their own setup with relative ease. But first, some details about what's actually going on here.

As noted in Tornado's [`tornado.websocket` documentation](https://www.tornadoweb.org/en/stable/websocket.html),
> Web browsers allow any site to open a websocket connection to any other, instead of using the same-origin policy that governs other network access from JavaScript. This can be surprising and is a [...]

And their [`WebSocketHandler.check_origin` documentation](https://www.tornadoweb.org/en/stable/websocket.html#tornado.websocket.WebSocketHandler.check_origin),
> Should return True to accept the request or False to reject it. By default, rejects all requests with an origin on a host other than this one.
>
> To accept all cross-origin traffic (which was the [...]

In short, as @OttoWinter [mentioned above](https://github.com/esphome/issues/issues/372#issuecomment-497930398), the `Origin` header being set is preventing this from working. Due to the nonexistence of a `check_origin` override, **all** cross-origin traffic is blocked. So if your Home Assistant is `hass.internal.com` and you're using `hass.external.com` to access it, you'll be denied. If however in your reverse proxy configuration you were to remove the `Origin` header, or override it to be the same as the `Host` header, you will be welcomed with milk and cookies at the front door and invited to come inside. This is because Tornado goes the full-open route and [allows full access to clients that omit this header](https://www.tornadoweb.org/en/stable/websocket.html#tornado.websocket.WebSocketHandler.check_origin):
> The `origin` argument is the value of the `Origin` HTTP header, the url responsible for initiating this request. **This method is not called for clients that do not send this header; such requests [...]

I found the actual check that's performed, [websocket.py:L536](https://github.com/tornadoweb/tornado/blob/master/tornado/websocket.py#L536). First, I copied `dashboard.py` to the `/data` volume so I can edit it (e.g. with vi) from my Home Assistant host:
```bash
# docker exec -it $(docker ps | awk '($(NF) ~ "^addon_.*_esphome$"){ print $1; }') bash -c 'cp /opt/esphome/esphome/dashboard/dashboard.py /data/dashboard.py'
```

Then I edited the file from my host using vi:
```bash
# vi /mnt/data/supervisor/addons/data/*_esphome/dashboard.py
```

From the actual check above I added the import line, a slightly modified version of the origin check, and a couple debug lines to `dashboard.py`:
```python
  15 | from urllib.parse import urlparse
...
 190 |     def check_origin(self, origin):
 191 |         parsed_origin = urlparse(origin)
 192 |         origin = parsed_origin.netloc
 193 |         origin = origin.lower()
 194 |
 195 |         host = self.request.headers.get("Host")
 196 |
 197 |         if origin == host:
 198 |             _LOGGER.info("[allow] Host '%s' == Origin '%s'", host, origin)
 199 |         else:
 200 |             _LOGGER.warning("[deny] Host '%s' !== Origin '%s'", host, origin)
 201 |
 202 |         return origin == host
```

Next I copied the dashboard file from the data volume to the running container, effectively the opposite from a minute ago:
```bash
# docker exec -it $(docker ps | awk '($(NF) ~ "^addon_.*_esphome$"){ print $1; }') bash -c 'cp /data/dashboard.py /opt/esphome/esphome/dashboard/dashboard.py'
```

Then I killed the ESPHome processes without restarting the container, which causes s6-supervise to restart them so that my above changes to the running volume don't get reset when the container restarts:
```bash
# docker exec -it $(docker ps | awk '($(NF) ~ "^addon_.*_esphome$"){ print $1; }') bash -c 'pkill -9 esphome'
```

Now when I try to use the ESPHome validator or compiler, I get a useful log in the addon's log:
```log
WARNING Host: hass.internal.com:443 Origin: hass.external.com
```

Okay, okay... I won't keep you waiting any longer! 🥁 *(drumroll please...)* Here's all you need to make that happen for us Apache users. Nginx users can Google how to set headers. 😄 Take note of the **RequestHeader unset Origin** and other directives - those are the magic sauce that will make ESPHome work through a reverse proxy. The domain name and port must match exactly, but a missing or incorrect scheme (http, https, or none) is fine. So in that regard, since effectively we're achieving the same result regardless of how we get there, it's probably easiest to just delete the `Origin` header altogether which will get us what we want.
```conf
# Home Assistant
<VirtualHost *:443>
    ServerName hass.external.com

    SSLEngine On
    SSLProxyEngine On

    RewriteEngine On
    RewriteCond %{HTTP:Upgrade} websocket [NC]
[... 17 lines cut ...]

All said and done, I don't like this and it shouldn't be *the* solution. What really should happen is a sane `check_origin` routine to be performed, ultimately allowing or denying cross-origin websocket traffic - perhaps a config directive that can be set in the addon, to allow requests from cross-domain origins. Something like:
```yaml
trusted-origins:
  - abc.xyz.com
  - ddd.efg.hi
```

And an idea for the Python side of things (which arguably needs some love from someone more familiar with the actual calls to read the addon's config):
```python
def check_origin(self, origin):
[... cut ...]
This way it's possible for those of us that use a reverse proxy to grant ourselves access without breaking anything for those who don't use a reverse proxy (their `origin` will be `None`) by default (e.g. if the config is empty). As far as security goes, it's still possible to send requests to the service by editing the `Origin` header, but that's not the only security check in place - If I understand things correctly, Home Assistant won't let you talk to the websocket without logging in and getting a token first anyways.

Wondering what everyone's thoughts here are? Would be nice to fix this upstream; it feels very hacky doing it this way.

geiseri, 2021-04-09:
This is still an issue in `1.16.2`

nohn, 2021-05-02:
Using esphome 1.16.2 and Apache 2.4.38-3+deb10u4 for me everything including logs and upload (didn't try the wizard though) works using this config:

```
RewriteEngine on

RequestHeader unset Origin
ProxyPass /esphome http://1.2.3.4:6052
ProxyPassReverse /esphome http://1.2.3.4:6052
RewriteCond %{HTTP:Upgrade} websocket [NC]
RewriteCond %{HTTP:Connection} upgrade [NC]
RewriteRule ^/esphome/(.*) "ws://1.2.3.4:6052/$1" [P,L]  
```

CarlosGS, 2021-06-16:
The solution in https://github.com/esphome/issues/issues/372#issuecomment-521621395 works perfectly. Tested with the NGINX plugin, this patch fixes ingress for ESPHome.

> just returning true there would be a security vulnerability (external sites could hijack the esphome session through your browser)

I'm a bit puzzled that this would be a vulnerability when using ingress - I thought of it as a "safe tunnel" that forces all incoming traffic through the secure HASS session. Hence any domain verification would be pointless after that authentication (specially HASS<->ESPHome).
Would you be so kind to explain what I'm getting wrong here? Aren't cookies tied to the main external domain? :confused:

I was going to propose to disable origin verification automatically when a global port isn't specified (the recommended ingress-only access). Looking forward to your feedback.

geiseri, 2021-06-17:
@CarlosGS I think its in cases you are not using a reverse proxy.  it is possible to use it without home assistant.  i don't know why you would want to but you can.  that being said I think running services like this should always happen behind a reverse proxy.

asaworld, 2021-08-19:
I think this is the same issue and this comment solves it. (https://github.com/esphome/issues/issues/1035#issuecomment-697560334)

thefunkygibbon, 2021-09-02:
> @CarlosGS I think it's in cases you are not using a reverse proxy. it is possible to use it without home-assistant. I don't know why you would want to but you can. that being said I think running services like this should always happen behind a reverse proxy. @OttoWinter if someone was to PR this would it be considered?

because not everyone is /can run supervised version of HA.  You need supervised to run 'addons'
myself included.

CarlosGS, 2021-09-08:
[image]
When running from Nabu Casa remote, the ESPHome dashboard shows "401: Unauthorized".
Could it be due to this same issue?

geiseri, 2021-09-19:
I am sure they needed to solve the proxy issue a little differently since they are putting it over the WAN.  There has to be some stupid magic with the local proxy setup.  I can't get it to work but I think the issue is that I access my home assistant from behind a reverse proxy using trafik.

CumpsD, 2023-06-25:
> #RequestHeader set Origin

@NobleKangaroo thanks for this research!

Fixed it on my nginx by setting Host and Origin to be identical with the following config:

```
      location / {
        proxy_pass http://127.0.0.1:6052;

        proxy_http_version 1.1;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection "upgrade";

        proxy_set_header Host "ha.local:6051";
        proxy_set_header Origin "https://ha.local:6051";
        proxy_set_header X-Real-IP $remote_addr;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header X-Forwarded-Proto $scheme;
      }
```

geiseri, 2023-07-21:
I wonder if this could be closed with an update to the documentation since it seems to work for me now with traefik 2.9 and the following configuration:

```yaml
http:
  middlewares:
    add-proxy-headers:
        headers:
          hostsProxyHeaders:
            - "X-Forwarded-For"
            - "X-Forwarded-Proto"
            - "X-Forwarded-Host"
            - "X-Forwarded-Method"
          stsSeconds: 31536000
          forceSTSHeader: true
          stsPreload: true
          stsIncludeSubdomains: true
          browserXssFilter: true
          customRequestHeaders:
             X-Forwarded-Proto: https
```

pergolafabio, 2023-07-21:
How can it work with the basic nginx addon? Do I need to add that http config in home assistant itself?

Lenart12, 2024-10-10:
I also had an error, this is how i fixed it with Apache
```xml
<VirtualHost *:80>
	ServerName home.example.com 
	Redirect / https://home.example.com/
</VirtualHost>

<VirtualHost *:443>
    ServerName home.example.com 

    ProxyPass / http://homeassistant.lan:8123/
    ProxyPassReverse / http://homeassistant.lan:8123/

    <Location "/">
        ProxyPass "ws://homeassistant.lan:8123/"
    </Location>

    # Main part of the fix here: set Host and Origin header
    RequestHeader set Host "homeassistant.lan:8123"
    RequestHeader set Origin "http://homeassistant.lan:8123"
            
    ErrorLog ${APACHE_LOG_DIR}/error-home.log
    CustomLog ${APACHE_LOG_DIR}/access-home.log combined

    Header always set Strict-Transport-Security "max-age=31536000; includeSubDomains"
    Include /etc/letsencrypt/options-ssl-apache.conf
    SSLCertificateFile /etc/letsencrypt/live/example.com/fullchain.pem
    SSLCertificateKeyFile /etc/letsencrypt/live/example.com/privkey.pem
</VirtualHost>
```

dekiesel, 2025-05-03:
> [@NobleKangaroo](https://github.com/NobleKangaroo) thanks for this research!
>
> Fixed it on my nginx by setting Host and Origin to be identical with the following config:
>
> ```
>       location / [...]

This doesn't seem to work for me, but it might be due to my setup: I have esphome on a machine called èsphome.lan`, and nginx on ǹginx.lan`. Home Assistant runs on a different machine.

All ports are standard ports, but this:

```
    location / {

        include /config/nginx/proxy.conf;
        resolver 127.0.0.11 valid=30s;
        set $upstream_app esphome.lan;
        set $upstream_port 6052;
        set $upstream_proto http;
        proxy_pass $upstream_proto://$upstream_app:$upstream_port;
	proxy_set_header Host $host:$upstream_port;
	#proxy_set_header Host $host:$upstream_port;
        proxy_redirect http:// https://;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header X-Real-IP $remote_addr;
	proxy_set_header Upgrade $http_upgrade;
	proxy_set_header Connection $connection_upgrade;

    }
```
doesn't work. Could anybody tell me what I am doing wrong?