esphome/issues #374: ESP32-CAM stops generating images working

Coren4, 2019-06-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on Ubuntu Server 18
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32-CAM board. Something like this -> https://www.aliexpress.com/item/Low-Power-ESP32-ESP32-CAM-WiFi-Module-ESP32-S-Kit-Serial-Dual-core-WiFi-Bluetooth-Development/33002147011.html?spm=a2g0s.9042311.0.0.27424c4dsuiYiS
**Affected component:**

https://esphome.io/components/esp32_camera.html

**Description of problem:**
When I start board everything is fine.
In logs I see;
"[08:53:55][D][esp32_camera:157]: Got Image: len=29818"
But after some time like 24h it stops generating image, and mentioned log doesn't appear any more.

In my HA UI it is added like that:
```
camera_image: camera.cam_on_node_04
entities:
  - light.node_04_light
title: 04 Kamera TV
type: picture-glance
```

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp_home_node_04
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "ssid"
  password: "pass"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

# ESP32-CAM
esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32

  name: cam_on_node_04
  # ... Frame Settings
  max_framerate: 1 fps
  brightness: 0
  contrast: 0
  saturation: 0

# Flashlight
output:
  - platform: gpio
    pin: GPIO4
    id: gpio_4
light:
  - platform: binary
    output: gpio_4
    name: node_04_light

```

**Traceback (if applicable):**

```

```

**Additional information and things you've tried:**
So as You can see in my yaml I already tried to limit max_framerate to 1FPS, because I read about overheating, but still it didn't helped.

I have already heatsink added on main chip of ESP32, but apparently it did not help.

I see that it already change between max and idle mode, when I click on that card in HA - it starts to generate images faster in logs, and when card is closed it slows down.

Maybe it is overheating, is there anything more I could do to overcome that issue? Are there any cheap boards that You could suggest for 24h monitoring? Because maybe I bought wrong ones?

I have 3 boards like that with similar problems.

OttoWinter, 2019-06-01:
Please try the test_pattern option.

If it doesn't happen with test_pattern, it's a overheating problem. You can reduce the idle framerate under 1fps (even 0fps I think) to counter that or use better heat sinks.

j23chan, 2019-06-03:
Had pretty much the exact same issue.  I tried upgrading to a 2A power plug and it did not help.  Then i upgraded again to a 2.4a plug with the following settings:

wifi:
  fast_connect: on
  power_save_mode: light

esp32_camera:
  max_framerate:  5fps

I doubt any of the settings really changed much.  I think it's really the power.  It's been 2 days and still no issues.  If it konks out again i'll put in some more investigative work.

Coren4, 2019-06-04:
But from what I read, these boards should not use more than 500mA, so anything above that shouldn't matter, also when I had problems with power source I was getting "Brownout detector was triggered" in my logs, so I think that power source is not problem here. But You might be onto something, I read that these board like to overheat when using WiFi constantly, so maybe `power_save_mode: light` helped to cool it down a little, I will try it too.

For now I have tested:
idle_framerate: 0.05 fps
max_framerate: 0.5 fps
...and it worked stable for two days.

Also I tried test patern and it worked well without problems.

OttoWinter, 2019-06-04:
> But from what I read, these boards should not use more than 500mA

Exactly, the ESP32 itself is not capable of consuming more.

Though maybe the camera adds to that a bit - but 2A are definitely not required.

What could be an issue is that the voltage supply is not "clean" - smaller ripples in volage can lead to issues with these sensitive electronics like cameras.

OttoWinter, 2019-07-11:
Closing due to inactivity - and it's likely a power supply issue.

closed 2019-07-11 by OttoWinter